esphome/issues #377: ESP32 xiaomi mijia flower frequent updates

sendorm, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS on Virtual Environment

**ESP (ESP32/ESP8266, Board/Sonoff):**

  platform: ESP32
  board: esp-wrover-kit

**Affected component:**

https://esphome.io/components/sensor/xiaomi_miflora.html

**Description of problem:**
The code below checks for the mijia values every 30-40 seconds, not the intended per 3 hours. 

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_tracker:
  scan_interval: 10800s

sensor:
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:64:36:6E'
    temperature:
      name: "Bonzai Temperature"
    moisture:
      name: "Bonzai Moisture"
    illuminance:
      name: "Bonzai Illuminance"
    conductivity:
      name: "Bonzai Soil Conductivity"
    battery_level:
      name: "Bonzai Battery Level"
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-01:
The variable sets the scan interval on the ESP32 BLE API - it does not control how often miflora updates are received; If the MiFlora decides to send out more advertising packets during the scan interval ESPHome will still publish them

Use sensor filters like sliding_window_moving_average to have fewer values pushed to HA.

sendorm, 2019-06-01:
I am more concerned about the battery life of the miflora sensors. So is there a way to make the esp32_ble_tracker only chech per 3 hours? More in line with the HASS component's usage with the code below. I've also tried the `miflora-mqtt-daemon`. It also has a `period` variable with the same usage. But ESPHome is much more robust and easier to integrate.  

```yaml
  - platform: miflora
    mac: 'C4:7C:8D:64:36:6E'
    name: Bonzai
    force_update: false
    monitored_conditions:
      - moisture
      - light
      - temperature
      - conductivity
      - battery
    scan_interval: 10800
```

OttoWinter, 2019-06-01:
See https://github.com/esphome/issues/issues/107#issuecomment-470898522

ESPHome only passively listens to what the miflora already publishes on its own - so esphome has exactly 0% impact on battery life of miflora because it's 100% passive. The scan_interval doesn't change that.

sendorm, 2019-06-01:
Nice to hear that.  Thank you for the information.

closed 2019-06-01 by sendorm