esphome/issues #378: list index out of range

potman69, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
vitual image

Please provide details about your environment below this line. -->

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 D1 mini pro

Please provide details about which ESP you're using below.
Pool timer

**Affected component:**
When validating I get the error but previously work before last two updates that happened this week
Please add the link to the documentation at https://esphome.io/index.html of the component in question.
-->

**Description of problem:**

INFO Reading configuration...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 466, in run_esphome
    config = read_config(args.verbose)
  File "/opt/esphome/esphome/config.py", line 793, in read_config
    res = load_config()
  File "/opt/esphome/esphome/config.py", line 653, in load_config
    return _load_config()
  File "/opt/esphome/esphome/config.py", line 641, in _load_config
    result = validate_config(config)
  File "/opt/esphome/esphome/config.py", line 551, in validate_config
    validated = comp.config_schema(input_conf)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 106, in validate_mapping
    key_value_map[key.schema] = key.default()
  File "/opt/esphome/esphome/components/time/__init__.py", line 104, in detect_tz
    return convert_tz(tz)
  File "/opt/esphome/esphome/components/time/__init__.py", line 74, in convert_tz
    dstoffset1 = transition_info[idx1][1]
IndexError: list index out of range

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: pool_controller
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "cheaptrick"
  password: "***********"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '**********'

ota:
  password: '***********'

text_sensor:
  - platform: version
    name: "ESPHome Version"

dallas:
  - pin: GPIO16

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Pool Pump"
    on_press:
      - switch.toggle: relay_pump

  - platform: gpio
    pin:
      number: GPIO2
      mode: INPUT_PULLUP
      inverted: True
    name: "Pool Pump"
    on_press:
      - switch.toggle: relay_light

switch:
  - platform: gpio
    name: "Pool Pump"
    pin: GPIO12
    id: relay_pump

  - platform: gpio
    name: "Pool Light"
    pin: GPIO13
    id: relay_light
    
time:
  - platform: sntp
    # ...
    on_time:
      # Every morning on weekdays in summertime
      - seconds: 0
        minutes: 00
        hours: 09
        days_of_week: 1,2,3,4,5,6,7
        months: 1,2,3,4,5,10,11,12
        then:
          - switch.turn_on: relay_pump
          - delay: 6 hours
          - switch.turn_off: relay_pump
      # Every morning on weekdays in wintertime
      - seconds: 0
        minutes: 00
        hours: 12
        days_of_week: 1,2,3,4,5,6,7
        months: 6,7,8,9
        then:
          - switch.turn_on: relay_pump
          - delay: 60 minutes
          - switch.turn_off: relay_pump    
    
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Pool Pump"
    update_interval: 60s
    
  - platform: uptime
    name: Uptime Sensor Pool Pump

  - platform: dallas
    address: 0xA40000031F055028
    name: "Pool Outlet"
  - platform: dallas
    address: 0xDD0000031EFB0428
    name: "Pool Return"
```

**Traceback (if applicable):**

```
INFO Reading configuration...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 466, in run_esphome
    config = read_config(args.verbose)
  File "/opt/esphome/esphome/config.py", line 793, in read_config
    res = load_config()
  File "/opt/esphome/esphome/config.py", line 653, in load_config
    return _load_config()
  File "/opt/esphome/esphome/config.py", line 641, in _load_config
    result = validate_config(config)
  File "/opt/esphome/esphome/config.py", line 551, in validate_config
    validated = comp.config_schema(input_conf)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 106, in validate_mapping
    key_value_map[key.schema] = key.default()
  File "/opt/esphome/esphome/components/time/__init__.py", line 104, in detect_tz
    return convert_tz(tz)
  File "/opt/esphome/esphome/components/time/__init__.py", line 74, in convert_tz
    dstoffset1 = transition_info[idx1][1]
IndexError: list index out of range
```

**Additional information and things you've tried:**
recreating the yaml from start after deleting deleting it first

OttoWinter, 2019-06-01:
Fixed by https://github.com/esphome/esphome/pull/586

closed 2019-06-01 by OttoWinter

definitio, 2019-06-09:
I have a similar problem with 1.13.5. Timezone: Europe/Saratov.
```
INFO Reading configuration...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File ".venv/bin/esphome", line 10, in <module>
    sys.exit(main())
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/__main__.py", line 466, in run_esphome
    config = read_config(args.verbose)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/config.py", line 803, in read_config
    res = load_config()
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/config.py", line 663, in load_config
    return _load_config()
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/config.py", line 651, in _load_config
    result = validate_config(config)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/voluptuous_schema.py", line 106, in validate_mapping
    key_value_map[key.schema] = key.default()
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/components/time/__init__.py", line 112, in detect_tz
    return convert_tz(tz)
  File "/home/username/Development/fw-lroom-aquarium/.venv/lib/python2.7/site-packages/esphome/components/time/__init__.py", line 87, in convert_tz
    utcoffset_on, _, tzname_on = transition_info[idx1]
IndexError: list index out of range
```

OttoWinter, 2019-06-09:
@definitio Hmm that must be an issue with pytz - their transition_info table contains an entry for 2038 (which means the timezone would change on 19th jan 2038 after the most recent timezone change in 2016); but looking through the internet it doesn't look like such a timezone change is scheduled (nor would it make sense to schedule a tz change 18 years in advance).

Will try to create a workaround.

glmnet, 2019-06-10:
ahhh yes, this branch fixes Argentina time zone too