esphome/issues #380: APDS-9960 Gesture Sensor is not listed in ESPHome control panel

artafakhari, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8226 - WEMOS D1 R2 & MINI
**Affected component:**

**Description of problem:**
I can't see the APDS-9960 in the devices list.
I tried all bin files and also compiled my own formware.bin in platformio.
But can't see this component in the device section lists in ESPHome control panel.
I2c Scan in Tools get the following results:
Supported Devices:
PCF8574A
TSL2561
APDS9960
Which means the device is recognized by the firmware
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-01:
I'm not sure you're talking about ESPHome here.

> I can't see the APDS-9960 in the devices list.

What devices list?

> I tried all bin files 

ESPHome has no pre-compiled bin files

> ESPHome control panel
> Supported Devices:
> PCF8574A
> TSL2561
> APDS9960

What control panel?

> I2c Scan in Tools get the following results:

There's no such thing in ESPHome that would output that list

artafakhari, 2019-06-01:
sorry it was my mistake
please delete the issue

closed 2019-06-01 by OttoWinter