esphome/issues #381: Samsung remote receiver

DelukaDZ, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Host system HassOS 2.11
Hass.io supervisor version 165
ESPHome 1.13.2
Home Assistant 0.93.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP 8266 Wemos Lolin nodemcu v3

**Affected component:**

Remote Receiver

**Description of problem:**

the problem is that the sensor triggers on every button i press on the Samsung remote and not only on the poweroff button.
its not triggered with any other remote i have

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  pin: 
    number: GPIO14
    inverted: yes
    mode: INPUT_PULLUP
  dump: samsung

binary_sensor:
  - platform: remote_receiver
    name: "Samsung Remote"
    samsung:
      data : 0xE0E040BF
    filters:
      - delayed_off: 200ms

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Asked for help on the Hassio forum
Used a other members and i have the same problem

OttoWinter, 2019-06-01:
Duplicate of https://github.com/esphome/issues/issues/369

closed 2019-06-01 by OttoWinter

DelukaDZ, 2019-06-01:
Well its still not working and no answer on how to fix it

OttoWinter, 2019-06-01:
-> 1.13.3