esphome/issues #383: Remote Receiver doesnt find RF codes

Flagun, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32S

**Affected component:**
https://esphome.io/components/remote_receiver.html#remote-receiver-binary-sensor

**Description of problem:**
when I try to sniff the RF codes in the consol I get only this respond: 
[D][remote.sony:064]: Received Sony: data=0x00000000, nbits=0

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: rf_ble_gateway
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"
  
remote_receiver:
  pin: 
    number: GPIO32
    inverted: true
  dump: all

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

when changed to dump: - rc_switch nothing is showing in the console
when changed to raw, the data is showing

OttoWinter, 2019-06-01:
Could you share some of that raw data?

RCSwitch is only for certain 433MHz devices - it could be that yours is not supported through RCSwitch (though you can also use the raw codes directly).

Flagun, 2019-06-01:
I clicked one button once and that's the whole output:
[19:51:34][D][remote.raw:028]: Received Raw: 141, -943, 31, -421, 78, -1048, 381, -386, 38, -2639, 75, -885, 52, -895, 323, -27, 123, -2201, 42, -472, 201, -246, 95, -807, 423, -404, 57, -305, 221, -1177, 572, -159, 188, -1818, 84, -191, 38, -773, 278, -491, 454, -827, 278, -435, 221, 
[19:51:34][D][remote.raw:028]:   -648, 172, -341, 70, -774, 89, -1097, 391, -1254, 33, -1141, 170, -115, 276, -294, 163, -283, 89, -51, 258, -161, 141, -754, 131, -464, 201, -79, 37, -175, 133, -1554, 145, -150, 246, -378, 166, -507, 150, -823, 527, -404, 392, -246, 70, -642, 307, 
[19:51:34][D][remote.raw:041]:   -171, 61, -789, 402, -654, 96, -83, 268, -334, 188, -209, 176, -1462, 376, -270, 32, -911, 316, -157, 128, -64, 112, -1211, 167, -314, 80, -240, 75, -554, 477, -328, 64, -660, 140, -695, 172
[19:51:34][D][remote.raw:028]: Received Raw: -206, 5986, -829, 439, -370, 2079, -552, 354, -1195, 171, -331, 211, -604, 273, -1483, 447, -5531, 145, -734, 579, -3252, 267, -5386, 198, -1364, 160, -438, 124, -2185, 87, -1688, 116, -5022, 198, -4897, 60, -1132, 142, -2452, 177, -2978, 
[19:51:34][D][remote.raw:041]:   81, -4747, 237

2nd click (the same button) 
[19:54:10][D][remote.raw:028]: Received Raw: 44, -574, 60, -6728, 69, -499, 64, -141, 330, -2267, 226, -3449, 90, -612, 95, -1537, 68, -295, 88, -233, 125, -162, 132, -145, 100, -427, 148, -114, 143, -419, 106, -853, 153, -1348, 395, -407, 41, -173, 275, -78, 112, -270, 391, -245, 89, 
[19:54:10][D][remote.raw:028]:   -425, 36, -214, 162, -662, 110, -519, 191, -2543, 292, -66, 33, -664, 79, -625, 582, -660, 263, -265, 60, -785, 75, -303, 123, -460, 222, -206, 279, -370, 57, -1956, 106, -1231, 537, -59, 54, -96, 165, -795, 102, -153, 164, -158, 66, -2691, 209, -422, 
[19:54:10][D][remote.raw:041]:   172, -356, 67, -986, 252, -196, 195, -1784, 290, -224, 164, -168, 208, -600, 147, -799, 121, -1046, 58, -825, 141, -1011, 227, -148, 121, -1027, 114, -497, 103, -1288, 113, -1340, 194, -266, 73

pawel334, 2019-06-03:
Hello,

I've got a similar problem with RC5 codes (ESPHome v 1.13.4). Every button is recognized as Sony: data=0x00000000, nbits=0.

Example raw data for rc5 remote:
Received Raw: 907, -881, 891, -877, 1771, -879, 888, -877, 888, -859, 907, -880, 891, -877, 890, -875, 890, -1739, 908, -881, 1769, -880, 882
Received Raw: 916, -876, 886, -881, 1769, -881, 885, -881, 881, -865, 929, -857, 884, -884, 884, -862, 911, -1738, 931, -858, 1772, -878, 890

closed 2019-06-06 by OttoWinter