esphome/issues #386: on_boot command seems delayed since move from 1.12 to 1.13.3

markcarline, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 command line and pip install
**ESP (ESP32/ESP8266, Board/Sonoff):**

LoHas B22 RGB Bulbs
**Affected component:**

Function of the on_boot command

**Description of problem:**

Prior to 1.13 and while i was on 1.12 i was able to switch on the bulb and it would turn on immediately. Now since the update to 1.13.3 i have to wait around 8 seconds.

Not a big issue but we got used to lights switching on as soon as power was applied.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ledbulb01
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
#  esp8266_restore_from_flash: yes
  on_boot:
    priority: 75
    then:
      - light.turn_on:
          id: ledbulb01
          brightness: 100%
          red: 100%
          green: 57%
          blue: 14%
          white: 50%

wifi:
  ssid: 'SSID'
  password: 'PW'
  manual_ip:
    static_ip: 192.168.1.141
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

# Example configuration entry
mqtt:
  broker: 192.168.1.101
  username: user
  password: password

# Enable logging
logger:

# Enable Home Assistant API
#api:

ota:

my9231:
 data_pin: GPIO13
 clock_pin: GPIO15
 num_channels: 6
 num_chips: 2
# update_on_boot: False

output:
  - platform: my9231
    id: output_warm_white
    channel: 0
  - platform: my9231
    id: output_blue
    channel: 1
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_red
    channel: 3
  - platform: my9231
    id: output_cold_white
    channel: 5

light:
  - platform: rgbww
    name: "ledbulb01"
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold_white
    warm_white: output_warm_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2800 K
    id: "ledbulb01"
    effects:
      - strobe:
          name: My Warm White
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 57%
              blue: 14%
              white: 50%
              duration: 2s
            - state: True
              brightness: 100%
              red: 100%
              green: 57%
              blue: 14%
              white: 50%
              duration: 2s
      - random:
          name: "My Slow Random Effect"
          transition_length: 30s
          update_interval: 30s

      - random:
          name: "My Fast Random Effect"
          transition_length: 4s
          update_interval: 5s
      - random:
          name: Random Effect With Custom Values
          transition_length: 5s
          update_interval: 7s
      - strobe:
          name: Strobe Effect With Custom Values
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 90%
              blue: 0%
              duration: 500ms
            - state: False
              duration: 250ms
            - state: True
              brightness: 100%
              red: 0%
              green: 100%
              blue: 0%
              duration: 500ms
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 1.5%
      - lambda:
          name: My Custom Effect
          update_interval: 1s
          lambda: |-
            static int state = 0;
            auto call = id(ledbulb01).turn_on();
            // Transtion of 1000ms = 1s
            call.set_transition_length(1000);
            if (state == 0) {
              call.set_rgb(1.0, 1.0, 1.0);
            } else if (state == 1) {
              call.set_rgb(1.0, 0.0, 1.0);
            } else if (state == 2) {
              call.set_rgb(0.0, 0.0, 1.0);
            } else {
              call.set_rgb(1.0, 0.0, 0.0);
            }
            call.perform();
            state += 1;
            if (state == 4)
              state = 0;

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

fabianschaefer, 2019-06-01:
Same issue here, it seems that the priority value is completely ignored. It takes 7.5s to turn the light no matter if the priority is set to -10 or 95.

fabianschaefer, 2019-06-01:
I looked through the commits and I think the issue started after commit [https://github.com/esphome/esphome/commit/766f6c045d56e929b3592df52e6166d22b68da3d#diff-b51d2d72fe3e37a65031db8a703fb3f2](url) where you added many stuff to load with priority 100 -> @coroutine_with_priority(100.0)
Just a thought, I'm not a python developer, but it might help to solve the issue.

OttoWinter, 2019-06-02:
@fabianschaefer No, those are different priorities - those are priorities for code generation; not component setup priorities

It is true that setup priority have changed in this latest release: https://github.com/esphome/esphome/blob/dev/esphome/core/component.cpp#L15-L23

I will make the changed priorities more clear in the docs. And also that they may change at any time without marking it as a breaking change.

closed 2019-06-02 by OttoWinter