esphome/issues #389: How to use "repeat" variable in remote transmitter actions 1.13.x

Reaper90, 2019-06-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP8266

**Affected component:**

remote_transmitter
https://esphome.io/components/remote_transmitter.html

**Description of problem:**
After upgrading to the newest version, i can't get my remotecodes for remote_transmitter to work. They did work in 1.12.x using the following configuration:

```yaml
remote_transmitter:
  pin: D1
  carrier_duty_percent: 100

switch:
  - platform: remote_transmitter
    id: switch_a_on
    rc_switch_raw:
      code: '000001000001010001010001'
      protocol: 1
    repeat: 10
  - platform: remote_transmitter
    id: switch_a_off
    rc_switch_raw:
      code: '000001000001010001010100'
      protocol: 1
    repeat: 10
  - platform: template
    name: Switch A
    optimistic: True
    assumed_state: True
    turn_on_action:
      - switch.turn_on: switch_a_on
    turn_off_action:
      - switch.turn_on: switch_a_off
```

i followed the instructions from validation output and changed my switch donfiguration code to this:

```yaml
switch:
  - platform: template
    name: Switch A
    optimistic: True
    assumed_state: True
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '000001000001010001010001'
        protocol: 1
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000001000001010001010100'
          protocol: 1
```

Now the validation output say that the config is valid, but when i'm trying to use it in Hass.io, it doesn't work and i think it's because the "repeat:" configuration variable is missing. But when i ty to add the repeat variable, it doesn't validate anymore. I already read the new documentation about remote_transmitter, but it's not quite clear to me where i have to use the "repeat:" variable to get it working. This is the code i tried:

```yaml
switch:
  - platform: template
    name: Switch C
    optimistic: True
    assumed_state: True
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '000001000001010001010001'
        protocol: 1
        repeat: 10
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000001000001010001010100'
          protocol: 1
          repeat: 10
```

Any solution for this in the new version?
Thanks in advance and best regards

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-02:
The proper syntax would be:

```yaml
switch:
  - platform: template
    name: Switch C
    optimistic: True
    assumed_state: True
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '000001000001010001010001'
        protocol: 1
        repeat:
 		  times: 10
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000001000001010001010100'
          protocol: 1
          repeat: 
			times: 10
```

Reaper90, 2019-06-02:
Thank you, works perfectly fine again. i think i tried this solution too, but it didn't work back then (maybe wrong spacing).

closed 2019-06-02 by Reaper90