esphome/issues #392: HA and Hass.io Addon unable to connect to .local host over API

soberstadt, 2019-06-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 165, HassOS 2.12

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, D1 Mini

**Affected component:**

https://esphome.io/components/api.html

**Description of problem:**
I tried to update a device (named `mqtt_arrow_5`) today to 1.13.3 and change from MQTT to use API. I flashed the new bin, but HomeAssistant and the Addon are not able to connect to `mqtt_arrow_5.local` but my local `esphome mqtt_arrow_5.yaml logs` can.

Locally from MacOS:
```
$ esphome mqtt_arrow_5.yaml logs 
INFO Reading configuration...
INFO Starting log output from mqtt_arrow_5.local using esphome API
INFO Connecting to mqtt_arrow_5.local:6053 (192.168.0.100)
INFO Successfully connected to mqtt_arrow_5.local
[23:28:36][I][app:063]: esphome version 1.13.3 compiled on Jun  2 2019, 23:20:15
```

On ESPHome Addon:
[image]

In Home Assistant:
[image]

When I `ping mqtt_arrow_5.local:6053` it resolves to `23.202.231.166`, but `ping mqtt_arrow_5.local` resolves to `192.168.0.100`. So I tried putting `192.168.0.100` into the Home Assistant integration registration it connected successfully.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: mqtt_arrow_5
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret mqtt_wifi_ssid
  password: !secret mqtt_wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret mqtt_password
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-03:
The reason it works on your MacOS device is because MacOS has a built-in mDNS resolver.

The reason it is not working on your other systems with that error is that your DNS setup is apparently wrong - some DNS resolvers resolve `.local` addresses even though they should not.

The only way to work around this is by using static IP addresses, as the error is in the DNS server setup.

closed 2019-06-03 by OttoWinter

soberstadt, 2019-06-03:
Ah, so maybe this is a problem with HassOS or Hass.io? Do you know if there a way to change the dns server for HassOS (I don't fully understand the difference between HassOS and Hass.io)?

on my mac:
```
$ nslookup mqtt_arrow_5.local             
Server:		192.168.0.1
Address:	192.168.0.1#53

Non-authoritative answer:
Name:	mqtt_arrow_5.local
Address: 23.202.231.166
Name:	mqtt_arrow_5.local
Address: 23.195.69.106
```

ssh'ed into Hass.io:
```
core-ssh:~# nslookup mqtt_arrow_5.local
nslookup: can't resolve '(null)': Name does not resolve

Name:      mqtt_arrow_5.local
Address 1: 23.202.231.166 a23-202-231-166.deploy.static.akamaitechnologies.com
Address 2: 23.195.69.106 a23-195-69-106.deploy.static.akamaitechnologies.com
```

soberstadt, 2019-06-03:
I see you have submitted a patch to HA that should help with this. One reason I think this is a config or software issue is that I had previously been able to add a esphome node to my HA instance. In-fact, the auto-detect worked.

OttoWinter, 2019-06-03:
@soberstadt I've seen this error before with the exact same issue: The DNS resolver resolves .local for some reason (as seen in your nslookup output).

soberstadt, 2019-06-03:
Ok, is there additional DNS configuration that would help on router or hassOS? If so some documentation might be helpful. Even if it's a warning.

PS, thank you so much for how quickly you respond to issues. You have given so much to the HA community with ESPHome!

OttoWinter, 2019-06-03:
> Ok, is there additional DNS configuration that would help on router or hassOS?

Don't know, anyway, in next version it will be "fixed".