esphome/issues #394: api.connected returns with only logging OTA active

glmnet, 2019-06-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

`api:`

**Description of problem:**
It seems that `api.connected:` https://esphome.io/components/api.html#api-connected-condition would return true when no Home Assistant is connected to the api, but there is a logger OTA connected to the node. I believe expected behavior is only to report true for clients monitoring the node like HA, otherwise may be a clarification statement on the docs that this will return true if an OTA logging is connected.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
https://community.home-assistant.io/t/make-esphome-node-fallback-when-not-connected-to-ha-api/116615/14?u=glmnet

OttoWinter, 2019-06-03:
The native API intentionally does not know what type of client is connected - like with MQTT. This is to allow for 3rd party clients/libs that interact with native API as HA would.

I guess the docs can be updated to make this more clear.

closed 2019-06-03 by OttoWinter

adarazs, 2019-06-04:
Thanks for the doc update, this got me while testing that condition and it's nice to have it mentioned in the docs! Also thanks for @glmnet for reporting it. :)