esphome/issues #398: Timezone question

jeff-h, 2019-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker, with docker-compose, running in Ubuntu 19.04 host. The relevant docker-compose fragment:

```
  esphome:
    image: esphome/esphome
    container_name: esphome
    volumes:
      - ${DOCKERDIR}/containers/esphome/config:/config:rw
    network_mode: host
    restart: unless-stopped
    environment:
      - TZ=Pacific/Auckland
```

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266

**Description of problem:**

When I click "Show Logs", the logs are all 12 hours out from my timezone (eg instead of 21:43 it shows 09:43). Being in GMT+12 here it's hard to say whether it's swapping AM for PM, or whether it's showing in GMT / UTC.

You'll see I attempted to fix this by adding the `TZ` environment in docker-compose, but that hasn't helped.

Would you say this is a problem with my docker machine config (I do have other containers with the correct TZ) or do I need to tell ESPHome my TZ some other way?

The problem raised its head for me because I have my sensor data being collected in Home Assistant, where the historical data graph is showing the correct (ie local NZ) time, but I've just tried to import that data from Home Assistant into Node-RED which is showing the same (wrong) timestamps as ESPHome has.

OttoWinter, 2019-06-04:
ESPHome uses the value reported by the OS for logging - depends on how the python process uses the data, but very likely it's localtime().

So yes this is a general docker setup problem - usually the TZ variable does not influence containers unless they explicitly access that variable. The solution usually is to pass in /etc/localtime into the container (google)

jeff-h, 2019-06-04:
Wow, that's a super fast response! Thanks so much for your work on this, I love it :)

I just added `- /etc/timezone:/etc/timezone:ro` to my volumes without success, but you've clarified this isn't an ESPHome issue, so I'll be quiet for now. Will post the solution here anyway once I find it, in case someone else has the same issue at some point.

brandond, 2019-06-04:
You want `/etc/localtime:/etc/localtime:ro` not /etc/timezone. It may also help to set the `TZ=:/etc/localtime` environment variable.

jeff-h, 2019-06-04:
@brandond thanks for the heads-up! You are right, it needs `localtime` not `timezone`. However, that alone didn't fix it. My first attempt was to have all three (`/etc/timezone`, `/etc/localtime` and a `TZ` environment variable) but no dice. Seems it has to be **only** `/etc/localtime` to work.

In my admittedly-extensive ignorance, it looks like this requirement actually is somewhat specific to the esphome docker image, in the sense that apparently ["Some applications use the one, others use the other"](https://unix.stackexchange.com/q/384971).

@OttoWinter would you like it added to the documentation?

OttoWinter, 2019-06-06:
> "Some applications use the one, others use the other".

While technically true, the more or less agreed upon standard *is* /etc/localtime. Why? The most common C standard library implementation (glibc) uses /etc/localtime for calculating the current time.

`/etc/timezone` seems to be a separate thing the Java developers introduced a long time after that.

`TZ` seems to be also used in some places - but glibc (the thing almost all programs use, including python) appears to prefer /etc/localtime over that.

brandond, 2019-06-06:
There's a reason that you should set TZ in addition to using /etc/localtime. tl;dr it shortcuts a bunch of glibc internal logic and saves a bunch of extra syscalls:

https://blog.packagecloud.io/eng/2017/02/21/set-environment-variable-save-thousands-of-system-calls/

OttoWinter, 2019-06-06:
@brandond Good to know! Thanks! I won't add it to the docker-compose example though because it's really not necessary for ESPHome, I don't think there's any big performance impact since ESPHome does not use that function anywhere except for logging. And logging is a) nowhere near fast enough and b) uses networking/serial ports which mean syscalls anyway. So it would unnecessarily complicate the example.

jeff-h, 2019-06-06:
Thanks all for the help and knowledge. I think this can be closed.

closed 2019-06-06 by jeff-h