esphome/issues #399: Docs and validation on template switch

glmnet, 2019-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

Template switch

**Description of problem:**
One detail: on the optimistic on switch the word cover is there and should be switch instead 

Now to the validation. I see often people confuses how template switch works. They do not implement turn_on/off_action nor optimistic. So switch does nothing. 
Is there a valid use case in that scenario or should the library warn the user. I believe they usually expect optimistic to be default true, without even knowing it. 

Also in docs there should be a note to not confuse turn_on_action with on_turn_on. IMHO 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

glmnet, 2019-06-04:
It will be cool if optimistic is automatically set to try if no action is set

OttoWinter, 2019-06-04:
> It will be cool if optimistic is automatically set to true if no action is set

Isn't this issue about not allowing action?

> I believe they usually expect optimistic to be default true, without even knowing it.

I guess so - reason it is this way is because originally template switch required the user to supply a lambda for the state; now it is of course also possible without.

Non-optimistic really only makes sense when the user supplies the system with actual data - via lambda or switch.template.publish
Making optimistic default to True might be a good idea, I will think about it.

I don't think I know a use case without turn_on/off action - so I guess we can make it so that at least one of them needs to be specified.

> Also in docs there should be a note to not confuse turn_on_action with on_turn_on. IMHO

While true, I've not seen a user ask about this before.

glmnet, 2019-06-04:
> I don't think I know a use case without turn_on/off action - so I guess we can make it so that at least one of them needs to be specified.

Actually this makes sense (I'm already using this as is):
````yaml
switch:
  platform: template
  name: My switch
  optimistic: true
````

I even called it a use pattern: how otherwise would you have a switch in ESP for some feature, e.g. toggling an ESPHome automation on/off,  from Home Assistant? If you do the turn_on/off_action you'll need a variable to set and later check the variable, it is easier if you just check the switch state directly.

OttoWinter, 2019-06-04:
Ahh. true.

closed 2020-12-05 by stale[bot]