esphome/issues #400: Momentary Switch does not return status in frontend

danye72, 2019-06-04:
The configuration for the Switch Momentary Switch does not return the status to the hassio frontend. However the relay is activated and deactivated correctly.
If I use optimistic: true the frontend shows a normal and non-timed switch.
I am using this configuration, while the version of esphome in use is 1.13.4

switch:
  - platform: gpio
    pin: GPIO15
    id: relay5
  - platform: template
    name: "Test"
    icon: "mdi:gate"
    turn_on_action:
      - switch.turn_on: relay5
      - delay: 1s
      - switch.turn_off: relay5

closed 2019-06-04 by close-issue-app[bot]

jhiller227, 2019-06-06:
I have the same issue.

jhiller227, 2019-06-06:
see this case.. it solved it. Template Switch does not return status in frontend #401