esphome/issues #401: Momentary Switch does not return status in frontend

danye72, 2019-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio.io in docker x86
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff basic v2,
**Affected component:**

GPIO Switch

**Description of problem:**

The configuration for the Switch Momentary Switch does not return the status to the hassio frontend. However the relay is activated and deactivated correctly.
If I use optimistic: true the frontend shows a normal and non-timed switch.
I am using this configuration, while the version of esphome in use is 1.13.4

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: giardino
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "WDANNI"
  password: "xxxxxxxxx"
  manual_ip:
    static_ip: 192.168.2.204
    gateway: 192.168.2.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

light:
  - platform: binary
    name: "Luce cantina"
    output: gpio_luce_cantina
    id: luce_cantina
  - platform: binary
    name: "Luci_giardino_1"
    output: gpio_luci_giardino_1
  - platform: binary
    name: "Luci_giardino_2"
    output: gpio_luci_giardino_2
    
output:
  - platform: gpio
    pin: GPIO12
    id: gpio_luce_cantina
  - platform: gpio
    pin: GPIO5
    id: gpio_luci_giardino_1
  - platform: gpio
    pin: GPIO4
    id: gpio_luci_giardino_2
    
switch:
  - platform: gpio
    pin: GPIO15
    id: relay5
  - platform: template
    name: "Test"
    icon: "mdi:gate"
    optimistic: true
    turn_on_action:
      - switch.turn_on: relay5
      - delay: 1s
      - switch.turn_off: relay5

    
binary_sensor:
  - platform: status
    name: "Giardino"
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Giardino Button 1"
    on_state:
      - light.toggle: luce_cantina
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    name: "Giardino Button 2"
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    name: "Giardino Button 3"
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "Giardino Button 4"

sensor:    
  - platform: wifi_signal
    name: "WiFi giardino"
    update_interval: 60s
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-06-04:
Your switch has no `value_template`. Without `optimistic: true` it has no way of knowing what the current value is to send back to Home Assistant. Optimistic tells it to assume on if turned on, or off if turned off. Providing a value_template allows you to provide code to retrieve the current states.

https://www.home-assistant.io/components/switch.template/#value_template

danye72, 2019-06-04:
I'm sorry I followed the example of the momentary switch and it doesn't work ...
```yaml
# Example configuration entry
switch:
  - platform: gpio
    pin: 25
    id: relay
  - platform: template
    name: "Gate Remote"
    icon: "mdi:gate"
    turn_on_action:
    - switch.turn_on: relay
    - delay: 500ms
    - switch.turn_off: relay
```
What is the system for having a timed switch, which returns the status on the frontend?

brandond, 2019-06-04:
That creates a button that, when turned on, toggles a relay on and off again. Currently it has no state - it neither assumes it's on/off, nor reads from something else to derive a value. You can find this used in several examples, such as: https://www.home-assistant.io/components/switch.template/#sensor-and-two-switches

I think what you want is something like this:
```yaml
switch:
  - platform: gpio
    pin: 25
    id: relay
  - platform: template
    name: "Gate Remote"
    icon: "mdi:gate"
    value_template: "{{ is_state('switch.relay', 'on') }}"
    turn_on_action:
    - switch.turn_on: relay
    - delay: 500ms
    - switch.turn_off: relay
```

danye72, 2019-06-04:
Unfortunately it doesn't work, I have this error:

[value_template] is an invalid option for [switch.template]. Please check the indentation.

OttoWinter, 2019-06-04:
@danye72 The behavior is correct - template switch does not do anything unless explicitly instructed to do so.

The example also purposefully does not add this behavior because it just adds complexity to a simple example. And most users don't need the reporting.

This would work:

```yaml
switch:
  - platform: gpio
    pin: GPIO15
    id: relay5
  - platform: template
    name: "Test"
    icon: "mdi:gate"
    lambda: 'return id(relay5).state;'
    turn_on_action:
      - switch.turn_on: relay5
      - delay: 1s
      - switch.turn_off: relay5
```

danye72, 2019-06-04:
@OttoWinter
Perfect!!! Your solution works well for me!
I will study the lambda option better.
Thank you.

OttoWinter, 2019-06-06:
Closing this issue because working as intended.

closed 2019-06-06 by OttoWinter