esphome/issues #402: BME280 is overheating and then shows inaccurate values

Shwamp, 2019-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-04:
> The oversampling should be changed in the ESPHOME example code to "oversampling: 1x".

You're free to change the docs, just press the "Edit this page on github" button.

> The datasheet also recommends the use of "forced mode"
> Without the ability to set the sensor to the forced power mode it will continue to show wrong values.

That is already used.

> The time used between the sensor should be either configurable or set to 60 seconds according to the datasheet.

-> `update_interval`; even defaults to 60s

Shwamp, 2019-06-04:
Great!! 😃 Will edit the docs!
What do you think about changing the default oversampling to x1 under the _"Oversampling Options"_? 
The current default "x16" will cause issues according to the data sheet.

OttoWinter, 2019-06-06:
> What do you think about changing the default oversampling to x1 under the "Oversampling Options"?

That would be a breaking change, and I don't really like those unless it's really required.

Also, while the x16 might warm up the sensor, I do think it will be a pretty constant amount it is warmed up with, so a simple offset should account for that pretty easily. And x16 oversampling *does* give you more accuracy too.

OttoWinter, 2019-07-11:
Closing due to inactivity - and the software is correct, the docs could just be updated.

closed 2019-07-11 by OttoWinter