esphome/issues #404: SSL Certificate issue on Hass.io addon after update to 1.13.4

mikenabhan, 2019-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on Ubuntu 18.04
**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp8266 NodeMCU
**Affected component:**

Home Assistant Addon

**Description of problem:**

I am running hass.io on Ubuntu 18.04 running the official ESPHome addon.

After my most recent update when I go back to the "network" section of the addon configuration and add the port back on to re-enable direct access via https://hass.mydomain.com:6052, the internal webserver cannot find the certificate.

This functionality worked properly prior to this update, so I assume that it is related to the introduction of ingress.

Nothing about my configuration has changed, simply the update disable direct access via port and re-enabling it as I would other addons does not work.

Below are the logs from my esphome addon as it starts:

`-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.13.4
 You are running the latest version of this add-on.
 System: Ubuntu 18.04.2 LTS  (amd64 / qemux86-64)
 Home Assistant version: 0.93.2
 Supervisor version: 165
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[20:45:37] INFO: Waiting for dashboard to come up...
[20:45:37] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[20:45:37] INFO: Starting NGINX...
2019/06/04 20:45:40 [error] 484#484: *1 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *2 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *3 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *4 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *5 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *6 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *7 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *8 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *9 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052
2019/06/04 20:45:40 [error] 484#484: *10 no "ssl_certificate" is defined in server listening on SSL port while SSL handshaking, client: 192.168.0.103, server: 0.0.0.0:6052`

SSL certificates are present in /ssl/ in the hass directory and here is my addon config:

`{
  "ssl": true,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}`
**Problem-relevant YAML-configuration entries:**
```yaml
Not relevent

```

**Traceback (if applicable):**

```
Not relevent
```

**Additional information and things you've tried:**

cherokee93, 2019-06-05:
I know it's not a solution, but why not simply use Hassio Ingress + Add to sidebar?

mikenabhan, 2019-06-05:
Well, that also isn't working properly for me, the black frames on the compile/console don't render anything that way, but that's a different issue and I am still trying to document it/trace it.  

There seems to be a lot of issues with ingress, I don't really feel that it is production ready.  

Attached is what I mean by that not working properly.  This is what happens if i try to view logs or compile/upload.

[image]

What is frustrating about that whole thing is that the process is still happening in the background, it just won't render which makes me believe its an ingress problem and not a background problem with ESPHome itself.

closed 2019-06-06 by OttoWinter