esphome/issues #405: TTGO T-Camera minimal config, still boot loops

keith721, 2019-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Intel Core i3 NUC, Debian Server, Docker, Hass.IO 
Linux hassio 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64
Docker version 18.09.1, build 4c52b90
Home Assistant 0.93.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, TTGO T-Camera
https://www.amazon.com/gp/product/B07RY29WM6/
TTGO Micro-32 V2.0 Wifi wireless Bluetooth Module ESP32 PICO-D4 IPEX

**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**
Upon download of firmware.bin to ESP32, board repeatedly fails to boot, falling back to safe boot mode.  At this time, it successfully inits wifi and waits for OTA download.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ttgo_cam01
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "MyHomeWifi"
  password: "123456789"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '123456789'

ota:
  password: '123456789'
  
esp32_camera:
  name: 'ttgocam01'
  external_clock:
    pin: 4
    frequency: 20MHz
  i2c_pins:
    sda: 18
    scl: 23
  data_pins: [34, 13, 14, 35, 39, 12, 15, 36]
  vsync_pin: 5
  href_pin: 27
  pixel_clock_pin: 25

```

**Traceback (if applicable):**

```
Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 49.63 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[07:28:47][I][logger:116]: Log initialized
[07:28:47][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[07:28:50][I][app:02[I][logger:116]: Log initialized
[07:28:53][C][ota:364]: There hav[I][logger:116]: Log initialized
[07:28:55][C][ota:364]: There[I][logger:116]: Log initialized
[07:28:58][C][ota:364]: There have been 3 suspected unsuc[I][logger:116]: Log initialized
[07:29:01][C][ota:364]: There h[I][logger:116]: Log initialized
[07:29:03][C][ota:364]: There have been 5 sus[I][logger:116]: Log initialized
[07:29:06][C][ota:364]: There have bee[I][logger:116]: Log initialized
[07:29:09][C][ota:364]: There have been 7 suspected unsu[I][logger:116]: Log initialized
[07:29:11][C][ota:364]: There have been 8 [I][logger:116]: Log initialized
[07:29:14][C][ota:364]: There [I][logger:116]: Log initialized
[07:29:14][C][ota:364]: There have been 10 suspected unsuccessful boot attempts.
[07:29:14][E][ota:369]: Boot loop detected. Proceeding to safe mode.
[07:29:14][C][wifi:029]: Setting up WiFi...
[07:29:14][D][wifi:268]: Starting scan...

```

**Additional information and things you've tried:**
Disabled I2C, disabled I2C display, minimized the YAML to only esp32_camera configuration.  Double-checked pin assignments.
Tried multiple 5VDC power supplies.
Failure occurs with two separate TTGO T-Camera boards.

Lefuneste83, 2019-06-05:
Hi there. I have the same issue with new TTGO Cameras boards.

It is occuring with the TTGO Camera With Microphone. The GPIO pinout is completly different from the regular TTGO Cam (without microphone).

I confirm the configuration to be as follow :

```
binary_sensor:
  - platform: gpio
    pin: GPIO19
    name: "ttgocam1 PIR"
    device_class: motion

i2c:
  sda: GPIO21
  scl: GPIO22

esp32_camera:
  name: "ttgocam1 Camera"
  external_clock:
    pin: GPIO4
  i2c_pins:
    sda: GPIO18
    scl: GPIO23
  data_pins: [GPIO34, GPIO13, GPIO14, GPIO35, GPIO39, GPIO12, GPIO15, GPIO36]
  vsync_pin: GPIO5
  href_pin: GPIO27
  pixel_clock_pin: GPIO25
  resolution: 640x480
  jpeg_quality: 10
  vertical_flip: false
  horizontal_mirror: false
```

The PIR and I2C parts are working fine. The GPIO for the PIR is wrongly adverstised as 33 by the sellers while it is 19 indeed. As soon as the camera definition is added the board boot loops.

Also on compilation I have the following warning message :

```
src/esphome/components/esp32_camera/esp32_camera.cpp: In member function 'virtual void esphome::esp32_camera::ESP32Camera::dump_config()':
src/esphome/components/esp32_camera/esp32_camera.cpp:54:10: warning: enumeration value 'FRAMESIZE_QXGA' not handled in switch [-Wswitch]
switch (this->config_.frame_size) {
^
src/esphome/components/esp32_camera/esp32_camera.cpp:54:10: warning: enumeration value 'FRAMESIZE_INVALID' not handled in switch [-Wswitch]
```

It seems that the driver has an issue with the board as I have checked 10 times the GPIO pinout and it seems to be correct.

Lefuneste83, 2019-06-05:
I got it working with the following conf except that I get constant disconnections from the Esphome API. As the esp32_camera relies upon the esphome api it is pretty unusable so far :

esphome:
  name: ttgo_cam_test
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "SSID"
  password: "PWD"
  fast_connect: on
  manual_ip:
    static_ip: 192.168.0.xx
    gateway: 192.168.0.xx
    subnet: 255.255.255.0
    dns1: 192.168.0.xx
    dns2: 8.8.8.8

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "PWD"

ota:
  password: "PWD"

binary_sensor:
  - platform: gpio
    pin: GPIO19
    name: "ttgocam1 PIR"
    device_class: motion

sensor:
  - platform: wifi_signal
    name: "ttgocam1 WiFi Signal"
    update_interval: 10s

time:
  - platform: homeassistant
    id: homeassistant_time

i2c:
  sda: GPIO21
  scl: GPIO22

font:
  - file: "fonts/times-new-roman.ttf"
    id: tnr1
    size: 20
  - file: "fonts/times-new-roman.ttf"
    id: tnr2
    size: 35

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.strftime(64, 0, id(tnr1), TextAlign::TOP_CENTER,"%d-%m-%Y", id(homeassistant_time).now());
      it.strftime(64, 64, id(tnr2), TextAlign::BASELINE_CENTER, "%H:%M", id(homeassistant_time).now());

esp32_camera:
  name: "ttgocam1 Camera"
  external_clock:
    pin: GPIO4
    frequency: 20MHz
  i2c_pins:
    sda: GPIO18
    scl: GPIO23
  data_pins: [GPIO34, GPIO13, GPIO14, GPIO35, GPIO39, GPIO12, GPIO15, GPIO36]
  vsync_pin: GPIO5
  href_pin: GPIO27
  pixel_clock_pin: GPIO25

keith721, 2019-06-05:
Confirming, code boots successfully and executes as expected so long as esp32_camera is not defined in the YAML. :(
'''
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 49.98 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[18:33:36][I][logger:116]: Log initialized
[18:33:36][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[18:33:36][I][app:027]: Running through setup()...
[18:33:36][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:33:36][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[18:33:36][C][wifi:029]: Setting up WiFi...
[18:33:36][D][wifi:268]: Starting scan...
[18:33:39][D][wifi:283]: Found networks:
[18:33:39][I][wifi:318]: - 'MyHomeWifi' [AA:BB:CC:DD:EE:FF]▂▄▆█
[18:33:39][D][wifi:319]:     Channel: 6
[18:33:39][D][wifi:320]:     RSSI: -10 dB
[18:33:39][D][wifi:322]: - 
[18:33:39][I][wifi:164]: WiFi Connecting to 'MyHomeWifi'...
[18:33:40][I][wifi:376]: WiFi connected!
[18:33:40][C][wifi:250]:   SSID: MyHomeWifi
[18:33:40][C][wifi:251]:   IP Address: 192.168.1.114
[18:33:40][C][wifi:253]:   BSSID: AA:BB:CC:DD:EE:FF
[18:33:40][C][wifi:254]:   Hostname: 'ttgo_cam01'
[18:33:40][C][wifi:258]:   Signal strength: -12 dB ▂▄▆█
[18:33:40][C][wifi:259]:   Channel: 6
[18:33:40][C][wifi:260]:   Subnet: 255.255.255.0
[18:33:40][C][wifi:261]:   Gateway: 192.168.1.1
[18:33:40][C][wifi:262]:   DNS1: 8.8.8.8
[18:33:40][C][wifi:263]:   DNS2: 8.8.8.8
[18:33:40][C][ota:029]: Over-The-Air Updates:
[18:33:40][C][ota:030]:   Address: ttgo_cam01.local:3232
[18:33:40][C][ota:032]:   Using Password.
[18:33:40][C][api:029]: Setting up Home Assistant API server...
[18:33:40][I][app:059]: setup() finished successfully!
[18:33:40][I][app:095]: esphome version 1.13.4 compiled on Jun  5 2019, 18:32:26
[18:33:40][C][wifi:368]: WiFi:
[18:33:40][C][wifi:250]:   SSID: MyHomeWifi
[18:33:40][C][wifi:251]:   IP Address: 192.168.1.114
[18:33:40][C][wifi:253]:   BSSID: AA:BB:CC:DD:EE:FF
[18:33:40][C][wifi:254]:   Hostname: 'ttgo_cam01'
[18:33:40][C][wifi:258]:   Signal strength: -12 dB ▂▄▆█
[18:33:40][C][wifi:259]:   Channel: 6
[18:33:40][C][wifi:260]:   Subnet: 255.255.255.0
[18:33:40][C][wifi:261]:   Gateway: 192.168.1.1
[18:33:40][C][wifi:262]:   DNS1: 8.8.8.8
[18:33:40][C][wifi:263]:   DNS2: 8.8.8.8
[18:33:40][C][i2c:028]: I2C Bus:
[18:33:40][C][i2c:029]:   SDA Pin: GPIO21
[18:33:40][C][i2c:030]:   SCL Pin: GPIO22
[18:33:40][C][i2c:031]:   Frequency: 50000 Hz
[18:33:40][I][i2c:033]: Scanning i2c bus for active devices...
[18:33:40][I][i2c:040]: Found i2c device at address 0x3C
[18:33:40][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'ttgo_cam01 PIR'
[18:33:40][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[18:33:40][C][gpio.binary_sensor:016]:   Pin: GPIO19 (Mode: INPUT)
[18:33:40][C][logger:137]: Logger:
[18:33:40][C][logger:138]:   Level: DEBUG
[18:33:40][C][logger:139]:   Log Baud Rate: 115200
[18:33:40][C][logger:140]:   Hardware UART: UART0
[18:33:40][C][homeassistant.time:010]: Home Assistant Time:
[18:33:40][C][homeassistant.time:011]:   Timezone: 'EST5EDT4,M3.2.0/2,M11.1.0/2'
[18:33:40][C][ssd1306_i2c:023]: I2C SSD1306
[18:33:40][C][ssd1306_i2c:023]:   Rotations: 0 °
[18:33:40][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[18:33:40][C][ssd1306_i2c:024]:   Address: 0x3C
[18:33:40][C][ssd1306_i2c:025]:   Model: SSD1306 128x64
[18:33:40][C][ssd1306_i2c:027]:   External VCC: NO
[18:33:40][C][ssd1306_i2c:028]:   Update Interval: 1.0s
[18:33:40][C][ota:029]: Over-The-Air Updates:
[18:33:40][C][ota:030]:   Address: ttgo_cam01.local:3232
[18:33:40][C][ota:032]:   Using Password.
[18:33:40][C][api:102]: API Server:
[18:33:40][C][api:103]:   Address: ttgo_cam01.local:6053
[18:34:19][D][api:572]: Client 'Home Assistant 0.93.2 (192.168.1.35)' connected successfully!
[18:34:19][D][time:030]: Synchronized time: Wed Jun  5 18:34:19 2019
'''

keith721, 2019-06-05:
Did not mean to close the issue.  Hope to work with others to resolve this clear pin assignment / definition issue with these boards.

OttoWinter, 2019-06-06:
Other boards work fine with ESP32 Camera - So to me it sounds like it is an issue with pin numbers indeed.

keith721, 2019-06-06:
Lewis Xhe also provided an updated electrical schematic for this 20190228 OV2640_V1.6 board:
https://github.com/Xinyuan-LilyGO/esp32-camera-series/blob/d24d3777f8f65d2cf97b908351487ebfed28dcd5/schematic/OV2640_V16.pdf

keith721, 2019-06-06:
So sorry to again accidentally close this research issue.  Re-opening until correct pin assignments are verified.

keith721, 2019-06-06:
FYI, the silk-screened PCB ID for this camera is OV2640_V1.6 20190228
The GPIO pin assignments silk-screened on the PCB beneath the camera agree with those shown in YAML configs above:
Y0=IO34
Y1=IO13
Y2=IO14
Y3=IO35
Y4=IO39
Y5=IO12
Y6=IO15
Y7=IO36

SIO_CLK=IO23
SIO_DAT=IO18
XCLK=IO4
VSYNC=IO5
HREF=IO27
PCLK=IO25

We know the photo/graphic on the board case is incorrect, specifically for the PIR sensor.  It appears all the pins for the LED display are correct. Could the silkscreen on the PCB be incorrect, as well?

keith721, 2019-06-06:
Began to wonder if the boards were defective, so decided to load Lewis Xhe's demonstration program tailored for this board. It works perfectly.  https://github.com/lewisxhe/esp32-camera-series

These are the pinout definitions from his INO sketch source file, which seem to agree with our assignments.  I suspect his comments regarding conflicting libraries and patching them may be the answer to this issue.  
```
/***************************************
 *  PinOUT
 **************************************/
#define PWDN_GPIO_NUM       -1
#define RESET_GPIO_NUM      -1
#define XCLK_GPIO_NUM       4
#define SIOD_GPIO_NUM       18
#define SIOC_GPIO_NUM       23

#define Y9_GPIO_NUM         36
#define Y8_GPIO_NUM         15
#define Y7_GPIO_NUM         12
#define Y6_GPIO_NUM         39
#define Y5_GPIO_NUM         35
#define Y4_GPIO_NUM         14
#define Y3_GPIO_NUM         13
#define Y2_GPIO_NUM         34

#define VSYNC_GPIO_NUM      5
#define HREF_GPIO_NUM       27
#define PCLK_GPIO_NUM       25
#define AS312_PIN           19
#define BUTTON_1            0
```

OttoWinter, 2019-06-07:
> I suspect his comments regarding conflicting libraries and patching them may be the answer to this issue.

Could you link to that?

keith721, 2019-06-07:
> 
> > I suspect his comments regarding conflicting libraries and patching them may be the answer to this issue.
> 
> Could you link to that?
Doh!  Certainly.
https://github.com/lewisxhe/esp32-camera-series

Lewis's comments seem to be more about the Adafruit BME280 library, which should not impact this board.  Although he does have comments regarding I2C, they're not specifically about conflicts.

OttoWinter, 2019-06-07:
> I checked the arduino-esp32 page, and they still have not yet added this specific board to their list. 

The board selection is not related to this issue - board only changes some basic things like memory size/pin aliases.

keith721, 2019-06-08:
Checking [this patch](https://github.com/geeksville/ezdevice-esp32/commit/35dd90840045cec940ab72a702dcc0730b5de241) from Kevin Hester and wondering if this might be the case for this board?

keith721, 2019-06-08:
Added the right push-button as a GPIO binary sensor, along with the PIR sensor.  They both initialize properly, then the boot loop begins.  This configuration finally halted.  It still appears  Core 1 is the culprit, if that helps any.  I was reading other threads and web pages today regarding ESP32 multi-core coding.  Rather a touchy thing, depending on how your code is assigned to one core or the other, and whether they attempt to share areas of code and/or data.
```
Using 'COM9' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 30:AE:A4:AA:BB:CC
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.7s
Compressed 15088 bytes to 9755...
Wrote 15088 bytes (9755 compressed) at 0x00001000 in 0.9 seconds (effective 135.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 523.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2100.4 kbit/s)...
Hash of data verified.
Compressed 948960 bytes to 513389...
Wrote 948960 bytes (513389 compressed) at 0x00010000 in 45.6 seconds (effective 166.3 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[18:51:06][I][logger:116]: Log initialized
[18:51:06][E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: 0 NOT_FOUND
[18:51:06][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[18:51:06][I][app:027]: Running through setup()...
[18:51:06][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:06][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:09][I][logger:116]: Log initialized
[18:51:09][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[18:51:09][I][app:027]: Running through setup()...
[18:51:09][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:09][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:12][I][logger:116]: Log initialized
[18:51:12][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[18:51:12][I][app:027]: Running through setup()...
[18:51:12][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:12][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:14][I][logger:116]: Log initialized
[18:51:14][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[18:51:14][I][app:027]: Running through setup()...
[18:51:14][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:14][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:17][I][logger:116]: Log initialized
[18:51:17][C][ota:364]: There have been 4 suspected unsuccessful boot attempts.
[18:51:17][I][app:027]: Running through setup()...
[18:51:17][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:17][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:20][I][logger:116]: Log initialized
[18:51:20][C][ota:364]: There have been 5 suspected unsuccessful boot attempts.
[18:51:20][I][app:027]: Running through setup()...
[18:51:20][D][binary_sensor:033]: 'ttgo_cam01 PIR': Sending state OFF
[18:51:20][D][binary_sensor:033]: 'ttgo_cam01 Button': Sending state ON
[18:51:20]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[18:51:20]Core 1 register dump:
[18:51:20]PC      : 0x4000bec0  PS      : 0x00060b30  A0      : 0x80140f5e  A1      : 0x3ffb1590  
[18:51:20]A2      : 0x3ffb9968  A3      : 0x00000000  A4      : 0x3ffb98ec  A5      : 0x00000000  
[18:51:20]A6      : 0x00000000  A7      : 0x00000012  A8      : 0x8008ea60  A9      : 0x3ffb1530  
[18:51:20]A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
```

keith721, 2019-06-10:
@OttoWinter I'm willing to use VSCode, PlatformIO, and anything else necessary to attempt debugging on WIndows 10 with this board.  My first attempt yesterday did not go well :(  but I'm very willing to try again.  Any hints, tips, or suggestions would be greatly appreciated.  Thank you!

Lefuneste83, 2019-06-10:
Hi Keith

I manage to get these working using 1.13.3 It is stil very weird. The board sometimes goes into OTA, sometimes sends a panic, sometimes bootloops. The weirdest thing is that it can get to work after 3 or 4 hard reset by unplugging replugging the USB. Once it is properly started I can see them in Home Assistant, get the PIR and the Camera. I have no idea why rebooting it a number of times changes something but I suppose it sends the board into a specific mode...

Anyhow, it is not likely a board defect, or at least not a single one, as I have 2 of these and they behalve similarly. 

I have tried to erase flash using esptool and also I have to unplug and replug them several times in order for the flash erase to succeed.

I must add that I have not been able to start them properly using 1.13.4 or 1.13.5 no matter how many times I reboot them.

My YAML is as follow : 

```
esphome:
  name: ttgo_cam_2
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "SSID"
  password: "PWD"
  fast_connect: on
  manual_ip:
    static_ip: 192.168.0.XX
    gateway: 192.168.0.XX
    subnet: 255.255.255.0
    dns1: 192.168.0.XX
    dns2: 8.8.8.8

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "PWD"

ota:
  password: "PWD"

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: gpio
    pin: GPIO19
    name: "ttgocam2 PIR"
    device_class: motion

  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "ttgocam2 Button"

sensor:
  - platform: wifi_signal
    name: "ttgocam2 WiFi Signal"
    update_interval: 60s

i2c:
  sda: GPIO21
  scl: GPIO22

font:
  - file: "fonts/times-new-roman.ttf"
    id: tnr1
    size: 20
  - file: "fonts/times-new-roman.ttf"
[... 183 lines cut ...]
[16:34:15][V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=50000Hz
[16:34:15][V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[16:34:15][D][binary_sensor:033]: 'ttgocam2 PIR': Sending state OFF
[16:34:15]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[16:34:15]Memory dump at 0x400f4df4: 00054614 91b08320 8890739c
[16:34:15]Core 1 register dump:
[16:34:15]PC      : 0x400f4dfa  PS      : 0x00060730  A0      : 0x8009475c  A1      : 0x3ffbc610  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d ttgo_cam_2 -t idedata
WARNING Decoded 0x400f4dfa: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c:86
[16:34:16]A2      : 0x00000006  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x80000020  
[16:34:16]A6      : 0x00000000  A7      : 0x3ffbd5f0  A8      : 0x3ffc361c  A9      : 0x3ffc35e8  
[16:34:16]A10     : 0x80000001  A11     : 0x80000001  A12     : 0x00000000  A13     : 0x00000001  
[16:34:16]A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
[16:34:16]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[16:34:16]
[16:34:16]Backtrace: 0x400f4dfa:0x3ffbc610 0x40094759:0x3ffbc630 0x40092871:0x3ffbc650
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400f4dfa: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c:86
WARNING Decoded 0x40094759: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c:3537
WARNING Decoded 0x40092871: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[16:34:16]
[16:34:16]Rebooting...
```
Hope this can help finding out the solution...

Lefuneste83, 2019-06-10:
This may be interesting : 

https://github.com/espressif/arduino-esp32/issues/2659

Lefuneste83, 2019-06-10:
I suspect the esp-cam code to cause an issue while trying to acces the PSRAM through I2C.

[esp32-hal-psram.c:47] psramInit(): PSRAM enabled

This line only appears when the camera is declared in the YAML and it should be adressed through  i2c_pins:
    sda: GPIO18
    scl: GPIO23

keith721, 2019-06-10:
I'm guessing it's this code in src/esphome/components/esp32_camera/esp32_camera.cpp that attempts to pin the camera task to Core 1, which is what's faulting.  Seriously, just a guess. 
```
  xTaskCreatePinnedToCore(&ESP32Camera::framebuffer_task,
                          "framebuffer_task",  // name
                          1024,                // stack size
                          nullptr,             // task pv params
                          0,                   // priority
                          nullptr,             // handle
                          1                    // core
  ); ```

brandond, 2019-06-11:
I believe the only difference between Q5 and Q6 is the package.

https://www.cnx-software.com/2017/04/04/new-espressif-esp32-single-and-dual-core-processors-in-5x5mm-package-optional-embedded-flash-coming-soon/

OttoWinter, 2019-06-11:
@keith721 No, that's probably not it - yes the task is pinned to core 1 but the code in the created task is properly protected with mutexes. (And if that were the issue, it would fail with all ESP32 camera boards)

keith721, 2019-06-13:
@OttoWinter I sent email to you this week.  If you're willing, I'll send one of these camera boards to you for research.

keith721, 2019-06-14:
I took another look at Lewis Xhe's Arduino sketch (which has supporting CPP source files) and saw a second use of the TTGO_OV2640_V16 macro I did not previously notice.  Could this be part of the issue?
```
#ifdef TTGO_OV2640_V16
    /* IO13, IO14 is designed for JTAG by default,
    * to use it as generalized input,
    * firstly declare it as pullup input */
    pinMode(13, INPUT_PULLUP);
    pinMode(14, INPUT_PULLUP);
#endif
```

keith721, 2019-06-14:
I inserted those two pinMode commands for GPIO13 and GPIO14 into the beginning of the CPP source code file, and can now successfully compile, load, and run using this TTGO T-Camera V1.6 w/MIC board. :)

Lefuneste83, 2019-06-15:
Great work Keith. I'll try this now...

Lefuneste83, 2019-06-15:
Yes I confirm a signicative improvement. My cams are much more stable and are able to reboot every time which was my main concern as I had plug/unplug them several times in order for them to connect. Since I applied your patch, I can unplug them and they will come back to life everytime, and show their sensors in HA. Great job !

keith721, 2019-06-15:
Following this up, the wifi performance is still a bit unstable.  Checking the other low-order GPIO pins used, I'll add GPIO12 to that list, and see if it gets better.  Worst thing that can happen is no change.

sunflowerABB, 2019-06-15:
I confirm that your changes in the file "/ttgo_cam01/src/main.cpp":
` 
    pinMode(12, INPUT_PULLUP);
    pinMode(13, INPUT_PULLUP);
    pinMode(14, INPUT_PULLUP);
    pinMode(15, INPUT_PULLUP);
`
do solve the problem for me. I'm located close to my wifi router though so cannot confirm the signal problems, even OTA updates work without any problems for me and there are no errors in the log.

This is my configuration.yaml
`esp32_camera:
  external_clock:
    pin: GPIO4
    frequency: 20MHz
  i2c_pins:
    sda: GPIO18
    scl: GPIO23
  data_pins: [GPIO34, GPIO13, GPIO14, GPIO35, GPIO39, GPIO12, GPIO15, GPIO36]
  vsync_pin: GPIO5
  href_pin: GPIO27
  pixel_clock_pin: GPIO25

  # Image settings
  name: test_cam
  max_framerate: 10 fps
  idle_framerate: 0.1 fps
  resolution: 640x480
  jpeg_quality: 10
  vertical_flip: true
  horizontal_mirror: false`

Keep us posted about future improvements but at least the camera is usable now.

keith721, 2019-06-15:
Re: wifi performance and OTA updates, I suspect the PCB trace antenna on my board is not connected.  I'll need to visually confirm this because there is no U.FL connector socket there.

sunflowerABB, 2019-06-15:
> Re: wifi performance and OTA updates, I suspect the PCB trace antenna on my board is not connected. I'll need to visually confirm this because there is no U.FL connector socket there.

There is no U.FL connector on my board (with mic) either, however a small printed camera connected through two golden pins appears to be soldered to the ESP32, should be ok.

keith721, 2019-06-16:
The RF output of the ESP32 chip is NOT connected to anything.  There are pads for the R14 U.FL connector (not installed) and R15 PCB trace, but neither is connected nor populated.  I'll have to solder a jumper wire and/or blob of solder to connect the on-board antenna trace, which should perform reasonably well.  It's considerably larger than most.
[image]

keith721, 2019-06-16:
Since we've identified the source of this camera's crash, and identified the proper pin for the PIR sensor, I believe it best to go ahead and close the thread.  Thank you to @OttoWinter for his fantastic ESPhome package and for allowing users of this particular TTGO T-Camera board to figure it out here.  Good luck to everyone!

closed 2019-06-16 by keith721