esphome/issues #406: Bang Bang Climate HA oddities

chaolue, 2019-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff / Sonoff POW

**Affected component:**

Bang Bang Climate

**Description of problem:**
A number of issues when using the new climate component under Home Assistant through the ESPHome API:

1. I'm unable to set the 'target_temp_low' via a HA automation or service call. I receive the following HA error - `some but not all values in the same group of inclusion 'temperature' @ data[<temperature>]`

Example service call:
`{
  "entity_id": "climate.our_heater",
  "temperature": 20,
  "target_temp_low": 21
}`

2. Maybe my misunderstanding of how the new bang bang climate component should work; but shouldn't a heating-only climate device only need a min temperature?

3. Cosmetically, when the visual temperature_step is set in ESPHome, the climate component temperature (sensor) value is also rounded. I would have expected that the temperature step only effected the slider for setting desired temperature, not the displaying of the actual read sensor temperature.

4. Also cosmetically, usually in HA when a climate device is not heating it's operation mode shows as 'Idle', while ESPHome climate devices stay in whatever operation mode they are set to (auto or heat) irrespective of whether the heating or not. 

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    sensor: dht_temperature
    name: Our Heater
    visual:
      min_temperature: 14
      max_temperature: 26
      temperature_step: 0.5
    default_target_temperature_low: 16
    default_target_temperature_high: 26
    away_config:
      default_target_temperature_low: 14
      default_target_temperature_high: 26
    heat_action:
    - switch.turn_on: relay
    idle_action:
    - switch.turn_off: relay

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

chaolue, 2019-06-05:
In regards to 1. above, I see that I need to enter both "target_temp_low" and "target_temp_high" for this service call to work

OttoWinter, 2019-06-06:
1. Exactly, both `target_temp_low` and `target_temp_high` must be set - that's a limitation of the HA core.

2. Well at some point the heating device should also shut off, right? That's what the max temperature is for - also see wikipedia entry for bang bang controller.

3. Also a limitation of HA - ESPHome is reporting the values correctly, HA is rounding them.

4. Then the other climate device in HA is not operating with the HA spec - while the climate device is in automatic control mode, HA mandates it should display as auto mode. The actual operation of the device in auto mode is not supported in HA (see also climate 1.0 in HA). ESPHome does report that value correctly, HA just does not display it.

All of these are HA limitations/issues, so closing the issue.

closed 2019-06-06 by OttoWinter

chaolue, 2019-06-06:
4. When using the HA's generic thermostat, it shows 'idle' as the climate device state when the heater entity id is off.

I guess I was expecting the climate component to work similar to this, obviously I was wrong. And hence my reasoning behind the other issues I've been having. So, I guess high/low settings make no difference when a climate device is in heat (or cool) mode - the heat/cool_actions just happen.

5. Is it a HA issue that a climate device when going in and out of away mode loses the previous state? It reverts to the default_target_temperature_low/high value.

OttoWinter, 2019-06-06:
4. Yes, I know. However, there's a bunch of problems with that approach (besides it being against HA standard): For example, there's no way to see from the frontend between an automatic idle state and a forced idle state.
5. No, that would be ESPHome - however I think that's more like a feature; when it goes into/out of away mode you want it to reset to the default values. At least in my setup it makes more sense like that.