esphome/issues #407: Request for feedback,

thompsa, 2019-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 dev kit C

**Description of problem:**

Adding support for the CircuitSetup Expandable 6 Channel ESP32 Energy Meter board which has two Atmel M90E32AS chips and an ESP32 header.

https://circuitsetup.us/index.php/product/expandable-6-channel-esp32-energy-meter/

Initial support here, looking for comments before creating a pull request.

https://github.com/thompsa/esphome/commit/40f83136974202f400410fd9f5a18c1a5f2b3b35

One issue I had was that the device requires SPI MODE3. I was unable to get this working with the bitbanging code so added in support for using the Arduino Framework SPI controller.

example output is ```
[12:46:48][D][sensor:092]: 'EMON Line Voltage': Sending state 231.42000 V with 1 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT1 Current': Sending state 0.00500 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT2 Current': Sending state 0.00500 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT3 Current': Sending state 0.00500 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON Line Frequency': Sending state 49.84000 hz with 1 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON Total Active Power CT1-3': Sending state -0.05440 W with 1 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT4 Current': Sending state 0.00600 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT5 Current': Sending state 0.00300 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON CT6 Current': Sending state 0.00100 A with 2 decimals of accuracy
[12:46:48][D][sensor:092]: 'EMON Total Active Power CT4-6': Sending state -0.04768 W with 1 decimals of accuracy```

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
   - platform: atm90e32
    cs_pin: 5
    voltage_a:
      name: "EMON Line Voltage"
    current_a:
      name: "EMON CT1 Current"
    current_b:
      name: "EMON CT2 Current"
    current_c:
      name: "EMON CT3 Current"
    frequency:
      name: "EMON Line Frequency"
    power:
      name: "EMON Total Active Power CT1-3"
    line_freq: 50Hz
    gain_pga: 2X
    gain_volt_a: 47660
    gain_ct_a: 12577
    gain_ct_b: 12577
    gain_ct_c: 12577
  - platform: atm90e32
    cs_pin: 4
    current_a:
      name: "EMON CT4 Current"
    current_b:
      name: "EMON CT5 Current"
    current_c:
      name: "EMON CT6 Current"
    power:
      name: "EMON Total Active Power CT4-6"
    line_freq: 50Hz
    gain_pga: 2X
    gain_volt_a: 47660
    gain_ct_a: 12577
    gain_ct_b: 12577
    gain_ct_c: 12577

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-06:
> One issue I had was that the device requires SPI MODE3. I was unable to get this working with the bitbanging code so added in support for using the Arduino Framework SPI controller.

SPI mode 3 just has clock polarity and phase a bit different

See here for more info: https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Operation

There are some problems with how you implemented the HW-SPI (for example, ESP32 has multiple SPI buses, each of those would need to be supported etc). Plus the implementation shouldn't need to care if the underlying implementation is bit banging or using HW spi.

I would recommend adding CPOL and CPHA support to the spi base - with virtual functions returning bools for each (not a `mode` function that returns an int, because that depends on how the mode is mapped to an int).

---------

On the rest of the code (preferably I would have this in a PR, it gets quite hard to communicate via links; you can even mark the PR as a draft if you like)

- Some of the functions don't use ESPHome's naming guides - see contributing guide
- A lot of the lines in `setup()` appear to be copied from a M90E32AS library. But looking at the datasheet, the values written there are what the sensor defaults to on power-up anyway. I'd say we remove those that set values that are already the defaults on power up.
- Some methods appear to be implemented (like GetExportReactiveEnergy), but do not have `Sensor` associated to them. If they're not going to be exposed as sensors, let's remove them.

- File `atm90e32_reg.h` - defining values as `static const` in a header file will make them be included in every compile unit that includes the header file, increasing the binary size a lot. Either move them into the .cpp compile unit or use other tricks like anonymous namespaces to include them only once.

- https://github.com/thompsa/esphome/commit/40f83136974202f400410fd9f5a18c1a5f2b3b35#diff-570d665ee4bbf01774bf625719756e29R85 - All these values have defaults, the `if` check can be removed.

thompsa, 2019-06-06:
Thanks for the feedback. I'll fix up what you have mentioned then submit a PR which will allow line by line comments.

closed 2019-06-06 by thompsa

digiblur, 2019-06-06:
@thompsa wow!  Excellent work!  I have one of these boards too and would prefer to do it over ESPHome instead of custom arduino sketches.  Looks like they way you are building it that this would also support the 2 channel board as well as the 6 with even ones stacked on it due to the declaring of the CS pin per chip.  I'd be glad to test this PR as well once you have it ready.

thompsa, 2019-06-06:
The 2 channel should work fine and I have one on order to test. Once it is all working I will submit a documentation PR to describe the sensor and also give yaml samples for both boards.

thompsa, 2019-06-06:
@OttoWinter 

> SPI mode 3 just has clock polarity and phase a bit different
>
> See here for more info: https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Operation
>
> There are some problems with how you [...]

I am not fussed on hw-spi, I just couldnt get the bit banging logic right for the different phase and needed to progress. I think CPOL and CPHA would make sense, I will try again.

> * Some methods appear to be implemented (like GetExportReactiveEnergy), but do not have `Sensor` associated to them. If they're not going to be exposed as sensors, let's remove them.

Ok. The chip exports a lot of different energy data and it is not clear if any others should be sensors. I'll clear out the unused ones and keep it clean.

> * File `atm90e32_reg.h` - defining values as `static const` in a header file will make them be included in every compile unit that includes the header file, increasing the binary size a lot. Either [...]

I don't imaging any other file other than atm90e32.cpp would ever include this file. The other option would be to use preprocessor defines which do not increase code size.

digiblur, 2019-06-06:
Sounds good. I have the two channel running right now on a Esp8266 as well to test with as well as the 6 channel.

OttoWinter, 2019-06-07:
> I don't imaging any other file other than atm90e32.cpp would ever include this file. 

The ESPHome main.cpp file base includes all .h files - and if the user is using custom code in .cpp files, it will also be included there.

But, I read up on static a bit more, and it seems like this is using internal linkage - so it won't be a problem. You can leave it as-is.

OttoWinter, 2019-06-07:
@thompsa I needed the SPI settings for another thing I'm working on - so I redid the SPI base a bit to add these options. And now everything's configured at compile time, which means faster SPI transactions at runtime.

https://github.com/esphome/esphome/pull/622/files

OttoWinter, 2019-06-07:
In your case it would be
```cpp
public spi::SPIDevice<spi::BIT_ORDER_MSB_FIRST, spi::CLOCK_POLARITY_HIGH, 
                      spi::CLOCK_PHASE_TRAILING, spi::DATA_RATE_200KHZ>
```

thompsa, 2019-06-07:
@OttoWinter 
I have pulled these changes and can confirm it is working with the Atmel chip. Thanks!