esphome/issues #409: Sonoff basic not connecting to wifi after updating to version 1.13.4

timon71, 2019-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Homeassistant 0.94.0 on Hassio with HassOS 2.12 and supervisor 166

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff basic

**Affected component:**

https://esphome.io/devices/sonoff_basic.html

**Description of problem:**

After updating to the latest version of ESPHome 1.13.4 my Sonoff basic wouldn't connect to wifi anymore after succesfully uploading the latest compiled firmware over-the air. 
I had to flash the sonoffs manualy to the previous formware to get them running again.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: bedlamp
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "*********"
  password: "**************"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '***********'

ota:
  password: '***********'

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Bedlamp Button"
    on_press:
      - switch.toggle: fakebutton

switch:
  - platform: template
    name: "Bedlamp schakelaar"
    optimistic: true
    id: fakebutton
    turn_on_action:
    - switch.turn_on: relay
    - light.turn_on: led
    turn_off_action:
    - switch.turn_off: relay
    - light.turn_off: led
  - platform: gpio
    id: relay
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "Bedlamp Green LED"
    output: basic_green_led
    id: led

sensor:
  - platform: wifi_signal
    name: "Bedlamp WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "Bedlamp Uptime"
  
text_sensor:
  - platform: version
    name: "Bedlamp ESPHome Version"
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-06:
Logs please, without them connectivity issues are impossible to diagnose.

Please attach the device to your computer via USB (but do not plug in mains) and see what error it shows while trying to connect to wifi.

timon71, 2019-06-06:
Sorry. Here is mij serial logging from my sonoff. Keeps restarting de WiFi adapter after giving an error "Auth expired".

I have mesh wifi with 3 tp-link deco m5 access point. Thats why you see my SSID 3 times with different mac addresses.

### logging from putty:
[D][wifi:268]: Starting scan...
[D][wifi:283]: Found networks:
[I][wifi:318]: - 'TIMON' (0C:80:63:B3:39:4E) ▂▄▆█
[D][wifi:319]:     Channel: 7
[D][wifi:320]:     RSSI: -60 dB
[I][wifi:318]: - 'TIMON' (0C:80:63:B3:38:7A) ▂▄▆█
[D][wifi:319]:     Channel: 7
[D][wifi:320]:     RSSI: -85 dB
[D][wifi:322]: - '' (12:80:63:B3:39:4E) ▂▄▆█
[I][wifi:164]: WiFi Connecting to 'TIMON'...
[W][wifi_esp8266:354]: Event: Disconnected ssid='TIMON' bssid=0C:80:63:B3:39:4E reason='Auth Expired'
[W][wifi:396]: Error while connecting to network.
[W][wifi:427]: Restarting WiFi adapter...
[I][wifi:164]: WiFi Connecting to 'TIMON'...
[W][wifi_esp8266:354]: Event: Disconnected ssid='TIMON' bssid=0C:80:63:B3:39:4E reason='Auth Expired'
[W][wifi:396]: Error while connecting to network.
[W][wifi:427]: Restarting WiFi adapter...

cliffordwhansen, 2019-06-06:
I have also just updated to the latest version:
```
$ esphome dummy.yaml version
Version: 1.13.4
```

I finally managed to get a serial log from the one device:
```
[W][wifi:427]: Restarting WiFi adapter...
[D][wifi:268]: Starting scan...
[D][wifi:283]: Found networks:
[D][wifi:285]:   No network found!
```

```
---
substitutions:
  devicename: lounge_floor_light
  upper_devicename: Lounge Floor Light

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.18.1.100
    gateway: 10.18.1.254
    subnet: 255.255.255.0

logger:
api:
ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    name: ${upper_devicename} Button
    on_press:
      - light.toggle: light_relay
  - platform: status
    name: ${upper_devicename} Status

switch:
  - platform: restart
    name: ${upper_devicename} Restart

output:
  - platform: esp8266_pwm
    id: basic_green_led
    frequency: 1000 Hz
    pin:
      number: GPIO13
      inverted: true
  - platform: gpio
    id: relay
    pin: GPIO12

light:
  - platform: monochromatic
    name: ${upper_devicename} LED
    output: basic_green_led
  - platform: binary
    name: ${upper_devicename}
    output: relay
    id: light_relay

sensor:
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
  - platform: uptime
    name: ${upper_devicename} Uptime

text_sensor:
  - platform: version
    name: ${upper_devicename} version
```

I even added the `fast_connect: on` setting and then I get:
```
[W][wifi:427]: Restarting WiFi adapter...                                                                                                                                
[I][wifi:164]: WiFi Connecting to 'xxx'...
[W][wifi_esp8266:354]: Event: Disconnected ssid='xxx' bssid=00:00:00:00:00:00 reason='AP Not Found'
[W][wifi:396]: Error while connecting to network.
```

After the 5 minute timeout:
```
[E][wifi:104]: Can't connect to WiFi, rebooting...
[I][app:122]: Forcing a reboot...

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
   [I][logger:116]: Log initialized
[C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[I][app:027]: Running through setup()...
[D][binary_sensor:033]: 'Lounge Floor Light Button': Sending state OFF
[C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[C][light:094]: Setting up light 'Lounge Floor Light LED'...
[D][light:261]: 'Lounge Floor Light LED' Setting:
[D][light:274]:   Brightness: 100%
[C][light:094]: Setting up light 'Lounge Floor Light'...
[D][light:261]: 'Lounge Floor Light' Setting:
[D][binary_sensor:033]: 'Lounge Floor Light Status': Sending state OFF
[D][text_sensor:015]: 'Lounge Floor Light version': Sending state '1.13.4 Jun  6 2019, 19:01:50'
[C][wifi:029]: Setting up WiFi...
[I][wifi:164]: WiFi Connecting to 'xxx'...
[D][sensor:092]: 'Lounge Floor Light Uptime': Sending state 0.15500 s with 0 decimals of accuracy
[W][wifi_esp8266:354]: Event: Disconnected ssid='xxx' bssid=00:00:00:00:00:00 reason='AP Not Found'
[W][wifi:396]: Error while connecting to network.
[W][wifi:427]: Restarting WiFi adapter...
```

The biggest problem I have now is flashing all my devices, I don't have easy access to them (but that is a me problem as I should not have installed them where I did ;) )

OttoWinter, 2019-06-06:
@cliffordwhansen That would be for the issues https://github.com/esphome/issues/issues/82 and https://github.com/esphome/issues/issues/100 - Please continue the `AP not found` discussion there. TL;DR, that errors seems to appear quite randomly for some people {haven't had a single repro yet :( } Try enabling verbose mode and see if that helps; I don't know why but for some reason it works for some people.

@timon71 That is an interesting/new issue: `Auth Expired`. Could you try restarting the router?

timon71, 2019-06-06:
I allready did. Even updatet my Deco ap’s to the latest firmware after rebooting. 

When I flashed my sonoffs back to an older version of esphome  it works like a charm.

qoobaa, 2019-06-07:
I can confirm the issue. I've just lost access to two Sonoff Basics - they're not connecting to the WiFi after update.

15819996676, 2019-06-07:
老大，更新到最新版本的命令是什么。怎么更新

abmantis, 2019-06-07:
I'm also having connection issues on a shelly 1. After rebooting 10 times it goes into safe mode, but still can't connect:
```
[23:03:36][C][ota:364]: There have been 10 suspected unsuccessful boot attempts.
[23:03:36][E][ota:369]: Boot loop detected. Proceeding to safe mode.
[23:03:36][C][wifi:029]: Setting up WiFi...
[23:03:36][D][wifi:268]: Starting scan...
[23:04:06][E][wifi:276]: Scan timeout!
[23:04:11][D][wifi:268]: Starting scan...
[23:04:41][E][wifi:276]: Scan timeout!
```

Lefuneste83, 2019-06-07:
Ok Guys I have noticed a pretty obvious regression in 1.13.4. I have spent the last 48 hours trying to nail it (I am new to ESPHOME sorry...). I had all my devices gone belly up...

Env : Home Assistant 94.0 on Docker/Portainer on Debian 9.2
Broker : MQTT Moquitto Add On (also tested with Mosquitto / Web server Add On)

Devices : mainly nodemcu-32s and mhetesp32devkit, also affects Wemos D1 mini and TTGO Cam with Mic

Platforms used : ledc, cwww, wifi_signal, dallas, gpio, xiaomi_mijia, ble_rssi...

Connectivity mode : Pure MQTT (but also witnessed the same behaviour with API mode)

Symptoms : 
1/ After hard reset, device would first connect fine then disconnect after 1 min roughly, then after 4 to 5 min reboot and regain connectivity, then would loose it again, and so on... All devices compiled against 1.13.4 would immediatly suffer from the issue.

2/ MQTT Broker would trace a similar message : Client whatever-840d8eaxxxxx has exceeded timeout, disconnecting.

3/ Wifi AP would trace a message such as : Sat Jun  8 01:00:11 2019 daemon.info hostapd: wlan0: STA 84:0d:8e:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE) 

Course of Actions : 
Changed Broker version, controlled the Wifi settings (OpenWRT AP) reversed to previous HA snapshot, uninstalled / reinstalled DEV and MASTER Branches of ESPHOME, tried to recompile against DEV and MASTER branches (on Hassio Add On and on PC), nothing changed the behaviour.

Solution :
Rolled back the ESPHOME add on version to 1.13.2 and recompiled the ESP32 code. Immediate correction of the symptoms as soon as recompiled code is pushed to a device. Tried the same thing with version 1.13.3, also working fine.

Working settings for ESPHOME 

```
{
  "esphome_version": "v1.13.3",
  "ssl": false,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}
```

Conclusion : Something is breaking the Wifi in 1.13.4. From the ventilation of my different sensors between devices and cross tests, I am quite sure it is an underlying lib which is at the origin of the issue and not a specific platform such as ble tracker which I first suspected. Maybe the wifi_signal is suspect as I tend tp put it on all devices. The change was brought by this version 1.13.4

Lefuneste83, 2019-06-08:
Ok logs now : 

YAML FILE : 

```
esphome:
  name: cave
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "MYSSID"
  password: "MYPWD"

# Enable logging
logger:
    level: VERY_VERBOSE

# Enable Home Assistant API
mqtt:
   broker: "192.168.0.XX"
   username: "MQTTLOGIN"
   password: "MQTTPWD"
   birth_message:
   will_message:

ota:
  password: 'OTAPWD'

esp32_ble_tracker:

sensor:
  - platform: wifi_signal
    name: "Wifi Signal Sensor"
    update_interval: 133s
    
  - platform: xiaomi_mijia
    mac_address: 4C:65:A8:D2:6C:96
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    battery_level:
      name: "Battery Level"
      
  - platform: ble_rssi
    mac_address: 4C:65:A8:D2:6C:96
    name: "RSSI value"
```

LOGS : 

```
Leaving...
Hard resetting via RTS pin...
================================================================================================================================================ [SUCCESS] Took 81.86 seconds ================================================================================================================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[02:02:28][I][logger:116]: Log initialized
[02:02:28][VV][preferences:033]: LOAD 0: valid=YES, 0=0x00000008 1=0x75320207 (Type=233825507, CRC=0x75320207)
[02:02:28][C][ota:364]: There have been 8 suspected unsuccessful boot attempts.
[02:02:28][VV][preferences:046]: SAVE 0: 0=0x00000009 1=0x4A164723 (Type=233825507, CRC=0x4A164723)
[02:02:28][I][app:027]: Running through setup()...
[02:02:28][V][app:028]: Sorting components by setup priority...
[02:02:29][D][esp32_ble_tracker:146]: Starting scan...
[02:02:29][C][wifi:029]: Setting up WiFi...
[02:02:29][V][wifi_esp32:033]: Enabling STA.
[02:02:29]I (753) wifi: wifi driver task: 3ffe5594, prio:23, stack:3584, core=0
[02:02:29]I (1590) wifi: wifi firmware version: 6d404d4
[02:02:29]I (1594) wifi: config NVS flash: enabled
[... 86 lines cut ...]
[02:03:34]Core 1 register dump:
[02:03:34]PC      : 0x400d72ed  PS      : 0x00060630  A0      : 0x800d80af  A1      : 0x3ffce130  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d cave -t idedata
WARNING Decoded 0x400d72ed: esphome::wifi::WiFiComponent::print_connect_params_()
[02:03:35]A2      : 0x3f405cb1  A3      : 0x3ffc5944  A4      : 0x00000000  A5      : 0x3ffc73ac  
[02:03:35]A6      : 0x00000178  A7      : 0x3ffcc680  A8      : 0x800d72ed  A9      : 0x3ffce110  
[02:03:35]A10     : 0x3ffce15c  A11     : 0x3f405cb1  A12     : 0x3f403a0f  A13     : 0x3ffce0e0  
[02:03:35]A14     : 0x000000fb  A15     : 0x3ffe5b30  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[02:03:35]EXCVADDR: 0x00000005  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff3  
[02:03:35]
[02:03:35]Backtrace: 0x400d72ed:0x3ffce130 0x400d80ac:0x3ffce1c0 0x400d8b55:0x3ffce1f0 0x400dba3a:0x3ffce210 0x400db47f:0x3ffce230 0x400dd36a:0x3ffce280 0x400e32e7:0x3ffce5c0 0x4008e1b1:0x3ffce5e0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d72ed: esphome::wifi::WiFiComponent::print_connect_params_()
WARNING Decoded 0x400d80ac: esphome::wifi::WiFiComponent::check_connecting_finished()
WARNING Decoded 0x400d8b55: esphome::wifi::WiFiComponent::loop()
WARNING Decoded 0x400dba3a: esphome::Component::call_loop()
WARNING Decoded 0x400db47f: esphome::Application::setup()
WARNING Decoded 0x400dd36a: setup()
WARNING Decoded 0x400e32e7: loopTask(void*)
WARNING Decoded 0x4008e1b1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[02:03:35]
[02:03:35]Rebooting...

```

15819996676, 2019-06-08:
怎么更新到最新版本。方法是什么，命令是什么
我的是homeassistnt

Lefuneste83, 2019-06-10:
I can confirm that this connection issue has been resolved with the 1.13.5 upgrade. 1.13.4 definitely had a regression on my devices despite other people apparently not having any issue with it.

timon71, 2019-06-10:
@OttoWinter I can confirm that in version 1.13.5 my wifi problem of my sonoff basics has been solved. Thnx

closed 2019-06-10 by timon71

OttoWinter, 2019-06-10:
@timon71 Well that would be interesting considering 1.13.5 contained no ESP8266 wifi  fixes (only ESP32 ones). So only @Lefuneste83's issue should be resolved. This shows again how weird this ESP8266 Wifi AP not found (or similar) issue is: It appears seemingly randomly.

It might also have to do with the esp toolchain - could you provide information on the host this is running on? I see you wrote it's HassOS, please also say what system (RPi? model?) it is running on and in the case of the RPi if you're using the 64-bit image.

qoobaa, 2019-06-10:
I upgraded one of my Sonoff Basics to 1.13.5 - it still doesn't connect to the WiFi network. I don't think the issue is resolved.

qoobaa, 2019-06-10:
I'm using Hass.io on a virtual machine (x64), I can paste some more details and logs later.

timon71, 2019-06-10:
@OttoWinter I’m running hassos as VM in VMWare workstation player. Hardware is a Dell optiplex 3050 mff intel i3 with Windows 10 64bits OS.

michenew, 2019-06-13:
I upgraded a Sonoff Basic to 1.13.5 version and it doesn't connect to Wifi. With 1.12.2 I had the same wifi problem and I solved using Verbose mode in logger. 
Now I've noticed the Sonoff doesn't connect to wifi if powered by 220V. If powered by FTDI interface, wifi is perfectly working. Could it be a hardware power issue?

nmaggioni, 2019-06-13:
After a few days of not being able to reach my Sonoff (4CH Pro R2) on v1.13.4 **(even though it was working fine with HomeAssistant)**, it magically bursted to life and was reachable again on port 6053. Upgrade to v1.13.5 was then completed smoothly.

Just Sonoffs being Sonoffs I guess.

OttoWinter, 2019-06-14:
> Now I've noticed the Sonoff doesn't connect to wifi if powered by 220V. If powered by FTDI interface, wifi is perfectly working. Could it be a hardware power issue?

Could be, although for some other people FTDI does not work either.

Maybe you could try adjusting the power_save_mode option in WiFi to light to see (it will reduce wifi power usage)

OttoWinter, 2019-06-18:
Moving the discussion to #455