esphome/issues #411: AARCH64

OvCa77, 2019-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

Complier

**Description of problem:**
Error compiling on AARCH64

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "a"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Traceback (if applicable):**

```
Compiling /data/test/.pioenvs/test/src/esphome/components/api/command_messages.cpp.o
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_message.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_server.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/src/esphome/components/api/basic_messages.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/src/esphome/components/api/command_messages.cpp.o] Error 132
========================== [ERROR] Took 9.59 seconds ==========================
```

**Additional information and things you've tried:**

OttoWinter, 2019-06-06:
Duplicate of https://github.com/esphome/issues/issues/129, which will be solved with 1.13.5

You can already try it by installing the ESPHome dev version, or by waiting for 1.13.5 to be released.

closed 2019-06-06 by OttoWinter

OvCa77, 2019-06-06:
I have tried dev version, but it is the same.

OttoWinter, 2019-06-06:
@OvCa77 Ok, let's please still continue the discussion in #129

Looks like esp8266 - the patch I deployed is only for ESP32, I though ESP8266 was working fine.

Could you verify at least ESP32 is working? Then I can do the same thing for ESP8266 too.