esphome/issues #412: Nextion Slider is unusable

AalianKhan, 2019-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPhome 1.13.4 running on Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini
**Affected component:**

https://esphome.io/components/display/nextion.html

**Description of problem:**
a slider is a component in the nextion editor where when you slide it the value of that component changes. What I want to do is when the slider has a value of 6 display some information on the display which is stored on the node. However, I can't do this because My Wemos D1 mini doesn't know the value of the slider. To get the value of the slider you use the command `get [component names].val` and this will get you the current value of the slider/component but currently the Nextion Component for ESPhome doesn't know how to parse the returned value. I can see this happening by adding a sensor component to get the values of components. In the sensor, you will configure which component you will want to parse the value for and the simple command shown above. if for some reason we can't send the command every second or so, we can also type in the same command in the nextion editor in the `on touch` section but if we go this route we won't be able to get values of variables since they don't have a trigger so to counter that we can choose if its a variable or a touchable component that sends the value when it's changed. 

Also, FYI variables can also be strings so I am not sure how you might handle that.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

AalianKhan, 2019-06-06:
here is the command I was talking about
https://www.itead.cc/wiki/Nextion_Instruction_Set#get:_Get_variable.2Fconstant_value_with_format

Andoramb, 2019-09-19:
I think this could be traced back to [Nextion sensor - Feature request](https://github.com/esphome/feature-requests/issues/201)

If it would be possible to interpret at least numbers from UART, that would make sliders work
and a lot of other things like: pageid (from 'sendme' command) or other user inputs, variables, that are not buttons

Andoramb, 2019-09-19:
Also, I tried this by using custom UART sensor, I literally copied the steps from the site, but it gives:

> error: expected class-name before '{' token

at
`class MyCustomComponent : public Component, public UARTDevice {`

closed 2020-12-05 by stale[bot]

martinkangur, 2020-12-23:
Any updates. I want to read slider values

rspaargaren, 2021-01-12:
Have a look at the following PR: https://github.com/esphome/esphome/pull/1464
This PR gives you full functionality on the slider and much more!