esphome/issues #414: A fatal error occurred: MD5 of file does not match data in flash! *** [upload] Error 2

vash3d, 2019-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 
**Affected component:**

https://esphome.io/devices/esp32.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: aqua_smart
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxxx"
  password: "xxxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

dallas:
  - pin: GPIO23

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/aqua_smart
Processing aqua_smart (platform: espressif32@1.8.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Retrieving maximum program size /data/aqua_smart/.pioenvs/aqua_smart/firmware.elf
Checking size /data/aqua_smart/.pioenvs/aqua_smart/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.5% (used 41040 bytes from 327680 bytes)
PROGRAM: [=====     ]  50.3% (used 823778 bytes from 1638400 bytes)
========================= [SUCCESS] Took 12.99 seconds =========================
INFO Successfully compiled program.
INFO Running:  platformio run -d /config/esphome/aqua_smart -t upload --upload-port /dev/ttyUSB1
Processing aqua_smart (platform: espressif32@1.8.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/aqua_smart/.pioenvs/aqua_smart/firmware.elf
Checking size /data/aqua_smart/.pioenvs/aqua_smart/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.5% (used 41040 bytes from 327680 bytes)
PROGRAM: [=====     ]  50.3% (used 823778 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB1
Uploading /data/aqua_smart/.pioenvs/aqua_smart/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:96:a3:ad
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 16848 bytes to 10928...
Wrote 16848 bytes (10928 compressed) at 0x00001000 in 1.0 seconds (effective 136.4 kbit/s)...
File  md5: 09cd5c6fed8b2916c9a0c52375a1fb38
Flash md5: cd72601290ea84044e04f833f370f86a
MD5 of 0xFF is 7b91d5c78935317cc54803dd1cb49a6f

A fatal error occurred: MD5 of file does not match data in flash!
*** [upload] Error 2
========================== [ERROR] Took 16.27 seconds ==========================
help_outline
```

**Additional information and things you've tried:**

I've seen there is already a thread with this error but it's closed without solution.
I've tried both with NodeMCU and generic esp32.
On the board there is an old sketch uploaded with an old version of esphome running on an Mac, cause espHome on rpi (hass.io) wasn't compatible with esp32 boards at that time. Maybe is this the problem?

vash3d, 2019-06-07:
I've compiled and saved the bin file and tried to upload it using esptool and got the same error.
Hence this issue is probably not directly related to esphome and should be closed.
Anyway, can someone give me some hint to solve this?
Thank you

OttoWinter, 2019-06-08:
This means your flash chip is broken or has some bad sectors. You can also try flashing another fw on there, but you'll very likely see the same problem.

closed 2019-06-08 by OttoWinter

kluszczyn, 2019-06-11:
I have the same issue, also with ESP32. It happened after about ~1month of esp usage, when uploading revised fw (based on 1.13.5). Upload (OTA) was successful, but then esp started in safe mode. Now not able to upload new fw, OTA nor serial. My esp is soldered into custom made PCB, it makes things more complicated to fix.

kluszczyn, 2019-06-22:
I resolved my issue. It turned out that GPIO2 was pulled HIGH (per my design, by pull up resistor) and during serial upload one need to keep it LOW. When GPIO2 is pulled HIGH and one initiate serial upload (GPIO0 LOW when powering up) esptool cannot communicate with flash and generates error messages as flash would be broken.

When flashing via OTA GPIO2 can be HIGH and flashing is still possible.