esphome/issues #416: ESP32 Cam will not work with esphome binary

boxy85, 2019-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
I cannot get the ESP32 board to run using ESP home no matter what config i try. I have removed all information regarding the camera from the Yaml file and it still doesnt load.

I have used 4 different boardtypes in my yaml file. I have tried just a standard ESP intstall with no other components. Any help would be appreciated

If i install the standard esp32 camera webserver using arduino then the camera works perfectly. I have a esp32-s Ai-thinker board with a camera attached.
I am using NodeMCU to upload the .bin file.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: cam2
  platform: ESP32
  board: nodemcu-32s
  
wifi:
  ssid: "redacted"
  password: "redacted"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "redacted"

ota:
  password: "redacted"
# Author by airijia.com
# ESP32-CAM
# Lovelace camera card https://github.com/custom-cards/camera-card
# Video tutorial https://youtu.be/-towjof_ixU

#wifi:
#  ssid: $ssid
#  password: $password
#  fast_connect: True
#api:
#  reboot_timeout: 0s
#ota:
#logger:

# ESP32-CAM

# Example configuration entry
esp32_camera:
  external_clock:
    pin: GPIO21
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO4, GPIO5, GPIO18, GPIO19, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  reset_pin: GPIO15
  resolution: 640x480
  jpeg_quality: 10
  # Image settings
  name: My Camera
# ... Frame Settings

# Flashlight
#output:
#  - platform: gpio
#    pin: GPIO4
#    id: gpio_4
#light:
#  - platform: binary
#    output: gpio_4
#    name: "light"

```

**Traceback (if applicable):**

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)flash read err, 1000ets_main.c 371
```

**Additional information and things you've tried:**

OttoWinter, 2019-06-08:
How did you upload the firmware file? I'm assuming it's neither esphome or esphome-flasher.

Most NodeMCU flashers are at least partially incompatible with the binaries ESP32s produce - and exactly that error will show up then.

boxy85, 2019-06-16:
Thanx OttoWinter. Was away for a week or so but finally I used the esphome flasher through the Pi and it worked. Had a few more different errors after that but finally got it working.

closed 2019-06-16 by boxy85