esphome/issues #420: Version: 1.13.5 Breaks Raw remote codes in yaml

darshkpatel, 2019-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker 
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/binary_sensor/remote_receiver.html
**Description of problem:**
I have been using docker images of esphome to compile and upload images to esp8266 devices. After updating to the latest image by `docker pull esphome/esphome`
running  
`sudo docker run --rm -v "${PWD}":/config -it esphome/esphome test.yaml version` 
gives `Version: 1.13.5`
 compiling using   
`sudo docker run --rm -v "${PWD}":/config -it esphome/esphome test.yaml compile`
would give me an error 
```
INFO Reading configuration...
Failed config

binary_sensor.remote_receiver: [source test.yaml:91]
  platform: remote_receiver
  id: code1
  on_press: 
    then: 
      - switch.toggle: relay1
  filters: 
    - delayed_off: 500ms
  
  expected a dictionary.
  raw: 
    - -4553
    - 4486
    - -528
    - 478
    - -527
    - 480
 ............................. same thing for all remote codes ................................
```
## The same config compiles and runs flawlessly if Version: 1.12.2 is used

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:

  - platform: remote_receiver
    id: code1
    #name: "Button 1"
    on_press:
      then:
        - switch.toggle: relay1
    filters:
      - delayed_off: 500ms
    raw: [-2665, 857, -479, 400, -477, 404, -475, 836, -484, 850, -910, 398, -473, 424, -454, 405, -473, 402, -478, 404, -474, 416, -462, 425, -454, 399, -478, 399, -482, 396, -485, 396, -477, 400, -478, 407, -470, 404, -477, 398, -480, 400, -477, 401, -477, 402, -476, 400, -926, 10000]

```

Do let me know if you require the complete yaml file

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

darshkpatel, 2019-06-09:
The beta branch seems to work fine for me 

`docker pull esphome/esphome:beta`
in case anyone else has the same problem.

darshkpatel, 2019-06-09:
[#601](https://github.com/esphome/esphome/pull/601)
[#578](https://github.com/esphome/esphome/pull/578)

might be the culprit

OttoWinter, 2019-06-14:
On second look, this is the same as https://github.com/esphome/issues/issues/439 - will merge the discussion there

closed 2019-06-14 by OttoWinter