esphome/issues #421: State topic option ignored for custom cover

definitio, 2019-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python virtual environment

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/cover/custom.html

**Description of problem:**
Custom cover publishes state to `<TOPIC_PREFIX>/<COMPONENT_TYPE>/<COMPONENT_NAME>/position/state`, that can't be changed via `state_topic` option, it does nothing.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
cover:
  - platform: custom
    lambda: |-
      auto custom_cover = new CustomCover();
      App.register_component(custom_cover);
      return {custom_cover};
    covers:
      - name: Cover
        state_topic: <room>/<device>/state
        command_topic: <room>/<device>/set
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-10:
state_topic is for the OPEN/CLOSED state - this is more of a feature request for new `position_state_topic` and `position_command_topic`, which can of course be added.

definitio, 2019-06-10:
Should I create an issue in feature-requests repository?

OttoWinter, 2019-07-11:
Yes.

closed 2019-07-11 by OttoWinter