esphome/issues #422: Cannot install lib

ezar, 2019-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

LibraryManager

**Description of problem:**
LibraryManager: Installing Roomba
Downloading...
Error: Can not unpack file '/data/roomba/.piolibdeps/_tmp_installing-7MVJnf-package/Roomba-1.4.zip?dl=0'

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: roomba
  platform: ESP8266
  board: esp01_1m
  includes:
    - includes/ESPHomeRoombaComponent.h
  libraries:
    - Roomba=https://www.dropbox.com/s/hpesby96dhceyhw/Roomba-1.4.zip?dl=0
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

ezar, 2019-06-09:
If I use http://www.airspayce.com/mikem/arduino/Roomba/Roomba-1.4.zip appears a 406 error :(

ezar, 2019-06-09:
I install using local file. I dont know why doesnt works using remote url.
  libraries:
    - file:///config/esphome/includes/Roomba-1.4.zip

closed 2019-06-09 by ezar

OttoWinter, 2019-06-10:
It's platformio's format - I don't think that supports installing from dropbox links.