esphome/issues #423: Custom Sensor

ezar, 2019-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

Sensor

**Description of problem:**
class RoombaDistanceSensor : public sensor::Sensor
{
  public:
    RoombaDistanceSensor(const std::string &name) : sensor::Sensor(name) {}
    std::string unit_of_measurement() override { return "mm"; }
    int8_t accuracy_decimals() override { return 1; }
};

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
custom_component:
 - lambda: |-
    auto custom_roomba = new RoombaComponent(
        "Roomba Distance",
        "Roomba Voltage",
        "Roomba Current",
        "Roomba Charge",
        "Roomba Capacity",
        "Roomba Charging",
        "Roomba Docked",
        "Roomba Cleaning",
        "Roomba On",
        "Roomba Off",
        "Roomba Dock",
        "Roomba Locate",
        "Roomba Spot Clean",
        "roomba/command", /* command_topic */
        4,                /* BRC Pin */
        5000              /* Update Interval */
      );
    App.register_component(custom_roomba);
    App.register_sensor(custom_roomba->distance_sensor);
    App.register_sensor(custom_roomba->voltage_sensor);
    App.register_sensor(custom_roomba->current_sensor);
    App.register_sensor(custom_roomba->charge_sensor);
    App.register_sensor(custom_roomba->capacity_sensor);
    App.register_binary_sensor(custom_roomba->chargingState_sensor);
    App.register_binary_sensor(custom_roomba->dockedState_sensor);
    App.register_binary_sensor(custom_roomba->cleaningState_sensor);
    App.register_switch(custom_roomba->on_switch);
    App.register_switch(custom_roomba->off_switch);
    App.register_switch(custom_roomba->dock_switch);
    App.register_switch(custom_roomba->locate_switch);
    App.register_switch(custom_roomba->spot_switch);
    return {custom_roomba};
```

**Traceback (if applicable):**

```
In file included from src/main.cpp:12:0:
src/ESPHomeRoombaComponent.h:118:37: error: 'sensor' has not been declared
class RoombaDistanceSensor : public sensor::Sensor
^
src/ESPHomeRoombaComponent.h:118:45: error: expected '{' before 'Sensor'
class RoombaDistanceSensor : public sensor::Sensor
^
src/ESPHomeRoombaComponent.h:119:1: error: invalid type in declaration before '{' token
{
^
src/ESPHomeRoombaComponent.h:120:3: error: expected primary-expression before 'public'
public:
^
src/ESPHomeRoombaComponent.h:120:3: error: expected '}' before 'public'
src/ESPHomeRoombaComponent.h:120:3: error: expected ',' or ';' before 'public'
src/ESPHomeRoombaComponent.h:122:39: error: virt-specifiers in 'unit_of_measurement' not allowed outside a class definition
std::string unit_of_measurement() override { return "mm"; }
^
src/ESPHomeRoombaComponent.h:123:32: error: virt-specifiers in 'accuracy_decimals' not allowed outside a class definition
int8_t accuracy_decimals() override { return 1; }
^
src/ESPHomeRoombaComponent.h:124:1: error: expected declaration before '}' token
};
^
*** [/data/roomba/.pioenvs/roomba/src/main.cpp.o] Error 1
```

**Additional information and things you've tried:**

OttoWinter, 2019-06-10:
The compile error is correct.

You don't have any `sensor:` line in your configuration, so the sensor base is never loaded.

You can either add an empty `sensor:` block in your config or use the custom sensor platform to register the sensors.

closed 2019-06-10 by OttoWinter

ezar, 2019-06-10:
Yes! You're the best @OttoWinter! I add a sensor: block ;)
I will migrate to custom sensor platform!
Regards,