esphome/issues #426: homeassistant binary sensor toggles on home assistant restart

glmnet, 2019-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/binary_sensor/homeassistant.html

**Description of problem:**
The switch receives unexpected updates on esphome which causes my lights to turn off and back on when home assistant restarts.

Note, there is **another** issue which happens when the esphome itself restarts, I sent a PR https://github.com/esphome/esphome/pull/632

**Problem-relevant YAML-configuration entries:**
````yaml
binary_sensor:
  - platform: homeassistant
    name: "Switch Galeria 2 from HA"
    entity_id: binary_sensor.switch_estudio_galeria_2
    on_state:
      then: 
        - light.toggle: galeria
  - platform: homeassistant
    name: "Switch Galeria 3 from HA"
    entity_id: binary_sensor.switch_estudio_galeria_3
    on_state:
      then: 
        - script.execute: toggle_proyectores
````

**Traceback (if applicable):**

```
[19:41:49][D][api:572]: Client 'Home Assistant 0.94.1 (10.0.1.206)' connected successfully!
[19:41:50][D][homeassistant.binary_sensor:023]: 'binary_sensor.switch_estudio_galeria_2': Got state OFF
[19:41:50][D][binary_sensor:033]: 'Switch Galeria 2 from HA': Sending state OFF
[19:41:50][D][light:248]: 'Galeria' Setting:
[19:41:50][D][light:257]:   State: OFF
[19:41:50][D][homeassistant.binary_sensor:023]: 'binary_sensor.switch_estudio_galeria_3': Got state OFF
[19:41:51][D][homeassistant.binary_sensor:019]: 'binary_sensor.switch_estudio_galeria_2': Got state ON
[19:41:51][D][binary_sensor:033]: 'Switch Galeria 2 from HA': Sending state ON
[19:41:51][D][light:248]: 'Galeria' Setting:
[19:41:51][D][light:257]:   State: ON
```

**Additional information and things you've tried:**

So, with this setup, before HA restarts `binary_sensor.switch_estudio_galeria_2` is `ON`, and  `binary_sensor.switch_estudio_galeria_3` is `OFF`.
HA sends updates of `OFF` and `ON` for `binary_sensor.switch_estudio_galeria_2`
Note that those binary sensors are standard binary_sensors in another ESPHome node, and therefore HA might not know the state of the sensor yet (I don't know which ESPHome connects first)

Also I believe the issue might be on the home assistant side of the code but I believe this is more appropriate place to post.

My best guess is that HA is sending `OFF` when the value is actually `unavailabe` but you might know better.

OttoWinter, 2019-06-10:
Please enable debug logging in HA (see logger section) and post what HA is sending to the ESP.

glmnet, 2019-06-10:
Well switch binary_sensor.switch_estudio_galeria_2 changes states:
* none 
* unavailable 
* off
* on

See detailed log, 
I filtered the log: eh_estudio has the binary_sensor.switch_estudio_galeria_2 
eh_patio subscribes to the binary_sensor (has the `platform: homeassistant`)

Even though off on events are timestamped at the same second the lights flickers.

I can send you the full log if needed, may be email or other medium?

````
2019-06-10 19:09:03 DEBUG (zeroconf-ServiceBrowser__esphomelib._tcp.local.) [homeassistant.components.zeroconf] Discovered new device eh_patio._esphomelib._tcp.local. {'host': '10.0.1.153', 'port': 6053, 'hostname': 'eh_patio.local.', 'type': '_esphomelib._tcp.local.', 'name': 'eh_patio._esphomelib._tcp.local.', 'properties': {'version': '1.13.0-dev', 'address': 'eh_patio.local'}}
2019-06-10 19:09:03 DEBUG (zeroconf-ServiceBrowser__esphomelib._tcp.local.) [homeassistant.components.zeroconf] Discovered new device eh_estudio._esphomelib._tcp.local. {'host': '10.0.1.151', 'port': 6053, 'hostname': 'eh_estudio.local.', 'type': '_esphomelib._tcp.local.', 'name': 'eh_estudio._esphomelib._tcp.local.', 'properties': {'version': '1.13.0-dev', 'address': 'eh_estudio.local'}}
2019-06-10 19:09:07 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.switch_estudio_galeria_2, old_state=None, new_state=<state binary_sensor.switch_estudio_galeria_2=unavailable; friendly_name=Switch estudio galeria 2, device_class= @ 2019-06-10T19:09:07.289219-03:00>>
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Connecting to eh_patio.local:6053 (('10.0.1.153', 6053))
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Connecting to eh_estudio.local:6053 (('10.0.1.151', 6053))
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Opened socket for
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Opened socket for
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Sending <class 'api_pb2.HelloRequest'>: client_info: "Home Assistant 0.94.1"
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Write: 00 17 01 0A 15 48 6F 6D 65 20 41 73 73 69 73 74 61 6E 74 20 30 2E 39 34 2E 31
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Sending <class 'api_pb2.HelloRequest'>: client_info: "Home Assistant 0.94.1"
2019-06-10 19:09:07 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Write: 00 17 01 0A 15 48 6F 6D 65 20 41 73 73 69 73 74 61 6E 74 20 30 2E 39 34 2E 31
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Got message of type <class 'api_pb2.HelloResponse'>: api_version_major: 1
server_info: "eh_patio (esphome v1.13.0-dev)"
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.HelloResponse'>: api_version_major: 1
server_info: "eh_estudio (esphome v1.13.0-dev)"
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Successfully connected ('eh_patio (esphome v1.13.0-dev)' API=1.1)
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Sending <class 'api_pb2.ConnectRequest'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Write: 00 00 03
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Successfully connected ('eh_estudio (esphome v1.13.0-dev)' API=1.1)
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Sending <class 'api_pb2.ConnectRequest'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Write: 00 00 03
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Got message of type <class 'api_pb2.ConnectResponse'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.ConnectResponse'>: 
2019-06-10 19:09:08 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to eh_patio.local
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Sending <class 'api_pb2.DeviceInfoRequest'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Write: 00 00 09
2019-06-10 19:09:08 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to eh_estudio.local
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Sending <class 'api_pb2.DeviceInfoRequest'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Write: 00 00 09
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Got message of type <class 'api_pb2.DeviceInfoResponse'>: name: "eh_patio"
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.DeviceInfoResponse'>: name: "eh_estudio"
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Sending <class 'api_pb2.ListEntitiesRequest'>: 
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Write: 00 00 0B
2019-06-10 19:09:08 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Sending <class 'api_pb2.ListEntitiesRequest'>: 
[... cut ...]
2019-06-10 19:09:09 DEBUG (MainThread) [aioesphomeapi.connection] eh_patio.local: Got message of type <class 'api_pb2.BinarySensorStateResponse'>: key: 1757620387
2019-06-10 19:09:09 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.LightStateResponse'>: key: 1110042541
2019-06-10 19:09:09 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.SwitchStateResponse'>: key: 3957643408
2019-06-10 19:09:14 DEBUG (MainThread) [aioesphomeapi.connection] eh_estudio.local: Got message of type <class 'api_pb2.ClimateStateResponse'>: key: 2366735579
````

OttoWinter, 2019-06-11:
@glmnet Hmm I think I see what the issue is now. When Home Assistant restarts, the state is as follows (basically explanation of your list above)

 - no state (entity does not exist yet/hasn't been registered)
 - unavailable (entity has been recovered from HA data storage, but is not connected yet)
 - off (ESPHome device connects to HA, available gets set to True again and a hass publish state is scheduled (before the binary sensor data arrives))
 - on (ESPHome device sends initial state of the binary sensor)

Not sure how to fix this - It's a bigger issue with the esphome integration in ha/ha itself with scheduling state updates. The thing you can probably do is use binary sensor filters to filter out short bursts of wrong data (delayed on and off with really small values applied after each other).

glmnet, 2019-06-11:
Ok. let's keep this issue open then, only the `delayed_off` is needed (when it is off already this is not an issue) I wonder if it would be easy to obey the `delayed_off` only when HA has been offline shortly before. May be with an slaved templated binary sensor + status and timers... hmm not so easy, but may be worth it, it's blazing fast already.

glmnet, 2019-06-13:
Well, in case somebody needs a temporal fix for this, the `delayed_off` was not an option as sometimes it happens very long time between the ON -> OFF transition and adding a filter with a large value would make the switch annoying.

I ended up using the _timer hidden in a script_ pattern:

````yaml

script:
  - id: api_just_connected
    then:
      - delay: 30s

- platform: status
  id: api_online
  on_press:
    then:
      - script.execute: api_just_connected

- platform: homeassistant
  name: "Switch from HA"
  entity_id: ...
  on_state:
    then: 
      if: 
        condition: 
          not: 
            script.is_running: api_just_connected
        then:
          light.toggle: light_id
````

glmnet, 2019-07-22:
This is causing me headaches, this affects more than home_assistant sensors on esphome device itself, in fact the problem is in home assistant esphome integration.

Basically binary sensors are on for a while then ESP home loses connection (who knows why) then state goes unavailable -> off ->on as researched before.
I have history bars greens they are painted with small gaps of gray then red and then a big green chunk again. 

This is causing some automations in HA to trigger randomly. (state changes from off to on I guess)

We definitely need to avoid assigned sensors their default value (off for binary sensors, but I guess numbers might be assigned 0) when esphome devices reconnects.

fah, 2019-07-22:
All that (what we can read in the log) looks to me as if the handling of the initial states is over complicated.
(Two entities trying to do the best, but fail in the end.)
Sometimes (in the history of programming) it is better to remove some parts of the code then trying to improve it.

Is it a solution to, by default, "do nothing" regarding the binary outputs (On events like reconnect and startup and ...)?

BTW: I short cut the relay of my sonoff pow r2 because of this issue. Now I reliable can measure the power consumption of my computers without repairing the disks every time. ;)

@OttoWinter, you are the best. From my side deep admiration for the idea and constant work you invest.

glmnet, 2019-07-23:
Well I’m doing more testing and I cannot replicate this issue. Works as expected transition is either
- off > unavailable > off
- on > unavailable > on

I see automations trigger when you setup the entity alone as state change trigger. 
Is unavailable an state itself? Is this designed this way?
I find no easy way to automate this. I have to add two triggers: on to off and off to on. So effectively filtering out unavailable state. All this might be ha stuff. Learning more. Please give feedback

OttoWinter, 2019-10-13:
> Is unavailable an state itself? Is this designed this way?

`unavailable` is a state, yes. In homeassistant, each entity has a `available` property that is called before writing any state to the state machine. If that returns `False`, the state is cleared and replaced with the string `unavailable`.

Anyway, I think I found a solution to delay the state update until the full state becomes available in the HA codebase. Will create PR and link it to here.

closed 2019-10-21 by OttoWinter