esphome/issues #427: Custom Component compile error when update from 1.12.0 to 1.13.5

netsnakecq, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP01-s
**Affected component:**

UARTDevice, mqtt::CustomMQTTDevice

**Description of problem:**
in tonecommand.h:
#include "esphome.h"

// part to asr 
class ToneCommand : public Component, public UARTDevice, public mqtt::CustomMQTTDevice {
  public:
    ToneCommand(UARTComponent *parent) : UARTDevice(parent) {}
    
    sensor::Sensor *index = new sensor::Sensor();
    text_sensor::TextSensor *service = new text_sensor::TextSensor();
    text_sensor::TextSensor *entity_id = new text_sensor::TextSensor();
    
    void setup() override {
      write_array(txdata[0],3); // asr init
      index->publish_state(0);

      subscribe("tone/command", &ToneCommand::on_message);
      subscribe_json("tone/json/command", &ToneCommand::on_json_message);
    }
    void loop() override {
      if(available()>0){
        delay(50);
        numdata = readBytes(buffer,3);
    ......

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  includes:
    - tonecommand.h
  name: tone_command
  platform: ESP8266
  board: esp01_1m

logger:
  baud_rate: 0

uart:
  - id: utonecommand
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
  - id: utonetts
    tx_pin: GPIO2
    baud_rate: 9600

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: !secret wifi_ssid1
    password: !secret wifi_password

ota:
  password: !secret ota_password

api:
#  password: !secret api_password
  services:
    - service: tonecommand
      variables:
        command: string
      then:
        - mqtt.publish:
            topic: 'tone/command'
            payload: !lambda 'return command;'

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  discovery: True
  discovery_retain: true

switch:
  - platform: uart
    name: "tone continue"
    id: stonecontinue
    uart_id: utonecommand
    data: [0xAB,0xAB,0x00]

  - platform: uart
    name: "tone tts"
    id: stonetts
    uart_id: utonetts
    data: '<I>18'

binary_sensor:
  - platform: gpio
    pin: GPIO0
    id: btntonecontrol
    name: "btnToneControl"
    internal: true
    filters:
      - invert:
    on_multi_click: 
    - timing: #Double Clicked
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at least 0.2s
      then:
        - switch.turn_on:
            id: stonestop
        - switch.turn_on:
            id: stonesleep
    - timing: #Single Long Clicked
        - ON for 3s to 5s
        - OFF for at least 0.5s
      then:
        - switch.turn_on:
            id: stonestop
        - delay: 00:00:01
        - switch.turn_on:
            id: stonecontinue
    - timing: #Single Short Clicked
        - ON for at most 2s
        - OFF for at least 0.5s
      then:
        - switch.turn_on:
            id: stonestop
        - delay: 00:00:01
        - switch.turn_on:
            id: stoneonce

sensor:
  - platform: wifi_signal
    name: "WiFi Signal_tone"
    update_interval: 60s
  - platform: custom
    lambda: |-
      auto tonecommand = new ToneCommand(id(utonecommand));
      App.register_component(tonecommand);
      return {tonecommand->index};
    sensors:
      - name: "command index"
        id: "commandindexid"

text_sensor:
  - platform: custom
    lambda: |-
      auto my_tone_command = new ToneCommand(id(utonecommand));
      App.register_component(my_tone_command);
      return {my_tone_command->service,my_tone_command->entity_id};
    text_sensors:
      - name: "tone service"
        id: "toneservice"
      - name: "tone entity_id"
        id: "tonentityid"
 ......
```

**Traceback (if applicable):**

```
Compiling .pioenvs/tone_command/libd05/ESP8266WiFi/WiFiClient.cpp.o
In file included from src/main.cpp:58:0:
src/tonecommand.h:4:56: error: expected class-name before ',' token
class ToneCommand : public Component, public UARTDevice, public MQTTClientComponent {
^
src/tonecommand.h:6:31: error: expected ')' before '*' token
ToneCommand(UARTComponent *parent) : UARTDevice(parent) {}
^
src/tonecommand.h:4:7: warning: direct base 'esphome::Component' inaccessible in 'ToneCommand' due to ambiguity [enabled by default]
class ToneCommand : public Component, public UARTDevice, public MQTTClientComponent {
^
src/tonecommand.h: In member function 'virtual void ToneCommand::setup()':
src/tonecommand.h:13:30: error: 'write_array' was not declared in this scope
write_array(txdata[0],3); // asr init
^
src/tonecommand.h: In member function 'virtual void ToneCommand::loop()':
src/tonecommand.h:20:20: error: 'available' was not declared in this scope
if(available()>0){
 ......
```

**Additional information and things you've tried:**

I tried examples in docs(custom UARTDevice), found same errors too, but in 1.12.1, 
it's none error and work fine

OttoWinter, 2019-06-10:
`public MQTTClientComponent` - that line is definitely not correct. You don't inherit from the MQTT client.

For the other stuff: write `uart::UARTComponent` instead of `UARTComponent` as a workaround. Or add `using namespace esphome::uart;` at the top of the file.

thomashermine, 2019-06-10:
Not sure if it's related, and I hope it can help, but I have the same issue with the custom component for [MAX7219 in dot grid config](https://github.com/ASMfreaK/esphome_max7219)

When compiling on 1.13.5 : 
````
In file included from src/main.cpp:10:0:
src/max7219grid.hpp:1:29: fatal error: esphome/defines.h: No such file or directory
compilation terminated.
*** [/data/reveil/.pioenvs/reveil/src/main.cpp.o] Error 1
````

It does compile on 1.12.2

netsnakecq, 2019-06-11:
According to OttoWinter's stuff, after adding 'namespace esphome::uart;', ok! thanks!
But another question: 
    1.12.2 have 'CustomMQTTDevice' component, 1.13.0 or above dropped it ??? I can't find out in 1.13.5!

OttoWinter, 2019-06-11:
@thomashermine Correct, `esphome/defines.h` now is called `esphome/core/defines.h` (see src/ directory of generated platformio project for more info).

@netsnakecq Hmm, I can't find it either; it must have been lost during the migration.

netsnakecq, 2019-06-13:
@OttoWinter，Are you some suggestion about  'CustomMQTTDevice' component above 1.13.0 ?

closed 2019-06-14 by OttoWinter