esphome/issues #428: Not receiving MQTT data

jsponz, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

MQTT

**Description of problem:**
I have a new project of a ESP32 as BLE gateway. I want to read the values of the Xiaomi MiJia BLE Sensors and send the values to my MQTT broker. 

I do receive the debug messages, but I don't get any standard messages under the endpoints:
temp/sensor/nicolastemperature/state, temp/sensor/nicolasbattery/state, temp/sensor/nicolashumidity/state

What's wrong? 

I am using the latest version 1.13.5

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: temp_gateway
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "myWifi"
  password: "myPasswod"

web_server:
  port: 80

mqtt: 
  broker: mqtt.myserver.com
  topic_prefix: temp

# Enable logging
logger:

ota:
  password: "changeme"

esp32_ble_tracker:
  scan_interval: 300s

binary_sensor:
  - platform: ble_presence
    mac_address: 3C:71:BF:B3:2A:8C
    name: "ESP32 BLE Tracker Google Home Mini"

sensor:
  - platform: xiaomi_mijia
    mac_address: 7A:80:8E:19:36:BA
    temperature:
      name: "NicolasTemperature"
    humidity:
      name: "NicolasHumidity"
    battery_level:
      name: "NicolasBattery"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-10:
Logs please.

Best would be verbose logs because they contain what is published + where.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter