esphome/issues #429: Time based cover not restore positions after shutdown

FreeLuck, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 
Home assistant 0.94.1 virtualenv

**ESP (ESP32/ESP8266, Board/Sonoff):**
 

ESP8266 nodemcuv2

**Affected component:**

Time-Based Cover

**Description of problem:**

Hello,
I have just updated to version 1.13.5 and configured a time based cover everything works but if you turn off the nodemcuv2 board when it is turned back on, home assistant loses its real position and is set at 50%.

**Problem-relevant YAML-configuration entries:**
```yaml

switch:
  - platform: gpio
    pin: 
      number: D2
      inverted: yes
    id: open_cover_switch
  - platform: gpio
    pin: 
      number: D3
      inverted: yes
    id: close_cover_switch

cover:
  - platform: time_based
    name: "Cover1"
    id: cover_1
    open_action:
     # cancel potential previous movement
      - switch.turn_off: close_cover_switch
      - switch.turn_on: open_cover_switch
    open_duration: 15s
    close_action:
     # cancel potential previous movement
      - switch.turn_off: open_cover_switch
      - switch.turn_on: close_cover_switch
    close_duration: 15s
    stop_action:
      - switch.turn_off: open_cover_switch
      - switch.turn_off: close_cover_switch

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-10:
That is correct, see `esp8266_restore_from_flash` for restoring from power loss.

closed 2019-06-10 by OttoWinter

FreeLuck, 2019-06-13:
Okay @OttoWinter I read and set esp8266_restore_from_flash to true but it keeps returning to 50%.

```
esphome:
  name: esp_cucina
  platform: ESP8266
  board: nodemcuv2
  arduino_version: latest      
  esp8266_restore_from_flash: true

switch:
  - platform: gpio
    pin: 
      number: D2
      inverted: yes
    id: open_cover_switch
    interlock: [close_cover_switch]
  - platform: gpio
    pin: 
      number: D3
      inverted: yes
    id: close_cover_switch
    interlock: [open_cover_switch]

cover:
  - platform: time_based
    name: "Cover1"
    id: cover_1
    open_action:
      - switch.turn_on: open_cover_switch
    open_duration: 15s
    close_action:
      - switch.turn_on: close_cover_switch
    close_duration: 15s
    stop_action:
      - switch.turn_off: open_cover_switch
      - switch.turn_off: close_cover_switch
```

I tried also with arduino_version: latest and is always the same, so I also tested with the old template cover configuration and does not save the last position, it always returns to closed. With the esphome version 1.12.2 this template cover configuration works perfectly while from 1.13.x it does not save.

```
esphome:
  name: `esp_cucina`
  platform: ESP8266
  board: nodemcuv2
  arduino_version: latest

switch:
  - platform: gpio
    pin: 
      number: D2
      inverted: yes
    id: open_cover_switch
  - platform: gpio
    pin: 
      number: D3
      inverted: yes
    id: close_cover_switch
  
cover:
  - platform: template
    name: "Cover1"
    id: cover_1
    open_action:
      # cancel potential previous movement
      - switch.turn_off: close_cover_switch
      # perform movement
      - switch.turn_on: open_cover_switch
      # wait until cover is open
      - delay: 15s
      # turn of relay to prevent keeping the motor powered
      - switch.turn_off: open_cover_switch
    close_action:
      - switch.turn_off: open_cover_switch
      - switch.turn_on: close_cover_switch
      - delay: 15s
      - switch.turn_off: close_cover_switch
    stop_action:
      - switch.turn_off: open_cover_switch
      - switch.turn_off: close_cover_switch
    optimistic: true
    
```

OttoWinter, 2019-06-14:
See https://github.com/esphome/issues/issues/424