esphome/issues #432: sonoff basic cannot connect to wifi

whxciotw, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker + Hassio 0.94.1
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic
**Affected component:**

**Description of problem:**
Sonoff cannot connect to wifi ssid

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: sideboard_light
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: Calmdown
  password: Comeongirl
  fast_connect: true  
  # manual_ip:
  #   static_ip: 192.168.19.88
  #   gateway: 192.168.19.1
  #   subnet: 255.255.255.0  
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
 - platform: gpio
   pin:
     number: GPIO0
     mode: INPUT_PULLUP
     inverted: True
   id: btn
   internal: true
   on_press:
     - switch.toggle: relay

switch:
 - platform: gpio
   name: sideboard_light
   icon: mdi:lightbulb-outline
   pin: GPIO12
   id: relay

status_led:
 pin:
   number: GPIO13
   inverted: true
```

**Traceback (if applicable):**

```
[01:42:05][W][wifi:394]: Timeout while connecting to WiFi.
[01:42:05][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:42:05][W][wifi_esp8266:354]: Event: Disconnected ssid='Calmdown' bssid=D8:C8:E9:97:85:21 reason='Association Leave'
[01:42:05][W][wifi:400]: Error while connecting to network.
[01:42:05][W][wifi:431]: Restarting WiFi adapter...
[01:42:10][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:42:40][W][wifi:394]: Timeout while connecting to WiFi.
[01:42:41][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:42:41][W][wifi_esp8266:354]: Event: Disconnected ssid='Calmdown' bssid=D8:C8:E9:97:85:21 reason='Association Leave'
[01:42:41][W][wifi:400]: Error while connecting to network.
[01:42:41][W][wifi:431]: Restarting WiFi adapter...
[01:42:46][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:42:58][W][wifi_esp8266:354]: Event: Disconnected ssid='Calmdown' bssid=D8:C8:E9:97:85:21 reason='Handshake Failed'
[01:42:59][W][wifi:400]: Error while connecting to network.
[01:42:59][W][wifi:431]: Restarting WiFi adapter...
[01:42:59][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:43:29][W][wifi:394]: Timeout while connecting to WiFi.
[01:43:34][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:43:34][W][wifi_esp8266:354]: Event: Disconnected ssid='Calmdown' bssid=D8:C8:E9:97:85:21 reason='Association Leave'
[01:43:34][W][wifi:400]: Error while connecting to network.
[01:43:34][W][wifi:431]: Restarting WiFi adapter...
[01:43:34][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:44:04][W][wifi:394]: Timeout while connecting to WiFi.
[01:44:09][I][wifi:164]: WiFi Connecting to 'Calmdown'...
[01:44:09][W][wifi_esp8266:354]: Event: Disconnected ssid='Calmdown' bssid=D8:C8:E9:97:85:21 reason='Association Leave'

```

**Additional information and things you've tried:**

redskinhu, 2019-06-11:
Hello

Try to remove binary_sensor component and test it. Yes it sounds weird but try it. 
See: https://github.com/esphome/issues/issues/187

smoothunit, 2019-06-14:
Exactly the same issue after upgrade. Manually flashed and older version and connected in 5 seconds.

whxciotw, 2019-06-14:
> Hello
> 
> Try to remove binary_sensor component and test it. Yes, it sounds weird but try it.
> See: #187

i tried, but still doesn't work. what's more, if i connect sonoff basic via USB TTL(3.3v), it works fine to connect wifi, but failed when plug into 220V

whxciotw, 2019-06-14:
it works now, because I remove both binary sensor and status led, just like below.
but it is really weird. I hope this issue will be fixed in the future releases.
thanks.
`esphome:
  name: sideboard_light
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: Calmdown
  password: xxxxxxxxxx
  fast_connect: true  
  # manual_ip:
  #   static_ip: 192.168.19.88
  #   gateway: 192.168.19.1
  #   subnet: 255.255.255.0  
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
 - platform: gpio
   name: sideboard_light
   icon: mdi:lightbulb-outline
   pin: GPIO12
   id: relay

`

ultratoto14, 2019-06-18:
Had the same problem, only on a subset of my sonoffs. Only basic ones.
Have been fixed by removing the GPIO0 button from the config file. Keeping Status in binary sensor works for me:
`esphome:
  name: multiprise_bureau
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: HOMELAND
  password: xxxxxxxxx

api:

logger:

ota:

binary_sensor:
#  - platform: gpio
#    pin:
#      number: GPIO0
#      mode: INPUT_PULLUP
#      inverted: True
#    name: "Multiprise Bureau - Button"
#    on_press:
#      - switch.toggle: fakebutton
  - platform: status
    name: "Multiprise Bureau - Status"
`
With GPIO0 used, always No Network Found.

ultratoto14, 2019-06-18:
Just a small addition, worked on 1.13.4 fails after,  1.13.5 and 1.13.6

ultratoto14, 2019-06-18:
Checked the differrences between 1.13.4 and 1.13.5.
When reverting this wifi fix [#627](https://github.com/esphome/esphome/pull/627) i'm able to connect again even if the GPIO0 is set in config file.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter