esphome/issues #433: Time not working, are DSTless timezones supported?

glmnet, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266
**Affected component:**

https://esphome.io/components/time.html

**Description of problem:**
Yaml compilation fails, see stack trace and comments below.

**Problem-relevant YAML-configuration entries:**
```yaml
time:
- platform: homeassistant
```

**Traceback (if applicable):**

```
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/home/guillermo/.local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/home/guillermo/git/esphome/esphome/__main__.py", line 529, in main
    return run_esphome(sys.argv)
  File "/home/guillermo/git/esphome/esphome/__main__.py", line 512, in run_esphome
    config = read_config(args.verbose)
  File "/home/guillermo/git/esphome/esphome/config.py", line 803, in read_config
    res = load_config()
  File "/home/guillermo/git/esphome/esphome/config.py", line 663, in load_config
    return _load_config()
  File "/home/guillermo/git/esphome/esphome/config.py", line 651, in _load_config
    result = validate_config(config)
  File "/home/guillermo/git/esphome/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/home/guillermo/git/esphome/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/home/guillermo/.local/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/home/guillermo/.local/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/home/guillermo/git/esphome/esphome/voluptuous_schema.py", line 106, in validate_mapping
    key_value_map[key.schema] = key.default()
  File "/home/guillermo/git/esphome/esphome/components/time/__init__.py", line 112, in detect_tz
    return convert_tz(tz)
  File "/home/guillermo/git/esphome/esphome/components/time/__init__.py", line 87, in convert_tz
    utcoffset_on, _, tzname_on = transition_info[idx1]
IndexError: list index out of range
```

**Additional information and things you've tried:**
Platform seems not to be the issue, tried them all 3 test1 / 2 / 3.yaml files
My time zone is GMT-3 and we do not observe daylight savings.

brandond, 2019-06-10:
See https://github.com/esphome/esphome/pull/586 - what version are you using?

glmnet, 2019-06-10:
commit 36f47ade70351e8d5ddbc6ec608b3092427f54bb (HEAD -> dev, origin/dev, origin/HEAD)
Author: Otto Winter <otto@otto-winter.com>
Date:   Sun Jun 9 17:03:51 2019 +0200

    Add Captive Portal (#624)
    
....

brandond, 2019-06-10:
Probably a dupe of this one then https://github.com/esphome/issues/issues/378#issuecomment-500219634

glmnet, 2019-06-10:
Ahhh yes, looks like I should move to another country then

OttoWinter, 2019-06-10:
Duplicate of #378

closed 2019-06-10 by OttoWinter