esphome/issues #434: platform: time - error of compilation

armacali, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 HassOS 2.12

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff TH

**Affected component:**
https://esphome.io/components/time.html

**Description of problem:**
the automation function with "time", does not allow compiling the programs

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: luz_calle
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "AMCWIFI"
  password: "xxxxxxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

status_led:
  pin:
    number: 13
    inverted: True

switch:
  - platform: gpio
    pin: 12
    name: "Luz Calle"
    id: "LuzCalle"
   
sensor:
  - platform: uptime
    name: "Luz Calle uptime" 

time:
  - platform: homeassistant
    id: "arielles"
```

**Traceback (if applicable):**

```
INFO Reading configuration...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 466, in run_esphome
    config = read_config(args.verbose)
  File "/opt/esphome/esphome/config.py", line 803, in read_config
    res = load_config()
  File "/opt/esphome/esphome/config.py", line 663, in load_config
    return _load_config()
  File "/opt/esphome/esphome/config.py", line 651, in _load_config
    result = validate_config(config)
  File "/opt/esphome/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 106, in validate_mapping
    key_value_map[key.schema] = key.default()
  File "/opt/esphome/esphome/components/time/__init__.py", line 112, in detect_tz
    return convert_tz(tz)
  File "/opt/esphome/esphome/components/time/__init__.py", line 87, in convert_tz
    utcoffset_on, _, tzname_on = transition_info[idx1]
IndexError: list index out of range
```

**Additional information and things you've tried:**

brandond, 2019-06-10:
https://github.com/esphome/esphome/pull/586

armacali, 2019-06-10:
I have the latest version of ESPHome (1.13.5), and it's still not compiled

OttoWinter, 2019-06-10:
Duplicate of #378

closed 2019-06-10 by OttoWinter