esphome/issues #435: ESP32 camera "setup failed" after

roybosch, 2019-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on a VM
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-S Camera (AI THINKER board)
**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**
Since the OTA update to 1.13.5 the ESP32 cam doesn't show any pictures or stream anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: espcam_voordeur
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xx"
  password: "xx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xx"

ota:
  password: "xx"

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Image settings
  name: espcam_voordeur
  max_framerate: 5 fps
  resolution: 640x480
  jpeg_quality: 10

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/espcam_voordeur
Processing espcam_voordeur (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Retrieving maximum program size /data/espcam_voordeur/.pioenvs/espcam_voordeur/firmware.elf
Checking size /data/espcam_voordeur/.pioenvs/espcam_voordeur/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.5% (used 40812 bytes from 327680 bytes)
PROGRAM: [======    ]  55.8% (used 914286 bytes from 1638400 bytes)
========================= [SUCCESS] Took 7.74 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of espcam_voordeur.local
INFO  -> xx
INFO Uploading /data/espcam_voordeur/.pioenvs/espcam_voordeur/firmware.bin (914400 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from espcam_voordeur.local using esphome API
WARNING Error resolving IP address of espcam_voordeur.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 1 seconds
INFO Connecting to espcam_voordeur.local:6053 (192.168.178.100)
INFO Successfully connected to espcam_voordeur.local
[22:49:23][I][app:096]: esphome version dev compiled on Jun 10 2019, 22:31:53
[22:49:23][C][wifi:372]: WiFi:
[22:49:23][C][wifi:254]:   SSID: [redacted]
[22:49:23][C][wifi:255]:   IP Address: xx
[22:49:23][C][wifi:257]:   BSSID: [redacted]
[22:49:23][C][wifi:258]:   Hostname: 'espcam_voordeur'
[22:49:23][C][wifi:262]:   Signal strength: -40 dB ▂▄▆█
[22:49:23][C][wifi:263]:   Channel: 11
[22:49:23][C][wifi:264]:   Subnet: 255.255.255.0
[22:49:23][C][wifi:265]:   Gateway: xx
[22:49:23][C][wifi:266]:   DNS1: xx
[22:49:23][C][wifi:267]:   DNS2: xx
[22:49:23][C][logger:137]: Logger:
[22:49:23][C][logger:138]:   Level: DEBUG
[22:49:23][C][logger:139]:   Log Baud Rate: 115200
[22:49:23][C][logger:140]:   Hardware UART: UART0
[22:49:23][C][esp32_camera:043]: ESP32 Camera:
[22:49:23][C][esp32_camera:044]:   Name: espcam_voordeur
[22:49:23][C][esp32_camera:045]:   Board Has PSRAM: YES
[22:49:23][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[22:49:23][C][esp32_camera:048]:   VSYNC Pin: 25
[22:49:23][C][esp32_camera:049]:   HREF Pin: 23
[22:49:23][C][esp32_camera:050]:   Pixel Clock Pin: 22
[22:49:23][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[22:49:23][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[22:49:23][C][esp32_camera:053]:   Reset Pin: -1
[22:49:23][C][esp32_camera:074]:   Resolution: 640x480 (VGA)
[22:49:23][E][esp32_camera:091]:   Setup Failed: ERROR
[22:49:23][C][ota:029]: Over-The-Air Updates:
[22:49:23][C][ota:030]:   Address: espcam_voordeur.local:3232
[22:49:23][C][ota:032]:   Using Password.
[22:49:23][C][api:103]: API Server:
[22:49:23][C][api:104]:   Address: espcam_voordeur.local:6053
[22:51:10][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

**Additional information and things you've tried:**
I've tried uploading again OTA. Everytime it says "Setup Failed" even though uploading went well.

FlashGordon7, 2019-06-11:
I see the same thing and when revetring back to 1.13.4 the camera works again.

OttoWinter, 2019-06-11:
Only thing in 1.13.5 that changed was that a small package upgrade of the ESP SDK core was downgraded again because of other issues the newer version had.

You can use the 1.13.4 version again with this:

```yaml
esphome:
  arduino_version: 1.0.2
```

(before it was 1.0.1)

FlashGordon7, 2019-06-12:
Just tested and it works. Thanks :-)

roybosch, 2019-06-12:
For me this didn't work. I lost the connection with the espcam after adding arduino version 1.0.2.

I think I have to flash the binary file again now.

solidminds, 2019-06-12:
Same problem here with a ESP32 Cam Aithinker board. Running 1.13.5 and added 'arduino version 1.0.2' to the config.
I found that I get and image when running at default resolution (640x480) but when i set it higher cam stops working.
Stops working as in, upload goes wo problems, but no image in Homeassistant. 

I only see the 'Got Image.....' logging below when resolution set to 640x480:
```
[18:42:22][D][esp32_camera:157]: Got Image: len=23474
[18:42:32][D][esp32_camera:157]: Got Image: len=23665
[18:42:32][D][esp32_camera:157]: Got Image: len=23618
```

redpizza69, 2019-06-13:
Same problem with ESP32 Cam AIThinker board running 1.13.5. No joy with adding "arduino version 1.0.2". I even changed the Resolution to see if that would help.

[17:44:22][C][esp32_camera:043]: ESP32 Camera:
[17:44:22][C][esp32_camera:044]:   Name: My Camera
[17:44:22][C][esp32_camera:045]:   Board Has PSRAM: YES
[17:44:22][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[17:44:22][C][esp32_camera:048]:   VSYNC Pin: 25
[17:44:22][C][esp32_camera:049]:   HREF Pin: 23
[17:44:22][C][esp32_camera:050]:   Pixel Clock Pin: 22
[17:44:22][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[17:44:22][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[17:44:22][C][esp32_camera:053]:   Reset Pin: -1
[17:44:22][C][esp32_camera:068]:   Resolution: 320x240 (QVGA)
[17:44:22][E][esp32_camera:091]:   Setup Failed: ERROR

FlashGordon7, 2019-06-13:
Try to change the board to esp32dev 
I run with that on 3pcs ESP32-cam Aithinker board and it works.
```
esphome:
  name: esp_cam_1
  platform: ESP32
  board: esp32dev
  arduino_version: 1.0.2

wifi:
  ssid: "iot"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xxx'
  reboot_timeout: 0s

ota:
  password: 'xxx'
  
# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: espcam1
  resolution: 1024x768
  max_framerate: 4 fps
  idle_framerate: 0 fps

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4

light:
  - platform: binary
    output: gpio_4
    name: esp_cam_1
```

redpizza69, 2019-06-14:
I cut and pasted your exact yaml (except for wifi, api, ota), and camera still doesn't work.

I do get these messages in the compile though:

INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp_cam_1
Processing esp_cam_1 (platform: espressif32@1.8.0; framework: arduino; board: esp32dev)
--------------------------------------------------------------------------------
LibraryManager: Installing id=1826 @ 1.0.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/api_message.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/api_server.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/command_messages.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/list_entities.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/service_call_message.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/subscribe_logs.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/user_services.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/api/util.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/esp32_camera/esp32_camera.cpp.o
src/esphome/components/esp32_camera/esp32_camera.cpp: In member function 'virtual void esphome::esp32_camera::ESP32Camera::dump_config()':
src/esphome/components/esp32_camera/esp32_camera.cpp:54:10: warning: enumeration value 'FRAMESIZE_QXGA' not handled in switch [-Wswitch]
switch (this->config_.frame_size) {
^
src/esphome/components/esp32_camera/esp32_camera.cpp:54:10: warning: enumeration value 'FRAMESIZE_INVALID' not handled in switch [-Wswitch]
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/light/addressable_light.cpp.o
Compiling /data/esp_cam_1/.pioenvs/esp_cam_1/src/esphome/components/light/light_state.cpp.o

and more....

CapnJimbo, 2019-06-15:
hi there..

I tried it too use the following

> Try to change the board to esp32dev
> I run with that on 3pcs ESP32-cam Aithinker board and it works.
> 
> ```
> esphome:
>   name: esp_cam_1
>   platform: ESP32
>   board: esp32dev
>   arduino_version: 1.0.2
> ```

when flashing completes following appears on the 'ESPHOMEFLASHER'

Wrote 933360 bytes (504751 compressed) at 0x00010000 in 44.8 seconds (effective 166.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[12:39:17]ets Jun  8 2016 00:22:57
[12:39:17]
[12:39:17]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:39:17]configsip: 0, SPIWP:0xee
[12:39:17]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:39:17]mode:DIO, clock div:2
[12:39:17]load:0x3fff0018,len:4
[12:39:17]load:0x3fff001c,len:952
[12:39:17]load:0x40078000,len:6084
[12:39:17]load:0x40080000,len:7944
[12:39:17]entry 0x40080310
[12:39:17]user code done

but my problem is the the ESP32 ai thinker wont connect to the wifi. I have a ttgo cam that connects to wifi and 2 ESP8266 that works and integrates to home assistant.

(tried with Arduino, camera web server - that was able to connect to wifi on the same module/ but i'd prefer a ESPHome option)

currently running Home Assistant 0.93.2 on a RPi3b+
ESPHome - You are currently running version 1.13.5

any assistance much appreciated
thank you in advance

RedGuilhas, 2019-06-17:
> hi there..
>
> I tried it too use the following
>
>
> when flashing completes following appears on the 'ESPHOMEFLASHER'
>
> Wrote 933360 bytes (504751 compressed) at 0x00010000 in 44.8 seconds (effective [...]

Any luck? I'm getting the same result as you. My AI Thinker won't connect to wifi with this status on the logs:

[16:14:14]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:14:14]configsip: 0, SPIWP:0xee
[16:14:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:14:14]mode:DIO, clock div:2
[16:14:14]load:0x3fff0018,len:4
[16:14:14]load:0x3fff001c,len:952
[16:14:14]load:0x40078000,len:6084
[16:14:14]load:0x40080000,len:7944
[16:14:14]entry 0x40080310
[16:14:14]user code done

roybosch, 2019-06-17:
Same here, I can't connect the board to Wifi with the latest firmware. 

Would be nice if we're able to choose the firmware version ourselves to prevent waiting for a solution. A bit like the library manager of the Arduino IDE.

FlashGordon7, 2019-06-17:
If you have a backup from the time you had the old version installed, you can restore that version (don't have to restore anything else, just uncheck everything except ESPHome) then remember to disable auto update if enabled.

roybosch, 2019-06-18:
How is this done? Using the snapshot function?

RedGuilhas, 2019-06-22:
> I'm running on 1.13.4 again :)

How did you install that version?

Thanks

Electronlibre2012, 2019-06-22:
Hello,

same boat,

i upgrade to 1.13.6 and i want to add a camera and in esphomeflasher i'm stuck to "user ocde done", not connecting...

Is there a solution please @OttoWinter ?

my log 
```
Showing logs:
[03:33:19]ets Jun  8 2016 00:22:57
[03:33:19]
[03:33:19]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[03:33:19]configsip: 0, SPIWP:0xee
[03:33:19]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[03:33:19]mode:DIO, clock div:2
[03:33:19]load:0x3fff0018,len:4
[03:33:19]load:0x3fff001c,len:952
[03:33:19]load:0x40078000,len:6084
[03:33:19]load:0x40080000,len:7944
[03:33:19]entry 0x40080310
[03:33:19]user code done
```

thanks!

Electronlibre2012, 2019-06-23:
Up please,

i have done what you suggest above and still not working :

> 
> 

> You can use the 1.13.4 version again with this:
> 
> ```yaml
> esphome:
>   arduino_version: 1.0.2
> ```
> 
> (before it was 1.0.1)

OttoWinter, 2019-06-23:
@Electronlibre2012 You flashed the device wrong - probably not through ESPHome or ESPHome-Flasher. Search this tracker for `SPI_FAST_FLASH_BOOT` and you will find the answer. I will hide the comments as they are off-topic to this issue.

Electronlibre2012, 2019-06-23:
Hello, thanks for your reply, but i flash it with your recommanded tool esphome flasher…i have download the last version 1.01…same issue

I have change Nothing in my flashing mode…compile Under ESPhome on my hassio pi, download the binary file, open esphomeFlasher and flash…that all…

I will look to your indication, but i dont understand why not say directly what is wrong for you…

Therefore, thanks for your response !

Good night

Provenance : Courrier pour Windows 10

De : Otto Winter
Envoyé le :lundi 24 juin 2019 00:07
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

@Electronlibre2012 You flashed the device wrong - probably not through ESPHome or ESPHome-Flasher. Search this tracker for SPI_FAST_FLASH_BOOT and you will find the answer. I will hide the comments as they are off-topic to this issue.

Electronlibre2012, 2019-06-23:
Hello, 2 hours later…same point…nothing found i understand with spi_fast_flash_boot….

Excuse me but i use your tool, and dont change anything that i have made before to flash the esp32cam : compile in esphome, download binary, flash with esphomeflasher…4 espcam are working untill 2 month…

Can you help please, i saw a lot of poor guys like me in the same situation, a lot of post…cant make it work until the 1.13.5 update of EspHome.

Deseperate…i am.

Provenance : Courrier pour Windows 10

De : Otto Winter
Envoyé le :lundi 24 juin 2019 00:07
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

@Electronlibre2012 You flashed the device wrong - probably not through ESPHome or ESPHome-Flasher. Search this tracker for SPI_FAST_FLASH_BOOT and you will find the answer. I will hide the comments as they are off-topic to this issue.

roybosch, 2019-06-24:
> > I'm running on 1.13.4 again :)
> 
> How did you install that version?
> 
> Thanks

I recovered a snapshot. Uncheck everything but ESPHome 1.13.4 and then click restore.

I think this is non-related to this topic, you can make a new topic for it.

Electronlibre2012, 2019-06-24:
Thanks! But i have no made a snapshot...

Dont understand why Otto say i have another problem because i have change
nothing : compile with esphome on HA, download bynaries and flash with
EspHomeFlasher v1.0.1...

Electronlibre2012, 2019-06-24:
Hello @OttoWinter 

i read the doc and :
SPI_FAST_FLASH_BOOT - This is the normal SPI flash boot mode.

when i flash : 
`
Using 'COM19' as serial port.
Showing logs:
[13:02:51]ets Jun  8 2016 00:22:57
[13:02:51]
[13:02:51]rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
[13:02:51]waiting for download
`
sorry i search everywhere...perhaps my English is not good enough but i understand nothing, found nothing about "user code done"...

Please could you explain a little so i can perhaps find where is my problem. I just compile with espHome in Ha, i dont understand what i can doing wrong...there is no option, just the board and i have no error at all when i compile with EspHome.

This is my config:

```
esphome:
  name: terrasse
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xxxxx"
  password: "xxxxxxx"
  manual_ip:
    static_ip: 192.168.1.48
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger: 
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "Popol97"
  
# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  vertical_flip: false
  horizontal_mirror: false
  resolution: 640x480 

  name: terrasse
  # ... Frame Settings
  
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Terrasse"
    update_interval: 60s
    id: signalWt

switch:
  - platform: restart
    name: "Restart Cam Terrasse"
  
binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: "Terrasse PIR Sensor"
    device_class: motion

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: $hostname lights
```

i try to add
`board: esp32dev
 arduino_version: 1.0.2`

but doesnt work better.

Please, help :)

roybosch, 2019-06-24:
> Thanks! But i have no made a snapshot... Dont understand why Otto say i have another problem because i have change nothing : compile with esphome on HA, download bynaries and flash with EspHomeFlasher v1.0.1...

I guess it's the same problem, but seen from another perspective: our existing installation got "corrupt" by updating OTA. Your new installation is corrupt by flashing the same version of the code. 

You should try reverting back to 3.14. If you don't have a snapshot, I believe it's possible to do it via some commandline. I don't know how that's done.

OttoWinter, 2019-06-24:
@Electronlibre2012 @roybosch That's a separate issue - this here is solely about "setup failed" message, not the SPI_FAST_FLASH_BOOT or similar. You can open a new issue about it, and include everything about how you flashed the device: what tool you used, what pins you held during upload etc.

Electronlibre2012, 2019-06-24:
Thanks Roy , but i run hassio not HA...and command line in hassio are not
complete at all...just a few options...

solidminds, 2019-06-24:
Maybe totally off-topic, but have you tried to use the 'EspEasy flash tool' and flash the included 'blank' firmware to overwrite the complete memory of your board. After that, just flash Esphome like you used to the first time.
I always 'rescue' my boards this way. Works with nodemcu, wemos, etc....

Electronlibre2012, 2019-06-24:
Hello, thanks i try right now !

Provenance : Courrier pour Windows 10

De : Solidminds
Envoyé le :lundi 24 juin 2019 19:52
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

Maybe totally off-topic, but have you tried to use the 'EspEasy flash tool' and flash the included 'blank' firmware to overwrite the complete memory of your board. After that, just flash Esphome like you used to the first time.
I always 'rescue' my boards this way. Works with nodemcu, wemos, etc....

Electronlibre2012, 2019-06-24:
Sorry , is it ESP easy Flasher ?

Have you got a link please, there is plenty

Provenance : Courrier pour Windows 10

De : Solidminds
Envoyé le :lundi 24 juin 2019 19:52
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

Maybe totally off-topic, but have you tried to use the 'EspEasy flash tool' and flash the included 'blank' firmware to overwrite the complete memory of your board. After that, just flash Esphome like you used to the first time.
I always 'rescue' my boards this way. Works with nodemcu, wemos, etc....

solidminds, 2019-06-24:
It's a tool by LetsControlIt.
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Get_started
I must say that I haven't used it with my Ai Thinker boards yet. But maybe it just works.

Electronlibre2012, 2019-06-24:
Thanks i try it !

Provenance : Courrier pour Windows 10

De : Solidminds
Envoyé le :lundi 24 juin 2019 20:10
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

It's a tool by LetsControlIt.
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Get_started
I must say that I haven't used it with my Ai Thinker boards yet. But maybe it just works.

Electronlibre2012, 2019-06-24:
I dont think its my board…i have plenty…like 12…lol

But i try it and i will tell you

Provenance : Courrier pour Windows 10

De : Solidminds
Envoyé le :lundi 24 juin 2019 20:10
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

It's a tool by LetsControlIt.
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Get_started
I must say that I haven't used it with my Ai Thinker boards yet. But maybe it just works.

Electronlibre2012, 2019-06-24:
Its for esp8266, not for esp32,

Tell me if i am wrong ?

Provenance : Courrier pour Windows 10

De : Solidminds
Envoyé le :lundi 24 juin 2019 20:10
À : esphome/issues
Cc : Electronlibre2012; Mention
Objet :Re: [esphome/issues] ESP32 camera "setup failed" after update to1.13.5 (#435)

It's a tool by LetsControlIt.
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Get_started
I must say that I haven't used it with my Ai Thinker boards yet. But maybe it just works.

OttoWinter, 2019-06-24:
@Electronlibre2012 I've specifically said not to post here anymore for the issue you're seeing and using a new issue for that. There have been a total of 5 (!) messages from you after the most recent warning. I will delete all those comments - and consider this a final warning (for the sake of the maintainers and all people subscribed to this thread)

daenny, 2019-06-30:
Is there any progress on this? I tried compiling under 1.13.4 (using a standalone docker) but i still get the "setup failed" error :(

harryd100, 2019-07-18:
For what it is worth, using Esphome 1.13.6 results in the same error. I used to have one ESP32-cam (AI Thinker) working with Esphome

harryd100, 2019-07-21:
Did some more investigation and found out that the ESP32 AI Thinker clones are of very bad quality (soldering) See: https://community.home-assistant.io/t/for-those-struggling-to-get-esp32-cam-clone-ai-thinker-module-working/127517

After re-soldering the esp32 pins, both modules work again also with 1.13.6

Electronlibre2012, 2019-08-11:
Hello,

i have the same problem ```[Esp32 camera:091]: Setup Failed: ERROR``` and for me i just replace the camera module and it works...
That was a brand new one, just out of the antistatic sealed bag...a faulty cam board sure...seems to be a scratch on the pastic wires...

Electronlibre2012, 2019-08-14:
Update : an another new ESP32-cam has the same problem, and i test the previous camera lens on a working ESP32-cam, flashed with EspHome, same result, same error.
So i take an old camera lens OV2640 and it works...that mean something has changed in the camera  lens provided with the ESP32-cam board from bangood...

Work with no error with a fisheye lens too

daenny, 2019-08-15:
I also changed the camera module in my board and this resolved my issues. I had bought 2 AI-Thinker (clone) modules from an ali-express vendor. Apparently the quality control is not great on these boards. I will investigate if I can get the other board working with the tips from @harryd100

jr-k, 2019-11-01:
> Did some more investigation and found out that the ESP32 AI Thinker clones are of very bad quality (soldering) See: [...]

You're right ! It's a nightmare to fix the solder but now it works ! Thanks.

heckler, 2019-11-01:
Let me relay my experience here, in case it helps others with the same issue:

I have two identical ESP32Cam modules out of Aliexpress, [SKU 32992663411](https://aliexpress.com/item/32992663411.html).  One of them started failing consistently with the `Setup Failed: ERROR` message.

I experimented with removing the camera module, carefully cleaning the contacts in the flat cable and re-attaching the module. Turned it back on (same firmware), and it started working just fine again, with no changes in software.

iz3man, 2020-03-15:
> I experimented with removing the camera module, carefully cleaning the contacts in the flat cable and re-attaching the module. Turned it back on (same firmware), and it started working just fine again, with no changes in software.

Damnit, that helped me as well. Had the same error. Opened the camera connector, removed cable, reinserted, closed it, and the camera works. Thanks!!

stanislav-brabec, 2020-11-24:
In my case, the problem was caused by uploading a sketch with ledc driver without channel directive. See #619 for more.

It seems that the damage could be permanent. You can try to upload fixed sketch. Maybe it will fix the internal hardware registers, and camera will start to work again.

doudz, 2021-02-16:
I got the same problem after trying to use ledc driver on GPIO4.
It works at first but after a while I have the same error on two AI Thinker boards
The problem is still here even  after removing the led control
`[E][esp32_camera:093]:   Setup Failed: ERROR
`
I try with esphome 1.15.* and 1.16.*

But the camera is still working good using CameraWebServer sample from Arduino IDE
So I suppose there's really a problem with the esphome code ?

stanislav-brabec, 2021-02-16:
@doudz 
Probably this makes the difference:
```
    config.ledc_channel = LEDC_CHANNEL_0;
```

It reprograms ledc channel 0 back to the camera.

doudz, 2021-02-16:
@stanislav-brabec ok, but why I'm still getting the error when switching back to esphome without led configuration ? I only keep the esp32_camera component

doudz, 2021-02-16:
Should I configure output component with ledc driver to channel 0, then flash, then remove it and setup esp32_camera again ?

stanislav-brabec, 2021-02-16:
@doudz The above channel 0 code refers to ESP32-CAM camera driver. If you configure ledc output component with channel 0, you cannot use camera. Camera driver reserves channel 0 and 1. You should configure ledc output to channel 2.

I think that the change is reboot-resistent and poweroff-resistent. But I am not sure. I did not study the datasheet and the bootloader in deep.

doudz, 2021-02-17:
I manage to make the board work again by :
- Flashing CameraWebServer sketch from Arduino IDE
- Rebooting twice
- Flashing ESPHome binary (with esp32_camera component and GPIO4 as switch, no more ledc driver)
- Rebooting twice (at first reboot, the error was still here)
and now it works !

ausycb, 2021-06-23:
I had same issue after updating to ESPHome 1.19.3
Changed arduino_version to really old release and then it worked again
esphome:
  arduino_version: 1.0.2

pszeremeta86, 2021-06-30:
The same here! Camera stopped working!

stecy, 2021-07-10:
Me too, the camera just stopped !

MarcodeFederizzi, 2021-07-29:
Same problem after updating.
Follwing ausycb solution worked again.
esphome:
arduino_version: 1.0.5
Thanks

SaturnusDJ, 2021-07-31:
Something to try. First I used the 5V and the ground next to it. Once I moved the ground as far away from 5V as possible, which in my case literally is the pin furthest away, the error appears a lot less.

SaturnusDJ, 2021-08-01:
Never mind. Still lots of problems.

reflexsa, 2021-09-08:
I seem to have similar issues on 2021.8.2.
Camera worked for a while but after uploading new code it never worked again, even after reverting to the 'working' code.
I've tried removing the cable and reinstalling as well as the older Arduino versions but nothing has rectified the problem as of yet.

brentbrooks70, 2021-09-08:
Running ESP32-CAM-MB which uses the AI Thunker pin assignments I was able to get it working by following 
@FlashGordon7 I changed esp32cam to esp32dev and also
@ausycb solution of changing the Arduino core to 1.05.
The limited capabilities this brings makes me think that just the Arduino CameraWebServer code and attaching the camera to MotionEye and setting the camera to http://_Camera-IP-Address:81/stream_ you can now let Motionye monitor for motion and capture when present. This seems a better solution than the ESPHome native API capabilities, maybe I missed something.

pixeldoc2000, 2021-11-06:
Might be related to #2224

doudz, 2021-11-14:
I finally solved the problem just by setting the "resolution" in esp32_camera component... without it fails to setup
```
esp32_camera:
  name: $name Camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: 640x480
```

miraixyz, 2021-11-24:
Just adding my 2 cents.  Had the same issue.  Camera worked perfectly until I upgraded.  Got no picture after upgrading. Added arduino_version: 1.0.5 to the yaml, and now it works again

closed 2023-01-19 by rradar