esphome/issues #436: MCP23017 hangs ESP8266

marci144, 2019-06-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-11:
The issue is that you're using the wrong pins for the i2c bus.

GPIO6 is used by the flash interface and will result in this error when used. ESPHome does already show a warning when compiling about this:

```
ESP8266: Pin 6 (6-11) might already be used by the flash interface. Be warned.
```

Double check what pins you have it attached to

closed 2019-06-11 by OttoWinter

marci144, 2019-06-11:
How stupid !!

It was a typo!!  I used pin 4 and 5 as usual !! But I wrote 5 and  in the sketch!

I never looked in that direction ! I change four different controllers.

OK tks, problem solved