esphome/issues #437: uploading fw to change to manual IP fails

bwanajag, 2019-06-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io 0.94.1, ESPHome 1.13.5
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (TTGO camera board) and ESP8266 (HYLETON 313 outlet)
**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
After initially flashing (via USB) both the ESP32 and ESP8266, the firmware doesn't have a manual IP configured.  When I edit the yaml file and configure manual IP settings and then try to upload (OTA), ESPHome tries to connect to the manual IP that was just added to the yaml file instead of the current dynamic IP.  I'm unable to find a way to change from dynamic to manual IP using OTA programming.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: hall_monitor
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "ssid"
  password: "password"
  manual_ip:
    static_ip: 10.50.0.47
    gateway: 10.50.0.1
    subnet: 255.255.255.0
    dns1: 10.50.0.1
    dns2: 10.50.0.1   

```

**Traceback (if applicable):**

```
INFO Successfully compiled program.
INFO Connecting to 10.50.0.47
ERROR Connecting to 10.50.0.47:3232 failed: [Errno 113] No route to host
```

**Additional information and things you've tried:**

For the ESP8266 that failed to update, I used tuya-convert to upload the ESPHome binary file.  I have no way (without breaking the device open) to connect via USB.

OttoWinter, 2019-06-11:
See `use_address`

closed 2019-06-11 by OttoWinter

kluszczyn, 2019-06-11:
I'm using `--upload-port 192.168.5.6` to specify ip address to upload fw (see <https://esphome.io/guides/cli.html>