esphome/issues #438: OTA fw upgrade from 1.12.2 to 1.13.5  - devices are not coming back online

ddean6232, 2019-06-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO Home Assistant 0.94.1, ESPHOME add-on v 1.13.5

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic, NodeMCU, WEMOS

**Affected component:**

**Description of problem:**
Upgraded ESPHOME add-on from 1.12.2 to 1.13.5 with no problems
No configuration changes to any .YAML files (included all relevant files below)
Upgraded Sonoff Basic from 1.12.2 to 1.13.5 via OTA and has not come back online, powered off for a hard reboot, still not coming back online.
Same for NodeMCU board
Same for WEMOS board.

**Problem-relevant YAML-configuration entries:**
```yaml
SONOFF BASIC
esphome:
  name: sonoff4964d4
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'xxxx'
  password: 'xxxx'
  manual_ip:
    static_ip: 192.168.0.53
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

#web_server:
#  port: 80

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "4964D4 Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Bedroom Lamp 4964D4"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

WEMOS D1 MINI
esphome:
  name: minitemp
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'xxxx'
  password: 'xxxxx'
  manual_ip:
    static_ip: 192.168.0.63
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

#web_server:
#  port: 80

dallas:
  - pin: D1
    update_interval: 5s

sensor:
  - platform: dallas
    address: 0xAB031554A2E2FF28
    name: "Thermo Temp"
    id: thermo

switch:
  - platform: gpio
    name: "Thermo Relay"
    pin: D7
    id: relay

time:
  - platform: homeassistant
    id: mytime
    on_time:
      - seconds: 0,10,20,30,40,50
        then:
          - display.page.show_next: my_display
          - component.update: my_display
font:
  - file: "OpenSans-Regular.ttf"
    id: my_font
    size: 14

image:
  - file: "logo.png"
    id: my_image
    resize: 128x32

i2c:
  sda: D5
  scl: D4
  scan: True

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    address: 0x3C
    id: my_display
    pages:
      - id: page1
        lambda: |-
          it.image(0, 0, id(my_image));
      # - id: page2
      #   lambda: |-
      #     it.print(0, 10, id(my_font), "This is page 2");
      - id: page3
        lambda: |-
         // Time
          it.printf(0, 10, id(my_font), "Relay: %s", id(relay).state ? "ON" : "OFF");
          it.strftime(85, 0, id(my_font), "%H:%M", id(mytime).now());    
         // LINE, draw a line from X start, Y start, X stop, Y stop
          it.line(0, 18, 128, 18);
          it.printf(0, 26, id(my_font), "Temp: %0.1fC", id(thermo).state);

NODEMCU
esphome:
  name: multisensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'xxxx'
  password: 'xxxx'
  manual_ip:
    static_ip: 192.168.0.61
    gateway: 192.168.0.1
    subnet: 255.255.255.0

api:

# Enable logging
logger:

ota:

sensor:
  - platform: dht
    model: DHT11
    pin: D7
    temperature:
      name: "Multisensor Temperature"
    humidity:
      name: "Multisensor Humidity"
    update_interval: 15s

output:
  - platform: esp8266_pwm
    pin: D1
    id: redgpio
  - platform: esp8266_pwm
    pin: D2
    id: greengpio
  - platform: esp8266_pwm
    pin: D3
    id: bluegpio

light:
  - platform: rgb
    name: "Multisensor Light"
    red: redgpio
    green: greengpio
    blue: bluegpio
   
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-11:
Most likely a duplicate of one of these issues: https://github.com/esphome/issues/issues?utf8=%E2%9C%93&q=label%3A%22Connection+Issue%22+

Please get USB logs off of the device, there's no way to help otherwise.

ddean6232, 2019-06-11:
RollBack to 1.13.3 and devices now working.  Something in the WiFi code probably

ddean6232, 2019-06-11:
where is the documentation to "get the USB logs off the device"  I dont know how to do that...

OttoWinter, 2019-06-12:
@ddean6232 Oh just connect the device via USB and view the serial monitor logs - like when you initially flashed it.

Boboman1, 2019-06-12:
@ddean6232 Having a similar problem like you, I removed the "logger:" component from the .yaml file, flashed the new fw and the sonoff basics appeared online and are working fine after 48h.

ddean6232, 2019-06-12:
@Boboman1 were you able to OTA upgrade after turning off "logger"

ddean6232, 2019-06-12:
SONOFF BASIS LOGGER VERBOSE OUTPUT,  FAST_CONNECT: ON
'''
Using 'COM15' as serial port.
Showing logs:
[11:33:12]state: 0 -> 2 (b0)
[11:33:12]state: 2 -> 3 (0)
[11:33:12]state: 3 -> 5 (10)
[11:33:12]add 0
[11:33:12]aid 13
[11:33:12]cnt 
[11:33:22]pm open,type:0 0
[11:33:22]state: 5 -> 0 (2)
[11:33:22]rm 0
[11:33:22]pm close 7
[11:33:22][W][wifi_esp8266:354]: Event: Disconnected ssid='*****' bssid=0C:80:63:0D:4F:33 reason='Handshake Failed'
[11:33:22]wifi evt: 1
[11:33:22]STA disconnect: 204
[11:33:22][W][wifi:400]: Error while connecting to network.
[11:33:22][W][wifi:431]: Restarting WiFi adapter...
[11:33:22][V][wifi_esp8266:035]: Disabling STA.
[11:33:22]del if0
[11:33:22]usl
[11:33:22]mode : null
[11:33:22][V][wifi_esp8266:392]: Event: Changed Mode old=STA new=OFF
[11:33:22]wifi evt: 8
[11:33:22][I][wifi:164]: WiFi Connecting to '*****'...
[11:33:22][V][wifi:166]: Connection Params:
[11:33:22][V][wifi:167]:   SSID: '*****'
[11:33:22][V][wifi:172]:   BSSID: Not Set
[11:33:22][V][wifi:174]:   Password: '*********'
[11:33:22][V][wifi:178]:   Channel: Not Set
[11:33:22][V][wifi:184]:   Manual IP: Static IP=192.168.0.53 Gateway=192.168.0.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[11:33:22][V][wifi:188]:   Hidden: NO
[11:33:22][V][wifi_esp8266:033]: Enabling STA.
[11:33:22]mode : sta(84:f3:eb:49:64:d4)
[11:33:22]ip:192.168.0.53,mask:255.255.255.0,gw:192.168.0.1
[11:33:22]add if0
[11:33:22][V][wifi_esp8266:392]: Event: Changed Mode old=OFF new=STA
[11:33:22]wifi evt: 8
[11:33:23][V][wifi_esp8266:360]: Event: Changed AuthMode old=WPA PSK new=WPA2 PSK
[11:33:23]wifi evt: 2
[11:33:23][V][wifi_esp8266:360]: Event: Changed AuthMode old=WPA2 PSK new=WPA PSK
[11:33:23]wifi evt: 2
[11:33:25]scandone
[11:33:25]state: 0 -> 2 (b0)
[11:33:25]state: 2 -> 3 (0)
[11:33:25]state: 3 -> 5 (10)
[11:33:25]add 0
[11:33:25]aid 13
[11:33:25]cnt 
[11:33:35]pm open,type:0 0
[11:33:35]state: 5 -> 0 (2)
[11:33:35]rm 0
[11:33:35]pm close 7
[11:33:35][W][wifi_esp8266:354]: Event: Disconnected ssid='*****' bssid=0C:80:63:0D:4F:33 reason='Handshake Failed'
[11:33:35]wifi evt: 1
[11:33:35]STA disconnect: 204
[11:33:35][W][wifi:400]: Error while connecting to network.
[11:33:35][W][wifi:431]: Restarting WiFi adapter...
[11:33:35][V][wifi_esp8266:035]: Disabling STA.
[11:33:35]del if0
[11:33:35]usl
[11:33:35]mode : null
[11:33:35][V][wifi_esp8266:392]: Event: Changed Mode old=STA new=OFF
[11:33:35]wifi evt: 8
[11:33:41][I][wifi:164]: WiFi Connecting to '*****'...
[11:33:41][V][wifi:166]: Connection Params:
[11:33:41][V][wifi:167]:   SSID: '*****'
[11:33:41][V][wifi:172]:   BSSID: Not Set
[11:33:41][V][wifi:174]:   Password: '*********'
[... 490 lines cut ...]
[11:36:55][V][wifi:172]:   BSSID: Not Set
[11:36:55][V][wifi:174]:   Password: '*********'
[11:36:55][V][wifi:178]:   Channel: Not Set
[11:36:55][V][wifi:184]:   Manual IP: Static IP=192.168.0.53 Gateway=192.168.0.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[11:36:55][V][wifi:188]:   Hidden: NO
[11:36:55][V][wifi_esp8266:033]: Enabling STA.
[11:36:55]mode : sta(84:f3:eb:49:64:d4)
[11:36:55]ip:192.168.0.53,mask:255.255.255.0,gw:192.168.0.1
[11:36:55]add if0
[11:36:55][V][wifi_esp8266:392]: Event: Changed Mode old=OFF new=STA
[11:36:55]wifi evt: 8
[11:36:55][V][wifi_esp8266:360]: Event: Changed AuthMode old=WPA PSK new=WPA2 PSK
[11:36:55]wifi evt: 2
[11:36:55][V][wifi_esp8266:360]: Event: Changed AuthMode old=WPA2 PSK new=WPA PSK
[11:36:55]wifi evt: 2
[11:36:57]scandone
[11:36:57]state: 0 -> 2 (b0)
[11:36:57]state: 2 -> 3 (0)
[11:36:58]state: 3 -> 5 (10)
[11:36:58]add 0
[11:36:58]aid 13
[11:36:58]cnt 
[11:37:00]
Serial port closed!
'''

ddean6232, 2019-06-12:
This SONOFF BASIC worked great before upgrade from 1.12.2 to 1.13.5

Boboman1, 2019-06-12:
Yes, OTA updates are working.

ddean6232, 2019-06-12:
Removed "logger" and it came right back online... Thanks @Boboman1 

@OttoWinter  Memory problem??

ddean6232, 2019-06-12:
WEMOS D1 and NODEMCU seem to be OK... just have to wait an extra long time....

Boboman1, 2019-06-12:
You're welcome! I tested a WEMOS D1mini with logger: enabled and its working good. No problems for more than 24h.

ddean6232, 2019-06-12:
Tried SONOFF POW doing OTA upgrade with no logger 1.12.2 to 1.13.5....  after upgrade never came back online.  I have about 20 SONOFFs... don't want to pull them all to do a hard upgrade...  any progress on what the problem is?

ddean6232, 2019-06-12:
Hopefully I will not have to upgrade any SONOFFs... only wired upgrade works without logger... like I said I dont want to have to pull 20 units for a hard upgrade!!!!

ddean6232, 2019-06-12:
Should announce that 1.13.5 does NOT work on SONOFF devices... @OttoWinter

OttoWinter, 2019-06-13:
@ddean6232 That is incorrect. 1.13.5 does work on Sonoff devices, I have some myself and it's working fine. Because the resolution is the same as with some other connection issues, I will mark this as a duplicate of the other ones https://github.com/esphome/issues/issues?utf8=%E2%9C%93&q=label%3A%22Connection+Issue%22+

TL;DR there's a bug in espressif's SDK that causes really weird things with networks - and doing random stuff fixes it. Almost as if a change in binary size or whatever is making things fail. Worst of all: I still can't reproduce this issue.

closed 2019-06-13 by OttoWinter

OttoWinter, 2019-06-18:
Moving the discussion to #455