esphome/issues #440: globals.set action does not respect variable type

caelor, 2019-06-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip (esphome-1.13.5)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

globals

**Description of problem:**
Using the globals.set action with an int type global results in a compile error

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_globalsset_bug
  platform: ESP8266
  board: nodemcuv2

logger:

globals:
  - id: test1
    type: int

script:
  - id: test_script
    then:
      - globals.set:
          id: test1
          value: '1'
```

**Traceback (if applicable):**

```
$ esphome bugreport.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d test_globalsset_bug
Processing test_globalsset_bug (framework: arduino; platform: espressif8266@1.8.0; board: nodemcuv2)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/test_globalsset_bug/src/main.cpp.o
Compiling .pioenvs/test_globalsset_bug/lib9a9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/test_globalsset_bug/lib9a9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/test_globalsset_bug/lib9a9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
In file included from src/esphome/core/automation.h:5:0,
from src/esphome/components/globals/globals_component.h:4,
from src/esphome.h:2,
from src/main.cpp:3:
src/esphome/core/helpers.h: In instantiation of 'esphome::TemplatableValue<T, X>::TemplatableValue(F) [with F = const char*; typename std::enable_if<(! esphome::is_callable<F, X ...>::value), int>::type <anonymous> = 0; T = int; X = {}]':
src/esphome/components/globals/globals_component.h:59:3:   required from 'void esphome::globals::GlobalVarSetAction<C, Ts>::set_value(V) [with V = const char*; C = esphome::globals::GlobalsComponent<int>; Ts = {}]'
src/main.cpp:56:44:   required from here
src/esphome/core/helpers.h:202:57: error: invalid conversion from 'const char*' to 'int' [-fpermissive]
TemplatableValue(F value) : type_(VALUE), value_(value) {}
^
*** [.pioenvs/test_globalsset_bug/src/main.cpp.o] Error 1
================================================================================================================= [ERROR] Took 2.26 seconds =================================================================================================================
```

**Additional information and things you've tried:**
Inclusion of initial_value doesn't seem to make any difference.

brandond, 2019-06-12:
You've quoted the '1' which makes it a string. What do you get if you remove the quotes, which makes the YAML parser treat it as an int?

caelor, 2019-06-12:
The config fails to parse, as the value must be a string (which the documentation confirms for both the globals.set action, and the globals definitions):

```
INFO Reading configuration...
Failed config

script:
  - [source bugreport.yaml:13]
    id: test_script
    then:
      - [source bugreport.yaml:15]
        globals.set:  [source bugreport.yaml:16]
          id: test1

          Must be string, got <class 'esphome.yaml_util.ESPInt'>. did you forget putting quotes around the value?.
          value: 1 [source bugreport.yaml:17]
```

caelor, 2019-06-13:
As a workaround, you can use a lambda to set the global in the meantime, e.g. replace

```
- globals.set:
    - id: test1
      value: '1'
```
with
```
- lambda: |-
    id(test1) = 1;
```

closed 2019-06-14 by OttoWinter