esphome/issues #441: Missing update of adc sensor with Home Assistant

leofalp, 2019-06-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io intsalled on Ubuntu Server 18.04.2 LTS

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU v3
**Affected component:**

Platform acd update

**Description of problem:**
I have a little problem with a NodeMCUv3 with ESPHome connected to Home Assistant through WiFi.

The configuration is simple: i have a Dallas sensor for the temperature and all is powered from little solar panel and 18650 battery. so i need Deep Sleep to preserve the battery and i'd like to monitor the battery voltage so i've link the + of battery with A0 Pin with a resistor. 

All works fine! There's only the problem of updating value to home hassistant. The temperature is perfect but the voltage updates whenever it wants, maybe after days.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp5
  platform: ESP8266
  board: nodemcuv2
  
deep_sleep:
  run_duration: 120s
  sleep_duration: 9min

wifi:
  ssid: "GuaiFaiMini"
  password: "**"

logger:

api:
  password: '**'

ota:
  password: '**'

dallas:
  - pin: D2

sensor:
  - platform: dallas
    address: 0x1602131B06C8AA28
    name: "Temperatura esterna"
    id: "temp_ext"
  
  - platform: adc
    pin: A0
    name: "Tensione batteria esp5"
    id: "batt_esp5"
    update_interval: 5s
    filters:
      - multiply: 4.2
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-13:
Connect the device via USB and attach serial logs of this happening to the issue.

leofalp, 2019-06-14:
here it is:
```
[0;36m[D][wifi:272]: Starting scan...[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=4294961770 (now=239)[0m
[0;36m[D][adc:056]: 'Tensione batteria esp5': Got voltage=1.00V[0m
[0;37m[V][sensor:013]: 'Tensione batteria esp5': Received new state 1.000000[0m
[0;38m[VV][sensor.filter:013]: Filter(0x3fff13d4)::input(1.000000)[0m
[0;38m[VV][sensor.filter:020]: Filter(0x3fff13d4)::output(4.200000) -> SENSOR[0m
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=60000 last_execution=4294894217 (now=275)[0m
[0;38m[VV][component:075]: set_timeout(name='0x1602131B06C8AA28', timeout=750)[0m
[0;38m[VV][component:128]: Running timeout '0x1602131B06C8AA28':1 with interval=750 last_execution=289 (now=1040)[0m
[0;38m[VV][dallas.sensor:245]: Scratch pad: 3D.01.4B.46.7F.FF.7F.10.D5 (96)[0m
[1;31m[E][dallas.sensor:248]: Reading scratchpad from Dallas Sensor failed[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=4294966770 (now=4475)[0m
[0;36m[D][adc:056]: 'Tensione batteria esp5': Got voltage=1.00V[0m
[0;37m[V][sensor:013]: 'Tensione batteria esp5': Received new state 1.000000[0m
[0;38m[VV][sensor.filter:013]: Filter(0x3fff13d4)::input(1.000000)[0m
[0;38m[VV][sensor.filter:020]: Filter(0x3fff13d4)::output(4.200000) -> SENSOR[0m
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m
scandone
[0;36m[D][wifi:287]: Found networks:[0m
[0;32m[I][wifi:322]: - 'GuaiFaiMini' [5m(E4:95:6E:43:23:37) [6m[0;32m▂▄▆█[0m[0m
[0;36m[D][wifi:323]:     Channel: 2[0m
[0;36m[D][wifi:324]:     RSSI: -38 dB[0m
[0;36m[D][wifi:326]: - [5m'RLNet-681'[6m [5m(CC:2D:E0:DF:6C:A1) [6m[0;33m▂▄▆[0;37m█[0m[0m
[0;32m[I][wifi:164]: WiFi Connecting to 'GuaiFaiMini'...[0m
[0;37m[V][wifi:166]: Connection Params:[0m
[0;37m[V][wifi:167]:   SSID: 'GuaiFaiMini'[0m
[0;37m[V][wifi:170]:   BSSID: E4:95:6E:43:23:37[0m
[0;37m[V][wifi:174]:   Password: [5m'***'[6m[0m
[0;37m[V][wifi:176]:   Channel: 2[0m
[0;37m[V][wifi:186]:   Using DHCP IP[0m
[0;37m[V][wifi:188]:   Hidden: NO[0m
[0;37m[V][wifi_esp8266:360]: Event: Changed AuthMode old=OPEN new=WPA2 PSK[0m
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with GuaiFaiMini, channel 2
dhcp client start...
[0;37m[V][wifi_esp8266:345]: Event: Connected ssid='GuaiFaiMini' bssid=E4:95:6E:43:23:37 channel=2[0m
wifi evt: 0
ip:192.168.8.110,mask:255.255.255.0,gw:192.168.8.1
[0;37m[V][wifi_esp8266:366]: Event: Got IP static_ip=192.168.8.110 gateway=192.168.8.1 netmask=255.255.255.0[0m
wifi evt: 3
[0;32m[I][wifi:380]: WiFi connected![0m
[0;35m[C][wifi:254]:   SSID: [5m'GuaiFaiMini'[6m[0m
[0;35m[C][wifi:255]:   IP Address: 192.168.8.110[0m
[0;35m[C][wifi:257]:   BSSID: [5mE4:95:6E:43:23:37[6m[0m
[0;35m[C][wifi:258]:   Hostname: 'esp5'[0m
[0;35m[C][wifi:262]:   Signal strength: -41 dB [0;32m▂▄▆█[0m[0m
[0;35m[C][wifi:263]:   Channel: 2[0m
[0;35m[C][wifi:264]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:265]:   Gateway: 192.168.8.1[0m
[0;35m[C][wifi:266]:   DNS1: 192.168.8.1[0m
[0;35m[C][wifi:267]:   DNS2: 0.0.0.0[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: esp5.local:8266[0m
[0;35m[C][ota:032]:   Using Password.[0m
[0;35m[C][api:030]: Setting up Home Assistant API server...[0m
[0;35m[C][deep_sleep:013]: Setting up Deep Sleep...[0m
[0;38m[VV][component:075]: set_timeout(name='', timeout=120000)[0m
[0;32m[I][app:060]: setup() finished successfully![0m
[0;32m[I][app:096]: esphome version dev compiled on Jun 14 2019, 12:04:45[0m
[... 72 lines cut ...]
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m
[0;38m[VV][api:598]: on_ping_request_[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=19474 (now=24490)[0m
[0;36m[D][adc:056]: 'Tensione batteria esp5': Got voltage=1.00V[0m
[0;37m[V][sensor:013]: 'Tensione batteria esp5': Received new state 1.000000[0m
[0;38m[VV][sensor.filter:013]: Filter(0x3fff13d4)::input(1.000000)[0m
[0;38m[VV][sensor.filter:020]: Filter(0x3fff13d4)::output(4.200000) -> SENSOR[0m
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m
[0;38m[VV][api:598]: on_ping_request_[0m
[0;38m[VV][api:598]: on_ping_request_[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=24474 (now=29488)[0m
[0;36m[D][adc:056]: 'Tensione batteria esp5': Got voltage=1.00V[0m
[0;37m[V][sensor:013]: 'Tensione batteria esp5': Received new state 1.000000[0m
[0;38m[VV][sensor.filter:013]: Filter(0x3fff13d4)::input(1.000000)[0m
[0;38m[VV][sensor.filter:020]: Filter(0x3fff13d4)::output(4.200000) -> SENSOR[0m
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m
[0;38m[VV][api:598]: on_ping_request_[0m
[0;38m[VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=29474 (now=34482)[0m
[0;36m[D][adc:056]: 'Tensione batteria esp5': Got voltage=1.00V[0m
[0;37m[V][sensor:013]: 'Tensione batteria esp5': Received new state 1.000000[0m
[0;38m[VV][sensor.filter:013]: Filter(0x3fff13d4)::input(1.000000)[0m
[0;38m[VV][sensor.filter:020]: Filter(0x3fff13d4)::output(4.200000) -> SENSOR[0m
[0;36m[D][sensor:092]: 'Tensione batteria esp5': Sending state 4.20000 V with 2 decimals of accuracy[0m

```

OttoWinter, 2019-06-14:
From the logs everything looks ok - the values are sent out properly, see `Sending state 4.20000 V`

What I think might be your issue is HA's state duplication filtering. If a state is equal to the last one, HA will not update the state (because it's the same and an update is not needed).

It looks like the ADC is permanently maxed out at 4.2V. It will only update if the value changes. This behavior is correct and expected.

leofalp, 2019-06-15:
Yes, i think that it could be a motivation but there's someting strange:
[image]
this is my graph with no change...
I don't think that battery is always at 4.2v...

maybe the solution could be this, in mqtt sensor description on home assistant but i don't know where i can add this:
```
force update

    (boolean)Sends update events even if the value hasn’t changed. Useful if you want to have meaningful value graphs in history.

    Default value: false
```
[source](https://www.home-assistant.io/components/sensor.mqtt/#force_update)

brandond, 2019-06-16:
The ADC is maxed out, in that it will never read higher than 4.2v. it looks like it is occasionally dropping below that, and updates correctly during those periods, but also spends a lot of time at 4.2v.

If you want to read higher than that you'll need to add some voltage divider resistors and then multiply the raw reading.

OttoWinter, 2019-06-17:
Yes, the ADC sensor is maxed out. That's why you're seeing the behavior.

For force_update please see feature requests (explanation why it doesn't exist there too).

Closing.

closed 2019-06-17 by OttoWinter