esphome/issues #443: Two DS18B20s on one bus not working

jikama, 2019-06-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
Connecting two configured DS18B20 will freeze the NodeMcu with ESPHOME. If one of the sensors is disconnected, the other one is functional and I see the value.

dallas:

pin: D5
update_interval: 10s
sensor:
platform: dallas
address: 0x3F030997792F5828
name: "Temperature #1"
platform: dallas
address: 0xD703139779A28828
name: "Temperature #2"

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-06-12:
If it literally freezes and stops responding I suspect you have them wired wrong, aren't supplying enough voltage to the esp8266, or are overdrawing the voltage regulator on the nodemcu board. How do you have them connected? I've successfully run 6 of them on a single pin. You just need to be sure that you're powering them correctly and have pull-up resistors in the right places.

jikama, 2019-06-12:
The DS18B20 is connected in parallel with the resistance 4k7 between VDD and DQ. If in code:

Dallas:
pin: D5 
update_interval: 10s 

so they work and find out their address.

jikama, 2019-06-12:
It will not be powered if I write the code as follows:

dallas:
  - pin: D5
    update_interval: 10s
    id: dallas_1
  - pin: D1
    update_interval: 10s
    id: dallas_2
sensor:
  - platform: dallas
    address: 0x3F030997792F5828
    name: "Teplota zdroje 5V"
    dallas_id: dallas_1
  - platform: dallas
    address: 0xD703139779A28828
    name: "Teplota zdroje 12V"
    dallas_id: dallas_2

I can read the temperature from both DS18B20 without any problems.

OttoWinter, 2019-06-13:
@jjkama Working for me here. I agree with @brandond this sounds like an issue with your setup. If you connect the device via USB and view serial monitor logs during this issue, what do you see? Also please post a photo/schematic of the exact setup you have.

jikama, 2019-06-13:
The NodeMCU cannot be connected to the serial monitor continuously when two ds18B20 are connected to one GPIO.
The wiring diagram is in the picture.
The whole setting is:
esphome:
  name: system
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
# Example configuration entry

i2c:
  sda: D3
  scl: D4
  scan: True
dallas:
  - pin: D5
    update_interval: 10s
   
sensor:
  - platform: dallas
    address: 0x3F030997792F5828
    name: "Teplota zdroje 5V"
  - platform: dallas
    address: 0xD703139779A28828
    name: "Teplota zdroje 12V"

  - platform: ina219
    address: 0x41
    shunt_resistance: 0.1 ohm
    
    current:
      filters:
       - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 0.001 -> 0.1
      name: "Proud baterie"
      
    bus_voltage:
      filters:
       - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 13.264 -> 13.43
      name: "Napeti baterie"
    update_interval: 10s
  
  - platform: ccs811
    baseline: 0x1481
    eco2:
      name: "CCS811 eCO2 Value"
    tvoc:
      name: "CCS811 Total Volatile Organic Compound"
    address: 0x5A
    update_interval: 60s

[image]

OttoWinter, 2019-06-13:
> The NodeMCU cannot be connected to the serial monitor continuously when two ds18B20 are connected to one GPIO.

Why would that be? There's no reason I see for this in the schematic.

> The wiring diagram is in the picture.

It appears your configuration contains other stuff as well that are not part of the diagram, please disconnect those while testing the DS18b20 to rule out other stuff.

jikama, 2019-06-13:
Thank you all for your help, finally it turned out to be a bad NodeMcu board v0.9, with another it already works as it should. Thanks again.

closed 2019-06-13 by jikama