esphome/issues #445: Wrong time with UTC+4 timezone

definitio, 2019-06-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python virtual environment

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Sonoff Dual)

**Affected component:**

https://esphome.io/components/time.html

**Description of problem:**
Time is UTC, offset is ignored, e.g. when local time is 17:00, esphome has 13:00.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
time:
  - platform: ...
    timezone: Europe/Saratov
    ...
```

**Traceback (if applicable):**

```
INFO Detected timezone '+04' with UTC offset 4
```

**Additional information and things you've tried:**

theniz, 2019-06-20:
I have the same issue.
in my case, my timezone is ASIA/SEOUL, also UTC+9.

time on esphome always show UTC time though i put timezone or offset.
But strangely when i put UTC-9, it worked. I don't know why.
so, why don't you set your offset as UTC-4?

definitio, 2019-06-20:
Thanks, I will use UTC-4.

hpirila, 2019-06-25:
I have similar thing. Actions happen at UTC time even time zone is set and time zone can be seen in logs as well.

`
//# configuration
time:
  - platform: homeassistant
    timezone: Asia/Bangkok
    on_time:
      - seconds: 0
        **minutes: 46
        hours: 7**
        days_of_week: MON-FRI
        then:
          **- switch.turn_off: motionswitch**

# log
//# expect the switch.turn_off: motionswitch happen at 7:46 Asia/Bangkok time 
//# but it happens at 14:46 Asia/Bangkok time, which is 07:46 UTC time

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.2.102 using esphome API
INFO Connecting to 192.168.2.102:6053 (192.168.2.102)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.2.102: timed out). Re-Trying in 1 seconds
INFO Connecting to 192.168.2.102:6053 (192.168.2.102)
INFO Successfully connected to 192.168.2.102
[14:45:12][I][app:096]: **esphome version 1.13.6** compiled on Jun 25 2019, 14:44:19
[14:45:12][C][wifi:372]: WiFi:
[14:45:12][C][wifi:254]:   SSID: 

[redacted]
[14:45:12][C][wifi:255]:   IP Address: 192.168.2.102
[14:45:12][C][wifi:257]:   BSSID: 

[redacted]
[14:45:12][C][wifi:258]:   Hostname: 'upstairshalllight'
[14:45:12][C][wifi:262]:   Signal strength: -61 dB ?_?¦
[14:45:12][C][wifi:263]:   Channel: 3
[14:45:12][C][wifi:264]:   Subnet: 255.255.255.0
[14:45:12][C][wifi:265]:   Gateway: 192.168.2.1
[14:45:12][C][wifi:266]:   DNS1: 8.8.8.8
[14:45:12][C][wifi:267]:   DNS2: 8.8.4.4
[14:45:12][D][time:030]: Synchronized time: Tue Jun 25 07:45:12 2019
[14:45:12][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button'
[14:45:12][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[14:45:12][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Upstairs hall motion sensor"'
[14:45:12][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[14:45:12][C][gpio.binary_sensor:016]:   Pin: GPIO3 (Mode: INPUT)
[14:45:12][C][switch.gpio:042]: GPIO Switch 'Upstairs hall light'
[14:45:12][C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT)
[14:45:12][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[14:45:12][C][template.switch:058]: Template Switch 'timer1s'
[14:45:12][C][template.switch:059]:   Restore State: NO
[14:45:12][C][template.switch:060]:   Optimistic: YES
[14:45:12][C][switch.gpio:042]: GPIO Switch 'blinker'
[14:45:12][C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT)
[14:45:12][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[14:45:12][C][template.switch:058]: Template Switch 'motionofftimer'
[14:45:12][C][template.switch:059]:   Restore State: NO
[14:45:12][C][template.switch:060]:   Optimistic: YES
[14:45:12][C][template.switch:058]: Template Switch 'Upstairs hall 20 min timer'
[14:45:12][C][template.switch:059]:   Restore State: NO
[14:45:12][C][template.switch:060]:   Optimistic: YES
[14:45:12][C][template.switch:058]: Template Switch 'Upstairs hall 2 h timer'
[14:45:12][C][template.switch:059]:   Restore State: NO
[14:45:12][C][template.switch:060]:   Optimistic: YES
[14:45:12][C][template.switch:058]: Template Switch 'Upstairs hall motion sensor switch'
[14:45:12][C][template.switch:059]:   Restore State: NO
[14:45:12][C][template.switch:060]:   Optimistic: YES
[14:45:12][C][esp8266_pwm:022]: ESP8266 PWM:
[14:45:12][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[14:45:12][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[14:45:12][C][esp8266_pwm:025]:   Inverted: YES
[14:45:12][C][logger:137]: Logger:
[14:45:12][C][logger:138]:   Level: DEBUG
[14:45:12][C][logger:139]:   Log Baud Rate: 115200
[14:45:12][C][logger:140]:   Hardware UART: UART0
[14:45:12][C][light:174]: Light 'led'
[14:45:12][C][light:176]:   Default Transition Length: 1.0s
[14:45:12][C][light:177]:   Gamma Correct: 2.80
[14:45:12][C][restart:021]: Restart Switch 'Upstairs hall ESP restart'
[14:45:12][C][restart:021]:   Icon: 'mdi:restart'
**[14:45:12][C][homeassistant.time:010]: Home Assistant Time:
[14:45:12][C][homeassistant.time:011]:   Timezone: '+07-7'**
[14:45:12][C][ota:029]: Over-The-Air Updates:
[14:45:12][C][ota:030]:   Address: 192.168.2.102:8266
[14:45:12][C][ota:032]:   Using Password.
[14:45:12][C][api:103]: API Server:
[14:45:13][C][api:104]:   Address: 192.168.2.102:6053
[14:45:24][D][api:573]: Client 'Home Assistant 0.94.1 (192.168.2.99)' connected successfully!
[14:45:24][D][time:030]: Synchronized time: Tue Jun 25 07:45:24 2019
[14:45:36][D][switch:025]: 'Upstairs hall light' Turning OFF.
[14:45:36][D][switch:045]: 'Upstairs hall light': Sending state OFF
[14:45:36][D][switch:021]: 'timer1s' Turning ON.
[14:45:36][D][switch:045]: 'timer1s': Sending state ON
[14:45:36][D][switch:025]: 'motionofftimer' Turning OFF.
[14:45:36][D][switch:025]: 'Upstairs hall 20 min timer' Turning OFF.
[14:45:36][D][switch:045]: 'Upstairs hall 20 min timer': Sending state OFF
[14:45:36][D][switch:025]: 'Upstairs hall 2 h timer' Turning OFF.
[14:45:36][D][switch:045]: 'Upstairs hall 2 h timer': Sending state OFF
[14:45:37][D][switch:025]: 'timer1s' Turning OFF.
[14:45:37][D][switch:045]: 'timer1s': Sending state OFF
**[14:46:00][D][switch:025]: 'Upstairs hall motion sensor switch' Turning OFF.
[14:46:00][D][switch:045]: 'Upstairs hall motion sensor switch': Sending state OFF**

`

hpirila, 2019-06-25:
And yes, I can confirm same as others above. Even my timezone is +7, it works correctly with value UTC-7.

```
time:
  - platform: homeassistant
    timezone: UTC-7
    on_time:
      - seconds: 0
        minutes: 19
        hours: 18
        days_of_week: MON-FRI
        then:
          - switch.turn_off: motionswitch

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/upstairshalllight
Processing upstairshalllight (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/upstairshalllight/.pioenvs/upstairshalllight/src/main.cpp.o
Linking /data/upstairshalllight/.pioenvs/upstairshalllight/firmware.elf
Retrieving maximum program size /data/upstairshalllight/.pioenvs/upstairshalllight/firmware.elf
Building /data/upstairshalllight/.pioenvs/upstairshalllight/firmware.bin
Checking size /data/upstairshalllight/.pioenvs/upstairshalllight/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  43.1% (used 35280 bytes from 81920 bytes)
PROGRAM: [====      ]  36.0% (used 368412 bytes from 1023984 bytes)
========================= [SUCCESS] Took 35.15 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.2.102
INFO Uploading /data/upstairshalllight/.pioenvs/upstairshalllight/firmware.bin (372560 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.2.102 using esphome API
INFO Connecting to 192.168.2.102:6053 (192.168.2.102)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.2.102: timed out). Re-Trying in 1 seconds
INFO Connecting to 192.168.2.102:6053 (192.168.2.102)
INFO Successfully connected to 192.168.2.102
[18:16:33][I][app:096]: esphome version 1.13.6 compiled on Jun 25 2019, 18:15:37
[18:16:33][C][wifi:372]: WiFi:
[18:16:33][C][wifi:254]:   SSID: 

[redacted]
[18:16:33][C][wifi:255]:   IP Address: 192.168.2.102
[18:16:33][C][wifi:257]:   BSSID: 

[redacted]
[18:16:33][C][wifi:258]:   Hostname: 'upstairshalllight'
[18:16:33][C][wifi:262]:   Signal strength: -60 dB ▂▄▆█
[18:16:33][C][wifi:263]:   Channel: 3
[18:16:33][C][wifi:264]:   Subnet: 255.255.255.0
[18:16:33][C][wifi:265]:   Gateway: 192.168.2.1
[... 30 lines cut ...]
[18:16:33][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[18:16:33][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[18:16:33][C][esp8266_pwm:025]:   Inverted: YES
[18:16:34][C][logger:137]: Logger:
[18:16:34][C][logger:138]:   Level: DEBUG
[18:16:34][C][logger:139]:   Log Baud Rate: 115200
[18:16:34][C][logger:140]:   Hardware UART: UART0
[18:16:34][C][light:174]: Light 'led'
[18:16:34][C][light:176]:   Default Transition Length: 1.0s
[18:16:34][C][light:177]:   Gamma Correct: 2.80
[18:16:34][C][restart:021]: Restart Switch 'Upstairs hall ESP restart'
[18:16:34][C][restart:021]:   Icon: 'mdi:restart'
[18:16:34][C][homeassistant.time:010]: Home Assistant Time:
[18:16:34][C][homeassistant.time:011]:   Timezone: 'UTC-7'
[18:16:34][C][ota:029]: Over-The-Air Updates:
[18:16:34][C][ota:030]:   Address: 192.168.2.102:8266
[18:16:34][C][ota:032]:   Using Password.
[18:16:34][C][api:103]: API Server:
[18:16:34][C][api:104]:   Address: 192.168.2.102:6053
[18:16:44][D][api:573]: Client 'Home Assistant 0.94.1 (192.168.2.99)' connected successfully!
[18:16:44][D][time:030]: Synchronized time: Tue Jun 25 18:16:44 2019
[18:18:24][I][ota:046]: Boot seems successful, resetting boot loop counter.
[18:19:00][D][switch:025]: 'Upstairs hall motion sensor switch' Turning OFF.
[18:19:00][D][switch:045]: 'Upstairs hall motion sensor switch': Sending state OFF
```

closed 2019-07-03 by OttoWinter