esphome/issues #447: ESP32 ai thinker not connecting

CapnJimbo, 2019-06-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

currently running Home Assistant 0.93.2 on a RPi3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 ai thinker camera
ESPHome - currently running version 1.13.5
**Affected component:**

https://esphome.io/components/wifi.html
https://esphome.io/components/esp32_camera.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
```
esphome:
  name: cam1
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: xx
  password: xx
#  fast_connect: True

web_server:
  port: 80

api:
  password: 'xx'

ota:
  password: 'xx'

logger:

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: cam1
  # ... Frame Settings

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: cam1 light
```

```

**Traceback (if applicable):**

```
Wrote 933360 bytes (504751 compressed) at 0x00010000 in 44.8 seconds (effective 166.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[12:39:17]ets Jun 8 2016 00:22:57
[12:39:17]
[12:39:17]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:39:17]configsip: 0, SPIWP:0xee
[12:39:17]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:39:17]mode:DIO, clock div:2
[12:39:17]load:0x3fff0018,len:4
[12:39:17]load:0x3fff001c,len:952
[12:39:17]load:0x40078000,len:6084
[12:39:17]load:0x40080000,len:7944
[12:39:17]entry 0x40080310
[12:39:17]user code done
```

**Additional information and things you've tried:**

I validated the yaml file in esphome, then compiled it, downloaded the binary and flashed with esphomeflasher.

once completed  the ESP32 ai thinker wont connect to the wifi. 

I have a ttgo cam that connects to wifi and 2 ESP8266 that works and integrates to home assistant.

https://www.youtube.com/watch?v=-towjof_ixU&t=293s

_(tried with Arduino, camera web server - that was able to connect to wifi on the same module/ but i'd prefer a ESPHome option)_

Really appreciate any assistance

Chris

glmnet, 2019-06-26:
Check this https://github.com/esphome/issues/issues/479

brandond, 2019-06-26:
Duplicate of #479

closed 2019-06-26 by OttoWinter