esphome/issues #449: total daily energy sensor no longer resets at midnight

questrov, 2019-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemo D1_Mini

**Affected component:**
https://esphome.io/components/sensor/total_daily_energy.html

**Description of problem:**
total daily energy sensor no longer resets at midnight, not sure when/why. Don't think I changed anything, seemed to work great for ages then started to accumulate. Maybe one of the updates changed something? 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

`esphome:
  name: wemo_151
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "myssid"
  password: "mypassword"
  manual_ip:
    static_ip: 192.168.0.151
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D2
  scl: D1
  scan: True
  
sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "INA219 Current"
      filters:
        - multiply: 100
    power:
      name: "INA219 Power"
      filters: 
        - multiply: 100
    bus_voltage:
      name: "INA219 Bus Voltage"
    shunt_voltage:
      name: "INA219 Shunt Voltage"
    max_voltage: 26.0V
    max_current: 3.2A
    update_interval: 6s
    
# Example configuration entry

  - platform: total_daily_energy
    name: "Total Daily Energy"
    power_id: solar_power

  # The power sensor to convert, can be any power sensor
  - platform: ina219
    # ...
    power:
      id: solar_power
      filters: 
        - multiply: 100

# Enable time component to reset energy at midnight
time:
  - platform: sntp
    id: my_time    ``

**Traceback (if applicable):**
<!--
Please copy the traceback here if compilation is failing. If possible, also connect to the ESP over USB and copy its logs into the backticks.
-->
```

```

**Additional information and things you've tried:**

updated to latest versions both hassio and esphome, uploaded the yaml file again.  Have not hard rebooted HA or ESP

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

OttoWinter, 2019-06-17:
Works ok here. Please check if your SNTP time source is getting data. If it is, you will see `Synchronized time ...` messages in your logs. If the time cannot be synchronized, the total daily energy sensor will obviously not work. Also, since you appear to be using native API, consider using `homeassistant` time source.

questrov, 2019-06-17:
Thanks Otto!  For some reason (unknown to me) indeed not getting Synchronized time ... from SNTP.  Easy edit to homeassistant time source (with a little help from your docs, link below) and now get the Synchronized time message OK.  Thanks again!
https://esphome.io/components/time.html

OttoWinter, 2019-07-11:
Ok, closing

closed 2019-07-11 by OttoWinter