esphome/issues #450: Compile error with "update_on_boot" of my9231

freegink, 2019-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** pip

**ESP (ESP32/ESP8266, Board/Sonoff):** Sonoff B1 R2

**Affected component:** My9231

**Description of problem:**
After adding property "update_on_boot", ESPHome indicated "update_on_boot" is an invalid option for my9321.  

And I checked the source code, it doesn't seem that the const "CONF_UPDATE_ON_BOOT" was used anywhere. Correct me if I'm wrong.

**Problem-relevant YAML-configuration entries:**
```yaml
my9231:
  data_pin: GPIO12
  clock_pin: GPIO14
  num_channels: 6
  num_chips: 2
  update_on_boot: false

```

**Traceback (if applicable):**

```
Failed config

my9231:
  - [source sonoff_b1_r2.yaml:27]
    data_pin: GPIO12
    clock_pin: GPIO14
    num_channels: 6
    num_chips: 2

    [update_on_boot] is an invalid option for [0]. Please check the indentation.
    update_on_boot: True
```

**Additional information and things you've tried:**

OttoWinter, 2019-06-17:
Ah yes, I think I removed the option in 1.13 because ESPHome now can restore state from flash. Will update the documentation to remove that option.

john-arvid, 2019-09-15:
Maybe also change the note on this one?
https://esphome.io/components/output/my9231.html?highlight=my9291#sonoff-b1-configuration-example

And I think that since the restore state is used it can read the last state from flash instead from the my9231. (Not that it ever read from the my9231 chip, but it will store the last state it sent to the my9231)

closed 2019-10-18 by OttoWinter