esphome/issues #451: Repeat discovery of new devices

poldim, 2019-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA in docker, running latest 0.94.3 build.  ESPHome in docker, running latest 0.13.6 build.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Combination of Wemos D1 mini and Sonoff Basic for these three devices

**Affected component:**

Not sure if it's ESPHome or core HA issue.

**Description of problem:**
Using same standard wifi config for most of my devices.  These three are still in development so I don't want to add them to my production HA.  The appear repeatedly. These are up to 427 instances since I last restarted which is less than 24 hours ago.

[image]

**POSSIBLY Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: 'XXX'
  password: 'XXX'
  ap:
    ssid: HAN-ESPH
    password: 'XXX'
  reboot_timeout: XXX

api:

ota:

web_server:
  port: 80

logger:
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

[Tried to see if others are having a similar issue.
](https://www.reddit.com/r/homeassistant/comments/c0l2py/anyone_elses_discovery_showing_repeat_entries_on/
)
Recreated HA and ESPHome dockers

OttoWinter, 2019-06-17:
Appears to be an issue in Home Assistant core - their latest discovery to zeroconf migration has removed a "de-duplication" the discovery was previously doing with discovered items.

balloob, 2019-06-17:
Didn't ESPHome already had to de-duplicate discoveries after a restart of Home Assistant ?

closed 2019-06-17 by OttoWinter