esphome/issues #452: A component took a long time in a loop() cycle

phdelodder, 2019-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

**Description of problem:**
[app:077]: A component took a long time in a loop() cycle (0.4 s).
[V][app:078]: Components should block for at most 20-30ms in loop().
[V][app:079]: This will become a warning soon.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-17:
There are a couple of components that do this right now - that's why it's just a verbose message right now.

Please post your configuration as instructed in the issue template. The message itself does not include what component took long to process.

phdelodder, 2019-06-17:
@OttoWinter updated the orignal issue

glmnet, 2019-06-20:
I see my AC IR transmitters receiving this warning and it makes sense as the IR transmitter codes are long:
9 ms just for header, then 24 bits of 1ms aprox, then full code sent twice is 80ms roughly.

What is the "solution" here? Is this just loop or any call like climate->perform_call(...) etc. May be this climate call is triggered from a loop call and that loop call being the offender.
I'll try to add more info.

OttoWinter, 2019-06-20:
@glmnet Yeah, quite a few components are probably capable of showing such a message.

The reason I added it was mainly for developers: ESPHome's main loop() really just not be slowed down much (for example by blocking calls in some libraries) - the text was intended to quickly show developers that they're doing something wrong. ESPHome will mostly work fine with a slower loop too, but it can things like GPIO binary sensor events to slow down too.

In many cases though (for example remote_receiver), there's no way to do it without blocking, so some cases will always exist.

I'll remove the "will become a warning" message and make it more clear.

OttoWinter, 2019-07-11:
Closing due to inactivity - and this has been changed to no longer include the "this will become a warning" in dev.

closed 2019-07-11 by OttoWinter