esphome/issues #456: Error while uploading through GUI but fine when using webserver

poldim, 2019-06-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA in docker, running latest 0.94.3 build. ESPHome in docker, running latest 0.13.6 build.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Combination of Wemos D1 mini and Sonoff Basics and S31s

**Affected component:**

OTA Upload

**Description of problem:**
If I upload the file through the GUI via the upload button, the file compiles without issues, begins upload, but then stalls at some random percentage through the upload.
Pulling up the WEBSERVER of the device, and uploading the firmware (from compile > download binary) works without issues.

**Device WEBSERVER LOG:**
```
[W][ota:319]: Timeout waiting for data!
[W][ota:276]: Update end failed! Error: ERROR[0]: No Error
```

**Traceback (if applicable):**
<!--
Please copy the traceback here if compilation is failing. If possible, also connect to the ESP over USB and copy its logs into the backticks.
-->
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome_lightwell_ground_light
Processing esphome_lightwell_ground_light (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size .pioenvs/esphome_lightwell_ground_light/firmware.elf
Checking size .pioenvs/esphome_lightwell_ground_light/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  42.4% (used 34724 bytes from 81920 bytes)
PROGRAM: [====      ]  35.7% (used 365052 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.46 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome_lightwell_ground_light.local
INFO  -> 10.30.30.135
INFO Uploading /config/esphome_lightwell_ground_light/.pioenvs/esphome_lightwell_ground_light/firmware.bin (369200 bytes)
Uploading: [==                                                          ] 4% 
ERROR Error sending data: [Errno 32] Broken pipe
```

**Additional information and things you've tried:**

- I thought this was related to poor wifi experience, but I'm having this issue with a device that's 20 feet from my UniFi AP so that shouldn't be a factor.
- erasing rom and reflashing

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

OttoWinter, 2019-07-12:
Tagging this as a generic connection issue - I think it's just by chance the web server worked and normal OTA didn't.

closed 2019-07-12 by OttoWinter

poldim, 2019-07-14:
This happens to me on many physical devices that range in different
hardware.  I don't think it's an isolated issue.  The webserver always
works while OTA does not.