esphome/issues #457: ct-clamp component invalid readings

Varabaiba, 2019-06-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

HFsi, 2019-06-18:
I can confirm the issue with very similar config.
I have a board with 3 clamps, and all 3 show the same behaviour, so I don´t think it'd be a hardware malfunction.
With a constant load, the raw value changes from, eg, 0.5 to 0.52 or 0.48, so for now the only solution I could find is to define a window average so the values tend to a more stable reading, but I believe there's an issue here.
Please let me know if I can do some test for you, Otto

OttoWinter, 2019-06-19:
@Varabaiba That does sound interesting. So just to confirm I understand this correctly:

When the "true" value goes _up_, the CT clamp responds almost immediately.
When the "true" values goes _down_ the CT clamp responds with a slight delay?

The CT clamp code itself definitely does not have any averaging function across measurements - I think it must be a lower level issue with the interaction with (a) ADS1115 connection or (b) ADS1115 measurement mode.

I think ADS1115 might not be too optimized for ct_clamp use in ESPHome yet (it uses single-shot mode, so it's slow and you will not get many samples of the full sine wave).

If you can, could you confirm the same happens with built-in ADC?

@HFsi Can you confirm the thing you're writing here is the same thing @Varabaiba is seeing (lagged CT clamp value when going down)? If you're using the built-in ADC of the ESP I would expect measurement errors around the range you're seeing.

HFsi, 2019-06-19:
Not completely....this part happens exactly to me:
>  Every time controller reboots, ct_clamp readings start with ~0.50A and go gradually down until normalized

The rest, don´t...the readings go up and down fine (having in consideration the fluctuation of a stable known value)

> @HFsi Can you confirm the thing you're writing here is the same thing @Varabaiba is seeing (lagged CT clamp value when going down)? If you're using the built-in ADC of the ESP I would expect measurement errors around the range you're seeing.

Ok, but the error happens in any range....I believe that the value could be wrong, but if the load is stable, it should remain the same, not going from 0.1 to 0.2 to 0.25 to 0.13 and so.

I´m using an ADS1115 with multiplexer 'A0_GND'

I have a brand new ESP32, do you think that ADC could be more stable?

Varabaiba, 2019-06-19:
@OttoWinter I decided to show the problem in a visual way, which would provide much better understanding. So here is the chart pic from HA. On segment 1, you would see downward "ladder", where reading interval is 60 seconds, so you see it spans over thee hours to "normalize". Until normalized - it doesn't correlate with "true" readings from hardware meter in the power setup. Second segment is a regular activity, so I was misled regarding all downward moves are similar , after "normalization" regular appliance spikes are served properly up and down (look at kettle). But if you pay attention to  Segment 3 and segment 4, as I reduced reading interval down to 10s (S3) and 1s (S4), then rebooted, the timing for normalization shrinks accordingly. Which obviously could not be related to hardware, unless it has some internal buffering for data.
https://drive.google.com/file/d/16ZxX928FIX7rjSK0_upQOzZ7Ay7zuSoY/view?usp=sharing 
 Regarding ADS1115, I was following the article: https://poweruc.com/blogs/news/non-invasive-sensor-yhdc-sct013-000-ct-used-with-arduino-sct-013
After carefully revisiting it, I have no doubts on hardware capabilities, but a concern whether regular 200ms sampling is proper default. 
And, not to forget - thank you for your response.

HFsi, 2019-06-19:
> as I reduced reading interval down to 10s (S3) and 1s (S4), then rebooted, the timing for normalization shrinks accordingly

Same behaviour with mine

> but a concern whether regular 200ms sampling is proper default.

I've found that 300 ms gives more accurate values, but 400 ms makes no difference

Varabaiba, 2019-06-19:
@HFsi thanks for supporting this issue :)
I've slightly edited comment while understood 200ms is not actually a sampling rate but duration.

Varabaiba, 2019-06-19:
@HFsi By the way, we can assume there could not be connectivity or hardware failure problem as it would be hard to replicate between two of us. Still, as we have similar hardware, some specifics according to Otto, may exist. I'm planning to switch to internal ADC when possible and test accordingly.

OttoWinter, 2019-06-19:
I see the issue now, the offset of the sampled voltage value begins at 0.5V. That works well for ESP8266 ADC because its range is 0-1V, but it doesn't work well for ADS1115 when in higher gain modes.

The reason it's slowly adjusting is because of a low pass filter on the offset value.

HFsi, 2019-06-19:
My clamp gives 2.5 V with no load, so I have to use 4.096 gain in ADS1115

closed 2019-07-02 by OttoWinter