esphome/issues #459: ESP32 WiFi Signal and Uptime broken

Xeroxxx, 2019-06-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32: esp-wrover-kit

**Affected component:**

Uptime and WiFi Signal

**Description of problem:**

Uptime and WiFi Signal is broken. They do not report to Hass.io via Hass.io API.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```
      
  - platform: uptime
    name: ESPFlora Uptime
  
  - platform: wifi_signal
    name: ESPFlora WiFi Signal
    update_interval: 120s

switch:
  - platform: restart
    name: ESPFlora Restart

binary_sensor:
  - platform: status
    name: ESPFlora Status

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esp_ota

ota:
  password: !secret esp_ota

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-19:
Works fine here - please attach Home Assistant logs and logs from the device.

Xeroxxx, 2019-06-19:
Any other logs to check?

Nothing is listed in homeassistant log /config/home-assistant.log

````
[16:02:41][I][app:101]: esphome version 1.14.0-dev compiled on Jun 18 2019, 09:53:22
[16:02:42][C][wifi:392]: WiFi:
[16:02:42][C][wifi:274]:   SSID: [redacted]
[16:02:42][C][wifi:275]:   IP Address: 
[16:02:42][C][wifi:277]:   BSSID: [redacted]
[16:02:42][C][wifi:278]:   Hostname: 'espflora'
[16:02:42][C][wifi:282]:   Signal strength: -64 dB ▂▄▆█
[16:02:42][C][wifi:283]:   Channel: 6
[16:02:42][C][wifi:284]:   Subnet: 255.255.255.0
[16:02:42][C][wifi:285]:   Gateway: 
[16:02:42][C][wifi:286]:   DNS1: 
[16:02:42][C][wifi:287]:   DNS2: 0.0.0.0
[16:02:42][C][logger:175]: Logger:
[16:02:42][C][logger:176]:   Level: DEBUG
[16:02:42][C][logger:177]:   Log Baud Rate: 115200
[16:02:42][C][logger:178]:   Hardware UART: UART0
[16:02:42][C][esp32_ble_tracker:459]: BLE Tracker:
[16:02:42][C][esp32_ble_tracker:460]:   Scan Interval: 300 s
[16:02:42][C][xiaomi_miflora:012]: Xiaomi Mijia
[16:02:42][C][xiaomi_miflora:013]:   Temperature 'Paprika 1 Temperature'
[16:02:42][C][xiaomi_miflora:013]:     Unit of Measurement: '°C'
[16:02:42][C][xiaomi_miflora:013]:     Accuracy Decimals: 1
[16:02:42][C][xiaomi_miflora:013]:     Icon: 'mdi:thermometer'
[16:02:42][C][xiaomi_miflora:014]:   Moisture 'Paprika 1 Moisture'
[16:02:42][C][xiaomi_miflora:014]:     Unit of Measurement: '%'
[16:02:42][C][xiaomi_miflora:014]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:014]:     Icon: 'mdi:water-percent'
[16:02:42][C][xiaomi_miflora:015]:   Conductivity 'Paprika 1 Soil Conductivity'
[16:02:42][C][xiaomi_miflora:015]:     Unit of Measurement: 'µS/cm'
[16:02:42][C][xiaomi_miflora:015]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:015]:     Icon: 'mdi:flower'
[16:02:42][C][xiaomi_miflora:016]:   Illuminance 'Paprika 1 Illuminance'
[16:02:42][C][xiaomi_miflora:016]:     Unit of Measurement: 'lx'
[16:02:42][C][xiaomi_miflora:016]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:016]:     Icon: 'mdi:brightness-5'
[16:02:42][C][xiaomi_miflora:012]: Xiaomi Mijia
[16:02:42][C][xiaomi_miflora:013]:   Temperature 'Paprika 2 Temperature'
[16:02:42][C][xiaomi_miflora:013]:     Unit of Measurement: '°C'
[16:02:42][C][xiaomi_miflora:013]:     Accuracy Decimals: 1
[16:02:42][C][xiaomi_miflora:013]:     Icon: 'mdi:thermometer'
[16:02:42][C][xiaomi_miflora:014]:   Moisture 'Paprika 2 Moisture'
[16:02:42][C][xiaomi_miflora:014]:     Unit of Measurement: '%'
[16:02:42][C][xiaomi_miflora:014]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:014]:     Icon: 'mdi:water-percent'
[16:02:42][C][xiaomi_miflora:015]:   Conductivity 'Paprika 2 Soil Conductivity'
[16:02:42][C][xiaomi_miflora:015]:     Unit of Measurement: 'µS/cm'
[16:02:42][C][xiaomi_miflora:015]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:015]:     Icon: 'mdi:flower'
[16:02:42][C][xiaomi_miflora:016]:   Illuminance 'Paprika 2 Illuminance'
[16:02:42][C][xiaomi_miflora:016]:     Unit of Measurement: 'lx'
[16:02:42][C][xiaomi_miflora:016]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:016]:     Icon: 'mdi:brightness-5'
[16:02:42][C][xiaomi_miflora:012]: Xiaomi Mijia
[16:02:42][C][xiaomi_miflora:013]:   Temperature 'Zuckererbse 1 Temperature'
[16:02:42][C][xiaomi_miflora:013]:     Unit of Measurement: '°C'
[16:02:42][C][xiaomi_miflora:013]:     Accuracy Decimals: 1
[16:02:42][C][xiaomi_miflora:013]:     Icon: 'mdi:thermometer'
[16:02:42][C][xiaomi_miflora:014]:   Moisture 'Zuckererbse 1 Moisture'
[16:02:42][C][xiaomi_miflora:014]:     Unit of Measurement: '%'
[16:02:42][C][xiaomi_miflora:014]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:014]:     Icon: 'mdi:water-percent'
[16:02:42][C][xiaomi_miflora:015]:   Conductivity 'Zuckererbse 1 Soil Conductivity'
[16:02:42][C][xiaomi_miflora:015]:     Unit of Measurement: 'µS/cm'
[16:02:42][C][xiaomi_miflora:015]:     Accuracy Decimals: 0
[16:02:42][C][xiaomi_miflora:015]:     Icon: 'mdi:flower'
[... 96 lines cut ...]
[16:02:42][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:02:42][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
````

`````
        "sensor": [
            {
                "accuracy_decimals": 0,
                "icon": "mdi:timer",
                "key": 34555166,
                "name": "ESPFlora Uptime",
                "object_id": "espflora_uptime",
                "unique_id": "3c71bf1d1c38-uptime",
                "unit_of_measurement": "s"
            },
            {
                "accuracy_decimals": 0,
                "icon": "mdi:wifi",
                "key": 2880365672,
                "name": "ESPFlora WiFi Signal",
                "object_id": "espflora_wifi_signal",
                "unique_id": "3c71bf1d1c38-wifisignal",
                "unit_of_measurement": "dB"
            },
````

OttoWinter, 2019-06-20:
@Xeroxxx If you recently switch the ESP32, the uptime/wifi signal entities will have a new unique_id. Please check if you have entities with a `_2` appended to it in the entity registry.

Xeroxxx, 2019-06-20:
I didn't. I can't see any unlisted entities with any number. Can I manually call the ESP API to check for any measurements?

Xeroxxx, 2019-06-20:
Tried with three different ESP32 and setup a clean one. No change.

OttoWinter, 2019-06-20:
@Xeroxxx It appears you're using the dev branch - please check with the latest release version (1.13.6) instead and the latest dev branch (also: dev branch is not intended for production use!)

Xeroxxx, 2019-06-22:
Yes I'm using dev-branch. I know its not for production use.

OttoWinter, 2019-07-12:
Closing because it's probably fixed in latest dev

closed 2019-07-12 by OttoWinter